bluez/bluez #370: RegisterProfile don't call NewConnection with already connected device

crab2313, 2022-07-26:
I tried to register a client profile with following code (in Rust, but it's a simple wrapper of the dbus method call):

```rust
use bluer::{rfcomm::{Profile, Role}, Session};
use uuid::Uuid;
use tokio_stream::StreamExt;

#[tokio::main(flavor = "current_thread")]
async fn main() {
    let uuid = Uuid::parse_str("8901dfa8-5c7e-4d8f-9f0c-c2b70683f5f0").unwrap();

    println!("{uuid}");

    let profile =  Profile {
        uuid,
        auto_connect: Some(true),
        role: Some(Role::Client),
        ..Default::default()
    };

    let session = Session::new().await.unwrap();
    let mut handle = session.register_profile(profile).await.unwrap();

    while let Some(v) = handle.next().await {
        println!("GOT = {:?}", v);
    }
}
```

The device is connected already, but I can't get a new connection immediately.  To get  a new connection I must disconnect the device and connect it again. The following patch fix the issue for me but I don't know whether this is the intended behavior.

```diff
diff --git a/src/device.c b/src/device.c
index 9a4f2ce45..d5d5b1750 100644
--- a/src/device.c
+++ b/src/device.c
@@ -4574,7 +4574,7 @@ void device_probe_profile(gpointer a, gpointer b)
 
        device->services = g_slist_append(device->services, service);
 
-       if (!profile->auto_connect || !device->general_connect)
+       if (!profile->auto_connect)
                return;
 
        device->pending = g_slist_append(device->pending, service);
```

Vudentz, 2022-07-26:
@crab2313 I far I can remember this is by design so we don't trigger connection before the registration completes, but the fact that you need to disconnect that seems to be the real bug here, DeviceConnect shall connect all matching profiles.

Vudentz, 2022-07-26:
Actually let me take this back, I think it is fine provided that the profile is marked to auto-connect, that said I think we need to check if the device is already connect in order to bypass general_connect:

```
diff --git a/src/device.c b/src/device.c
index 9a4f2ce45..775003796 100644
--- a/src/device.c
+++ b/src/device.c
@@ -4574,7 +4574,8 @@ void device_probe_profile(gpointer a, gpointer b)
 
        device->services = g_slist_append(device->services, service);
 
-       if (!profile->auto_connect || !device->general_connect)
+       if (!profile->auto_connect || (!btd_device_is_connected(device) &&
+                                       !device->general_connect))
                return;
 
        device->pending = g_slist_append(device->pending, service);
```

crab2313, 2022-07-27:
> Actually let me take this back, I think it is fine provided that the profile is marked to auto-connect, that said I think we need to check if the device is already connected in order to bypass general_connect:
> 
> ```
> diff --git a/src/device.c b/src/device.c
> index 9a4f2ce45..775003796 100644
> --- a/src/device.c
> +++ b/src/device.c
> @@ -4574,7 +4574,8 @@ void device_probe_profile(gpointer a, gpointer b)
>  
>         device->services = g_slist_append(device->services, service);
>  
> -       if (!profile->auto_connect || !device->general_connect)
> +       if (!profile->auto_connect || (!btd_device_is_connected(device) &&
> +                                       !device->general_connect))
>                 return;
>  
>         device->pending = g_slist_append(device->pending, service);
> ```

This patch works. Thank you.

crab2313, 2022-07-27:
> Actually let me take this back, I think it is fine provided that the profile is marked to auto-connect, that said I think we need to check if the device is already connected in order to bypass [...]

This patch works. Thank you.

> Actually let me take this back, I think it is fine provided that the profile is marked to auto-connect, that said I think we need to check if the device is already connected in order to bypass [...]

This patch works. Thank you.

crab2313, 2022-07-28:
@Vudentz Will next version of bluez conatin this patch?

Vudentz, 2022-08-01:
> @Vudentz Will next version of bluez conatin this patch?

Yep, I just sent the patch upstream

closed 2022-08-02 by BluezTestBot