bluez/bluez #371: Unable to pair with BLE device

HelloHungryImDad, 2022-07-28:
I have been trying to pair with a BLE peripheral in order to use the INDICATE function on one of the characteristics. Using the nRF connect app I am able to pair, connect, and receive the output from the indicate function.

However from Manjaro linux (kernel 5.15.55-1-MANJARO) using bluez v5.64-2 I can connect to the device, but pairing always fails.

The procedure I have followed is thus:
1. Put device into pairing mode
2. Scan devices using bluetoothctl
3. Trust the device
4. Attempt to pair the device

I have tried various agents and settings in `/etc/bluetooth/main.conf` all with the same result. The key things I can see are that the authentication fails due to invalid parameters, and that bluez is unhappy about receiving an SMP: Identity Address Information packet from the device.

See logs below. Any help or advice would be greatly appreciated as I've exhausted all my options after several days of googling and trial and error.

bluetoothctl (scan section omitted):
```
[bluetooth]# trust B8:7C:6F:9A:EE:AA
Changing B8:7C:6F:9A:EE:AA trust succeeded
[bluetooth]# pair B8:7C:6F:9A:EE:AA
Attempting to pair with B8:7C:6F:9A:EE:AA
[CHG] Device B8:7C:6F:9A:EE:AA Connected: yes
[CHG] Device B8:7C:6F:9A:EE:AA Name: HPO-300T
[CHG] Device B8:7C:6F:9A:EE:AA Alias: HPO-300T
Failed to pair: org.bluez.Error.AuthenticationFailed
[CHG] Device B8:7C:6F:9A:EE:AA ServicesResolved: yes
```

btmon shows:
```
Bluetooth monitor ver 5.64
= Note: Linux version 5.15.55-1-MANJARO (x86_64)                                   0.702278
= Note: Bluetooth subsystem version 2.22                                           0.702284
= New Index: 74:4C:A1:74:81:34 (Primary,USB,hci0)                           [hci0] 0.702285
= Open Index: 74:4C:A1:74:81:34                                             [hci0] 0.702286
= Index Info: 74:4C:A1:74:81:34 (MediaTek, Inc.)                            [hci0] 0.702287
@ MGMT Open: bluetoothd (privileged) version 1.21                         {0x0001} 0.702288
@ MGMT Command: Pair Device (0x0019) plen 8                       {0x0001} [hci0] 19.588658
        LE Address: B8:7C:6F:9A:EE:AA (NXP (China) Management Ltd.)
        Capability: KeyboardDisplay (0x04)
< HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 8     #1 [hci0] 19.588831
        Own address type: Random (0x01)
        Filter policy: Ignore not in accept list, exc. directed unresolved RPA (0x03)
        PHYs: 0x01
        Entry 0: LE 1M
          Type: Passive (0x00)
          Interval: 60.000 msec (0x0060)
          Window: 60.000 msec (0x0060)
> HCI Event: Command Complete (0x0e) plen 4                             #2 [hci0] 19.698980
      LE Set Extended Scan Parameters (0x08|0x0041) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6         #3 [hci0] 19.699097
        Extended scan: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                             #4 [hci0] 19.699476
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 53                               #5 [hci0] 19.763769
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0013
            Props: 0x0013
              Connectable
              Scannable
              Use legacy advertising PDUs
            Data status: Complete
          Legacy PDU Type: ADV_IND (0x0013)
          Address type: Public (0x00)
[... 272 lines cut ...]
        Num handles: 1
        Handle: 512
        Count: 1
> ACL Data RX: Handle 512 flags 0x02 dlen 9                            #58 [hci0] 21.073246
      ATT: Error Response (0x01) len 4
        Read By Group Type Request (0x10)
        Handle: 0x0033
        Error: Attribute Not Found (0x0a)
< ACL Data TX: Handle 512 flags 0x00 dlen 9                            #59 [hci0] 21.082221
      ATT: Write Request (0x12) len 4
        Handle: 0x000f
          Data: 0200
> HCI Event: Number of Completed Packets (0x13) plen 5                 #60 [hci0] 21.122112
        Num handles: 1
        Handle: 512
        Count: 1
> ACL Data RX: Handle 512 flags 0x02 dlen 5                            #61 [hci0] 21.170517
      ATT: Write Response (0x13) len 0
```

dmesg shows:
```
[181822.611280] Bluetooth: hci0: Identity key blocked for b8:7c:6f:9a:ee:aa
[181822.611794] Bluetooth: hci0: unexpected SMP command 0x09 from b8:7c:6f:9a:ee:aa
```

Vudentz, 2022-07-28:
@HelloHungryImDad Looks like the device address is blocked, try removing it before pairing or use bluetoothctl> unblock

HelloHungryImDad, 2022-07-28:
@Vudentz Thanks for the suggestion. I tried this but it still produces the same error:

```
[bluetooth]# unblock B8:7C:6F:9A:EE:AA
Changing B8:7C:6F:9A:EE:AA unblock succeeded
[bluetooth]# remove B8:7C:6F:9A:EE:AA
[DEL] Device B8:7C:6F:9A:EE:AA BLEsmart_00060002B87C6F9AEEAA
Device has been removed
[bluetooth]# scan on
Discovery started
[CHG] Controller 74:4C:A1:74:81:34 Discovering: yes
[CHG] Device 09:AB:F2:21:C2:64 RSSI: -94
[NEW] Device B8:7C:6F:9A:EE:AA BLEsmart_00060002B87C6F9AEEAA
[CHG] Device 62:4F:ED:CE:90:E8 RSSI: -93
[CHG] Device 62:4F:ED:CE:90:E8 TxPower: 26
[CHG] Device 64:8B:70:81:A1:9B RSSI: -91
[CHG] Device 64:8B:70:81:A1:9B TxPower: 5
[bluetooth]# scan off
Discovery stopped
[CHG] Device 64:8B:70:81:A1:9B TxPower is nil
[CHG] Device 64:8B:70:81:A1:9B RSSI is nil
[CHG] Device 62:4F:ED:CE:90:E8 TxPower is nil
[CHG] Device 62:4F:ED:CE:90:E8 RSSI is nil
[CHG] Device B8:7C:6F:9A:EE:AA RSSI is nil
[CHG] Device 09:AB:F2:21:C2:64 RSSI is nil
[CHG] Controller 74:4C:A1:74:81:34 Discovering: no
[bluetooth]# trust B8:7C:6F:9A:EE:AA
[CHG] Device B8:7C:6F:9A:EE:AA Trusted: yes
Changing B8:7C:6F:9A:EE:AA trust succeeded
[DEL] Device 39:CB:CC:78:49:EE 39-CB-CC-78-49-EE
[bluetooth]# pair B8:7C:6F:9A:EE:AA
Attempting to pair with B8:7C:6F:9A:EE:AA
[CHG] Device B8:7C:6F:9A:EE:AA Connected: yes
Failed to pair: org.bluez.Error.AuthenticationFailed
```

`bluetoothctl info` provides the following if this is of any help:
```
[bluetooth]# info B8:7C:6F:9A:EE:AA
Device B8:7C:6F:9A:EE:AA (public)
        Name: HPO-300T
        Alias: HPO-300T
        Paired: no
        Trusted: yes
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Current Time Service      (00001805-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (6e400001-b5a3-f393-efa9-e50e24dcca9e)

```

Vudentz, 2022-07-28:
Check if the device is not using one of these keys:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=0a0f54b40af878206dac2c822f9997f7d03355ae

HelloHungryImDad, 2022-07-28:
That would explain it then. It's not a Google Titan, but the IRK matches as you say. It's a very unfortunate collision it's by chance. I'll speak to the manufacturer to see what they can so. Thank you very much for the prompt solution @Vudentz. It's been driving me crazy!

closed 2022-07-28 by HelloHungryImDad