bluez/bluez #372: Sixaxis gamepad gets removed from trusted list after plugging it in with a USB cable

d3p4z, 2022-08-01:
I can setup my sixaxis gamepad successfully to make it work using bluetooth. But every time I connect it with the USB cable it gets removed from my trusted list and I get a new authorization request to set it up again. I use it with the cable more often than not, but I find myself keeping accepting authorization requests to be sure it works whenever I want to use it wirelessly.

I wonder if would be possible to add a configuration option to autoaccept authorizations from a given address+UUID, or make bluez detect that the device is already configured and skip it.

I'm using bluez 5.64 under Void Linux, no systemd.

kakra, 2022-08-01:
If this is an Xbox controller, please try upgrading its firmware.

kakra, 2022-08-01:
Okay, you can try switching on `JustWorksRepairing=always`. I was just pointing out the Xbox controller because their older firmware versions are known to have pairing problems, it's only stable since the BLE firmwares.

- https://www.phoronix.com/news/BlueZ-5.54-Released
- https://patchwork.kernel.org/project/bluetooth/patch/20200212212729.Bluez.v2.1.I333a90ad3c75882c6f008c94a28ca7d3e8f6c76e@changeid/

d3p4z, 2022-08-01:
I really don't see any difference after switching on `JustWorksRepairing=always` in `/etc/bluetooth/main.conf`. I keep getting the "authorize service" requests. I've restarted the daemon and the full system just in case.

The device never appears as paired, only trusted. I don't know if that's the expected behavior.
To make it work wirelessly, I first connected the pad using USB and I got the "Authorize service" request from bluetoothctl. Once accepted, the pad is added to the device list and trusted, and remains in my device list working wirelessly. But it isn't paired, I've not been able to pair it like I've done with other devices.

Here's how it shows in bluetoothctl:
```
[bluetooth]# devices
Device 44:D8:32:A1:C3:D9 Sony PLAYSTATION(R)3 Controller
[bluetooth]# info 44:D8:32:A1:C3:D9
Device 44:D8:32:A1:C3:D9 (public)
	Name: Sony PLAYSTATION(R)3 Controller
	Alias: Sony PLAYSTATION(R)3 Controller
	Class: 0x00000508
	Icon: input-gaming
	Paired: no
	Trusted: yes
	Blocked: no
	Connected: no
	WakeAllowed: yes
	LegacyPairing: no
	UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v054Cp0268d0000
```
Every time I connect the pad through USB, it gets removed from the trusted list and asks me authorization:
```
[CHG] Device 44:D8:32:A1:C3:D9 Trusted: no
Authorize service
[agent] Authorize service 00001124-0000-1000-8000-00805f9b34fb (yes/no): yes
[CHG] Device 44:D8:32:A1:C3:D9 Trusted: yes
[CHG] Device 44:D8:32:A1:C3:D9 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device 44:D8:32:A1:C3:D9 Connected: yes
```

This is what the logs show me every time I connect it through USB:
```
2022-08-01T19:11:56.31425 daemon.notice: Aug  1 21:11:56 bluetoothd: bluetoothd[949]: sixaxis: compatible device connected: Sony PLAYSTATION(R)3 Controller (054C:0268 /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1:1.0/0003:054C:0268.0003/hidraw/hidraw0)
2022-08-01T19:11:56.31436 daemon.info: Aug  1 21:11:56 bluetoothd[949]: sixaxis: compatible device connected: Sony PLAYSTATION(R)3 Controller (054C:0268 /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1:1.0/0003:054C:0268.0003/hidraw/hidraw0)
2022-08-01T19:11:56.31902 daemon.info: Aug  1 21:11:56 bluetoothd[949]: sixaxis: setting up new device
2022-08-01T19:11:56.31909 daemon.notice: Aug  1 21:11:56 bluetoothd: bluetoothd[949]: sixaxis: setting up new device
```

Vudentz, 2022-08-01:
> I really don't see any difference after switching on `JustWorksRepairing=always` in `/etc/bluetooth/main.conf`. I keep getting the "authorize service" requests. I've restarted the daemon and the [...]

So the sixaxis plugin seems to always do a new setup every time you plug it over USB, it seems it was done on purpose to but perhaps we could store some information that can be used to authorize instead of asking the agent to authorize everytime, in fact trusted is what we normally use to skip the authorization but the plugin is automatically resetting everytime you plug it, instead I think what we should do is leave up to the user to trust it or not and then we can just check if already trusted skip authorizing again instead of flipping it automatically which doesn't respect the user's preference.

d3p4z, 2022-08-02:
@Vudentz Leaving up to the user to trust it or not, and checking if it's trusted to decide if future authorizations are really needed sounds perfect to me. I hope something like that can be implemented.

d3p4z, 2022-08-02:
@Vudentz I tried your changes (applied your pr as a patch) and I didn't receive authorization requests at all when plugging in the pad, even after untrusting the device and removing it from the device list (it was there from before). I had to revert your changes to be able to receive the initial authorization request.

The bluetoothd log didn't show anything when plugging in the pad, and I was running it with debug enabled.

d3p4z, 2022-08-02:
I've been doing some quick tests. Keeping your changes but changing:
```
	if (device && btd_device_has_uuid(device, HID_UUID) &&
			(btd_device_is_connected(device) ||
			 device_is_trusted(device))) {
```
To just:
```
       if (device && btd_device_has_uuid(device, HID_UUID) &&
                       btd_device_is_connected(device)) {
```
In `plugins/sixaxis.c` lines 338-340 makes the fix work as expected here. I get authorization requests ONLY when the device isn't trusted / not added to the device list. I only reverted the test to like it was before, but seems that your removal of the two `btd_device_set_trusted` calls was enough.

I don't have prior C coding experience or know enough about this project, but the fact that the test is working fine on my side only after such change doesn't make any sense to me. Was that `device_is_trusted(device)` test passing when it shouldn't?

Vudentz, 2022-08-02:
> ```
> 2022-08-02T13:21:21.82289 daemon.err: Aug  2 15:21:21 bluetoothd[951]: src/plugin.c:plugin_init() Can't load plugin /usr/lib64/bluetooth/plugins/sixaxis.so: /usr/lib64/bluetooth/plugins/sixaxis.so: undefined symbol: device_is_trusted
> ```
> 
> This must be the reason why things don't work when `device_is_trusted` is in the if check.

Yep, we need to add btd_ prefix in order for the function to be accessible by plugins.

d3p4z, 2022-08-02:
> Yep, we need to add btd_ prefix in order for the function to be accessible by plugins.

I can't make the pad work after applying your latest commit.
I get this warning during compilation:
```
  CC       plugins/sixaxis_la-sixaxis.lo
plugins/sixaxis.c: In function 'setup_device':
plugins/sixaxis.c:340:5: warning: implicit declaration of function 'btd_device_is_trusted'; did you mean 'btd_device_set_trusted'? [-Wimplicit-function-declaration]
  340 |     btd_device_is_trusted(device))) {
      |     ^~~~~~~~~~~~~~~~~~~~~
      |     btd_device_set_trusted
```
Compiles, and when I start the daemon it outputs this error to the log:
```
2022-08-02T16:52:41.26896 daemon.err: Aug  2 18:52:41 bluetoothd[954]: src/plugin.c:plugin_init() Can't load plugin /usr/lib64/bluetooth/plugins/sixaxis.so: /usr/lib64/bluetooth/plugins/sixaxis.so: undefined symbol: btd_device_is_trusted
```

Vudentz, 2022-08-02:
@d3p4z you will need to apply both patches in the set:

https://patchwork.kernel.org/project/bluetooth/patch/20220802163441.2522333-1-luiz.dentz@gmail.com/

closed 2022-08-02 by d3p4z