bluez/bluez #373: After resume the host stops listening to incoming Bluetooth pairing requests

birdie-github, 2022-08-01:
This is a dupe of https://bugzilla.kernel.org/show_bug.cgi?id=216314 but I'm not sure who's the culprit so I'm filing the report in both projects.

This looks like a regression in kernel 5.18 or bluez. I've upgraded both in the meantime so it's hard to say. The device is: Intel Corp. Wireless-AC 9260 Bluetooth Adapter

After booting, my BT headphones connect to my host automatically. Once I suspend, this no longer happens, there's **nothing in the system log**:
```
# bluetoothctl
Agent registered
AdvertisementMonitor path registered

[ I'm switching on my BT headphones - no messages, nothing, ok, switching them off ]

[bluetooth]# power off
Changing power off succeeded
[CHG] Controller 48:89:E7:XX:YY:ZZ Powered: no
[CHG] Controller 48:89:E7:XX:YY:ZZ Discovering: no
[CHG] Controller 48:89:E7:XX:YY:ZZ Class: 0x00000000
[bluetooth]# power on
[CHG] Controller 48:89:E7:XX:YY:ZZ Class: 0x007c0104
Changing power on succeeded
[CHG] Controller 48:89:E7:XX:YY:ZZ Powered: yes

[ I'm switching on my BT headphones again ]

[CHG] Device 17:10:28:AA:BB:CC Connected: yes
[NEW] Endpoint /org/bluez/hci0/dev_17_10_28_AA_BB_CC/sep1 
[NEW] Endpoint /org/bluez/hci0/dev_17_10_28_AA_BB_CC/sep2 
[NEW] Transport /org/bluez/hci0/dev_17_10_28_AA_BB_CC/fd1 
[CHG] Transport /org/bluez/hci0/dev_17_10_28_AA_BB_CC/fd1 Volume: 0x000f (15)
[DEL] Transport /org/bluez/hci0/dev_17_10_28_AA_BB_CC/fd1 
[NEW] Transport /org/bluez/hci0/dev_17_10_28_AA_BB_CC/sep1/fd2 
[CHG] Transport /org/bluez/hci0/dev_17_10_28_AA_BB_CC/sep1/fd2 Volume: 0x000f (15)
```

If I click the bluman system tray icon -> Recent connections -> My BT headphones -> It connects and works but it's not convenient at all because it worked automatically before.

I'm looking at

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/log/drivers/bluetooth/btintel.c

and there's nothing which could indicate a regression. Maybe it's actually an issue with bluez?

birdie-github, 2022-08-01:
Software configuration:
```
Fedora 36 with all updates installed
Kernel: 5.8.10
BlueZ: bluez-5.65-1.fc36.x86_64
```

birdie-github, 2022-08-01:
After power cycling the adapter, all consequent BT pairing attempts succeed.

kparal, 2022-08-01:
Downstream Fedora bug with some details and lots of people affected:
https://bugzilla.redhat.com/show_bug.cgi?id=2100761

birdie-github, 2022-08-01:
According to multiple comments it's a regression in the kernel, not bluez, but no one has any clue why. Maybe bluez developers could check it and give their insight. Lost of people are affected.

Of course, one could perform a regression testing (`git bisect`) but at the moment I'm not ready to do it.

Vudentz, 2022-08-01:
@birdie-github Please add the HCI trace when resuming, leave btmon running before suspending it should record the HCI command sequence when resuming perhaps something is missing.

birdie-github, 2022-08-01:
@Vudentz 

[bluetooth.dump.gz](https://github.com/bluez/bluez/files/9237718/bluetooth.dump.gz)

Vudentz, 2022-08-01:
@birdie-github Looks like disable Scan during suspend but we don't re-enable it during resume:

```
< HCI Command: Write Scan Enable (0x03|0x001a) plen 1                                                                                                                                                                                                                     #9 [hci0] 12.425215
        Scan enable: No Scans (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                                                              #10 [hci0] 12.429189
      Write Scan Enable (0x03|0x001a) ncmd 2
        Status: Success (0x00)
```

Looks like the following shall have it fixed, but it probably need to be tagged for stable:

https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/commit/?id=68253f3cd715e819bc4bff2b0e6b21234e259d56

birdie-github, 2022-08-01:
Thanks a ton, I will report it to Fedora. Please delete my previous comment with the dump as it contains a lot of private info I don't want to disclose. I thought hiding it would hide it for everyone but it still can be seen.

closed 2022-08-01 by birdie-github

birdie-github, 2022-08-01:
Thanks a ton, I will report it to Fedora.

birdie-github, 2022-08-01:
I've applied the patch and reloaded the module (`bluetooth.ko`) without rebooting however the issue persists :( Maybe a full reboot is necessary but that seems highly unlikely. Normally when you `rmmod` and `modprobe` a module, its new version is loaded from the disk and the module on the disk is now new with a patch applied.

Vudentz, 2022-08-02:
> I've applied the patch and reloaded the module (`bluetooth.ko`) without rebooting however the issue persists :( Maybe a full reboot is necessary but that seems highly unlikely. Normally when you [...]

But it shall have caused a reset which would cleanup the states which was the reason why power off/power on made it work again, so Im quite surprised you are able to reproduce it under this conditions.

birdie-github, 2022-08-02:
OK, a full reboot has fixed the issue. No idea why a module reloading didn't.

kparal, 2022-08-02:
Artem, thanks a lot for a great work. I wonder how to best poke kernel devs to make sure this is tagged stable? In my experience they don't watch kernel bugzilla too much.

birdie-github, 2022-08-02:
Kamil, that's nothing. I've pinged all the three developers related to the patch, let's hope at least one of them will submit it to stable.  My experience of dealing with Greg Kroah-Hartman (he's responsible for stable releases) directly has been a complete failure, so I've long stopped writing to him.

bam80, 2025-01-27:
Hi guys, I found this from #1079.

I'm having hard time to find out if [the patch](https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/commit/?id=68253f3cd715e819bc4bff2b0e6b21234e259d56) here is actually merged upstream and thus in my Fedora's 6.12.9-200.fc4 Linux kernel, could you help?
It looks to me it was upstreamed long time ago, at least since [Linux 5.18.18](https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.18.18), am I right?
Thanks.