bluez/bluez #379: Bluetooth issue with my mouse after resume from sleep

iant321, 2022-08-23:
I posted my issue in the blueman-project (#1833), but after some responses it seems that it belongs here. So I post it here with findings from the other thread.

blueman: 2.3.2
BlueZ: 5.65
Distribution: Fedora 36
Desktop environment: KDE
Kernel: 5.18.17

I have a bluetooth mouse Logitech M720 and a Lenovo T460 laptop. Here is the scenario to reproduce my issue. Initially, the laptop is in the sleep state with the lid closed. When I open the lid, I see the kde lock screen. If I begin to move my external mouse immediately after this (as I have to position my mouse pointer in the password field), the mouse pointer does not react, and after unlocking with my trackpad/keyboard I see that bluetooth is off. The blueman icon in the system tray is present, with the red cross over its image. I can click on it and switch bluetooth on, after that my external mouse is connected and works ok.

However, if, while on the lock screen, I wait for 4-5 seconds, and only after that begin to move the external mouse, it gets connected, the mouse pointer moves and everything is ok.

The logs show that the difference between the two cases. In the first case, there are several messages like

Aug 21 12:12:23 apollo bluetoothd[48786]: profiles/deviceinfo/deviceinfo.c:read_p
npid_cb() Error reading PNP_ID value: Request attribute has encountered an unlike
ly error

BEFORE the lines

Aug 21 12:12:23 apollo systemd[1]: Starting blueman-mechanism.service - Bluetooth management mechanism...
Aug 21 12:12:23 apollo systemd[1]: Started blueman-mechanism.service - Bluetooth management mechanism.

In the second case, there are several messages AFTER blueman is started, showing that the mouse was successfully detected.

For the first case, the desired behavior would be that my bluetooth mouse would be irresponsive for a couple of seconds and then would connect after bluetooth is switched on.

The full log is in the original thread: https://github.com/blueman-project/blueman/issues/1833

Thanks for reading.

Vudentz, 2022-08-23:
Please add the HCI traces (using btmon).

iant321, 2022-08-23:
Thanks for your reply. I attach two files. Scenario1.log is for the first scenario (immediate mouse movement, bluetooth locks in off state), Scenario2.log is the second (on the lock screen, wait for ~4sec before starting to move the mouse), it gets connected as expected. The files are pretty big but I did not cut them as I do not now which parts are irrelevant. It looks like only several first and end lines of the files contain significant information.

[Scenario1.log](https://github.com/bluez/bluez/files/9410357/Scenario1.log)
[Scenario2.log](https://github.com/bluez/bluez/files/9410363/Scenario2.log)

iant321, 2022-09-21:
No response for 29 days... Anyone?

Vudentz, 2022-09-22:
```
= Close Index: E4:A7:A0:8D:FC:79                               [hci0] 18.908947
@ MGMT Event: Index Removed (0x0005) plen 0           {0x0001} [hci0] 18.908953
= Delete Index: E4:A7:A0:8D:FC:79                              [hci0] 18.908959
= New Index: 00:00:00:00:00:00 (Primary,USB,hci0)              [hci0] 19.156625
= Open Index: 00:00:00:00:00:00                                [hci0] 19.160808
= Index Info: 00:00:00:00:00:00 (Intel Corp.)                  [hci0] 19.160813
```

The device is being unregistered during resume, then we have to reload the entire firmware and perform the init sequence, so we only really start connecting much later:

```
< HCI Command: LE Create Connec.. (0x08|0x000d) plen 25  #5008 [hci0] 23.628549
        Scan interval: 60.000 msec (0x0060)
        Scan window: 60.000 msec (0x0060)
        Filter policy: Accept list is not used (0x00)
        Peer address type: Random (0x01)
        Peer address: C7:D5:9A:A2:E1:9A (Static)
        Own address type: Public (0x00)
        Min connection interval: 7.50 msec (0x0006)
        Max connection interval: 11.25 msec (0x0009)
        Connection latency: 44 (0x002c)
        Supervision timeout: 2160 msec (0x00d8)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                #5009 [hci0] 23.630534
      LE Create Connection (0x08|0x000d) ncmd 2
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 19                #5010 [hci0] 23.784512
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 3585
        Role: Central (0x00)
        Peer address type: Random (0x01)
        Peer address: C7:D5:9A:A2:E1:9A (Static)
        Connection interval: 11.25 msec (0x0009)
        Connection latency: 44 (0x002c)
        Supervision timeout: 2160 msec (0x00d8)
        Central clock accuracy: 0x00
```

On my laptop (Lenovo P1-Gen2) running fedora:

```
@ MGMT Event: Controller Suspended (0x002d) plen 1                                                             {0x0001} [hci0] 17:00:24.214149
        Suspend state: Disconnected and not scanning (1)
< HCI Command: Set Event Mask (0x03|0x0001) plen 8                                                                #6109 [hci0] 17:00:37.480587
        Mask: 0x3dbff807fffbffff
          Inquiry Complete
          Inquiry Result
          Connection Complete
          Connection Request
          Disconnection Complete
          Authentication Complete
          Remote Name Request Complete
          Encryption Change
          Change Connection Link Key Complete
          Link Key Type Changed
          Read Remote Supported Features Complete
          Read Remote Version Information Complete
          QoS Setup Complete
          Command Complete
          Command Status
          Hardware Error
          Flush Occurred
          Role Change
          Mode Change
          Return Link Keys
          PIN Code Request
          Link Key Request
          Link Key Notification
          Loopback Command
          Data Buffer Overflow
          Max Slots Change
          Read Clock Offset Complete
          Connection Packet Type Changed
          QoS Violation
          Page Scan Mode Change
          Page Scan Repetition Mode Change
          Flow Specification Complete
          Inquiry Result with RSSI
          Read Remote Extended Features Complete
          Synchronous Connection Complete
          Synchronous Connection Changed
          Sniff Subrating
          Extended Inquiry Result
          Encryption Key Refresh Complete
          IO Capability Request
          IO Capability Request Reply
          User Confirmation Request
          User Passkey Request
          Remote OOB Data Request
          Simple Pairing Complete
          Link Supervision Timeout Changed
          Enhanced Flush Complete
          User Passkey Notification
          Keypress Notification
          Remote Host Supported Features Notification
          LE Meta
> HCI Event: Command Complete (0x0e) plen 4                                                                       #6110 [hci0] 17:00:37.482114
      Set Event Mask (0x03|0x0001) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Controller Resumed (0x002e) plen 8                                                               {0x0001} [hci0] 17:00:37.482191
        Wake reason: Resume from non-Bluetooth wake source (0)
        BR/EDR Address: 00:00:00:00:00:00 (OUI 00-00-00)
= bluetoothd: Controller resume with wake event 0x0                                                                            17:00:37.485341

```

The adapter doesn't not disappear when resuming we don't have to do much just update the Event Mask, so I suspect something is cutting the power of USB bus which effectively make resume behave like a cold boot, that said I don't know why it would cause to not resume powered, well perhaps because AutoEnable is not set:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/main.conf#n285

iant321, 2022-09-22:
In my /etc/bluetooth/main.conf AutoEnable=True. I checked this first thing. The bluetooth adapter _does_ resume powered. As I said in the OP, If I do not touch my external bluetooth mouse for a few seconds after opening the lid, everything is ok and bluetooth is on, connecting the mouse etc. It is if I do move the mouse within 3-4 seconds upon opening the lid, the powered bluetooth reacts and switches itself off. I suspect that this has something to do with the error lines 

= bluetoothd: profiles/input/hog-lib.c:report_read_cb() Error rea..   20.111369

at the end of Scenario1.log log file. I do not mind waiting for a bit longer to allow the firmware to be loaded. However, I would expect that until this is completed, my mouse would just stay irresponsive and would not interfere with the bluez soft. The problem is that it _does_ interfere. Somehow the powered on bluetooth detects the motion of the mouse, begins some operations and fails, resulting in switching it off.

Vudentz, 2022-09-22:
> Vudentz, thanks for your reply. In my /etc/bluetooth/main.conf AutoEnable=True. I checked this first thing. Also, In the KDE settings - Power management, bluetooth "On" option is set.
> 
> OK, I see that for some reason my bluetooth adapter is being unregistered while sleeping. Still moving the mouse while the init process is unfinished should not break the process. I would expect that until it is completed, my mouse would just stay irresponsive. I suspect that the problem has something to do with the error lines
> 
> = bluetoothd: profiles/input/hog-lib.c:report_read_cb() Error rea.. 20.111369
> 
> at the end of Scenario1.log log file.
> 
> So we now have two questions: (1)why is my bluetooth adapter is being unregistered during sleep; (2)why moving the mouse while the init process is unfinished results in bluetooth switched off.

Well something is switching Bluetooth Off:

```
@ MGMT Command: Set Powered (0x0005) plen 1           {0x0001} [hci0] 20.086153
        Powered: Disabled (0x00)
```

Which causes the device to be disconnected:

```
< HCI Command: Disconnect (0x01|0x0006) plen 3           #4926 [hci0] 20.097177
        Handle: 3585 Address: C7:D5:9A:A2:E1:9A (Static)
        Reason: Remote Device Terminated due to Power Off (0x15)
```

> It is likely that if I resolve the first issue, the second will not happen. Could you advice where to look? Thanks.

Given the following line:

`= bluetoothd: src/adv_monitor.c:btd_adv_monitor_power_down() Unex..   20.086267`

Im pretty sure this is happening on:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/adapter.c#n3115

Which is only called if some application is setting Powered to False, so that is _not_ the daemon doing it automatically but something else turning Bluetooth off and by that disconnecting.

iant321, 2022-09-24:
I disabled the module PowerManager in blueman manager settings (which also disabled KillSwitch) and the problem went away (at least when I closed and opened the laptop lid for 5 times). Before I posted my initial message I (following some suggestions from Internet) tried to check/uncheck the checkbox "Automatic switch on" (not sure about exact English translation as my system is non-English) in the PowerManager settings. This did not have any effect. However completely disabling the manager apparently solved the problem. Thanks to all who helped me with the issue.

iant321, 2022-09-24:
Sorry, accidentally closed the issue. I leave it open for now so that if anyone wants to comment, he/she can do it. Will close in a couple of days.

closed 2022-09-27 by iant321