bluez/bluez #380: How to return a particular GATT error code?

weliem, 2022-08-23:
My application is a Peripheral and when a read/write request for a characteristic comes in I return the value or an error.

I now need to return a particular GATT error code, e.g. 0x81 which is an application specific error code as used in some BT SIG services.

However, it looks like I can only return a string like "org.bluez.Error.NotPermitted". 

How do I return a code?

Vudentz, 2022-08-23:
@weliem Here is the mapped errors:

https://github.com/bluez/bluez/blob/master/src/gatt-database.c#L2195

So we are able to return 0x80 but not 0x81, not sure if that perhaps is a typo and should have been 0x81 though, btw what service is that requires an application error?

weliem, 2022-08-23:
@Vudentz  Thanks for sharing the link of where the errors are mapped. I wonder why it is done this way? Is there any benefit to mapping errors as opposed to simply returning a valid GATT error code? It looks like an unnecessary limitation....

In the bluetooth standard it says:

```
GATT - Section 4.9.5:
 
If the Characteristic Value that is written is the wrong size, or has an invalid value as defined by the profile, then the write shall not succeed, and an ATT_ERROR_RSP PDU with the Error Code parameter set to Application Error (0x80 – 0x9F) shall be sent by the server.
 
Application Error 0x80 – 0x9F Application error code defined by a higher layer specification.

```

I am implementing the BTSIG [GLS profile](https://www.bluetooth.com/specifications/specs/glucose-service-1-0-1/) which uses application error codes in the 0x80-0x9F range. This service defines the following Attribute Protocol Application Error codes (page 8, section 1.7 Error Codes):
- Procedure Already in Progress **0x80**
- Client Characteristic Configuration descriptor improperly configured **0x81**

CTS, RSCS, IDS, PAMS, HRS, and UDS also define their own error codes so it is quite common it the BT SIG profiles. Therefore I was hoping I would be in full control of which ATT error code I return. How about allowing a string with the number, e.g. "128" instead of "org.bluez.Error.XYZ"?

Vudentz, 2022-08-23:
> @Vudentz Thanks for sharing the link of where the errors are mapped. I wonder why it is done this way? Is there any benefit to mapping errors as opposed to simply returning a valid GATT error code? It looks like an unnecessary limitation....

Right, because application do care about following the standard correctly and would never return something invalid breaking the stack in the process... The application errors was actually introduced past Bluetooth 4.0 and if you ask me that is a mistake because it forces applications to be compliant but they are almost never part of the qualification listing.

> In the bluetooth standard it says:
>
> ```
> GATT - Section 4.9.5:
>
> If the Characteristic Value that is written is the wrong size, or has an invalid value as defined by the profile, then the write shall [...]

And decoding these errors is also application specific so the ATT layer has no idea of what is going on... brilliant idea by the spec authors, I do wonder what happens if application return an invalid error which leads to disconnect or something that forces to pairing in a loop, if that is within 0x80 - 0x9F the stack can't even fail because that is valid range.

> I am implementing the BTSIG [GLS profile](https://www.bluetooth.com/specifications/specs/glucose-service-1-0-1/) which uses application error codes in the 0x80-0x9F range. This service defines the [...]

There is an error message we could utilize, that said encoding it as error message is kind of a hack of you ask me, but we can probably check if is within 0x80-0x9F range, but anyway if the error cause broken interaction or some sort of vulnerability we can no longer prevent that.

ErikMoll, 2022-08-24:
In the Bluetooth Core Specification the ATT error codes are defined in volume 3, part F - ATT, section 3.4.1.
Note that this table includes 2 ranges for error codes that are used by higher layers, such as GATT services:

- Application Error codes use 0x80 – 0x9F.
- Common Profile and Service Error Codes use 0xE0 – 0xFF and are defined in the Core Specification Supplement.

There are many SDKs that do define the same set, e.g. [TI](https://software-dl.ti.com/simplelink/esd/simplelink_cc2640r2_sdk/3.40.00.10_new/exports/docs/blestack/ble_user_guide/doxygen/ble/html/group___a_t_t___e_r_r___c_o_d_e___d_e_f_i_n_e_s.html) and [Zephyr](https://docs.zephyrproject.org/apidoc/latest/att_8h_source.html) and allow the application to send the desired errors. 

Not being able to receive or send such error codes seems an omission in the Bluetooth implementation.

weliem, 2022-08-24:
@Vudentz 

As far as I can see, Bluez already does some basic checks before it even calls my application over the DBus. So whenever my application receives the WriteValue request, the PDU/opcode/permissions have all been checked already. The only thing left for my application is to validate the byte array and to accept/reject it. Therefore, I'd say that 'by definition' the error I return is conceptually an application error! The Bluetooth specification has a range of codes for that defined so I think it would be appropriate that my application can return a value in the defined range. I don't see how that could harm the stability of the Bluez stack....or create some kind of vulnerability.

Vudentz, 2022-08-25:
> @Vudentz
>
> As far as I can see, Bluez already does some basic checks before it even calls my application over the DBus. So whenever my application receives the WriteValue request, the [...]

I guess you are saying that we can limit the errors code that application can return (0x80-0x9F), that said we won't be able to indicate back to the application if it happens to misbehave when it set an error outside of that range.

Vudentz, 2022-08-25:
Looks like we already do encode the application errors into the D-Bus error code for the client:

https://github.com/bluez/bluez/blob/master/src/gatt-client.c#L292

So I guess we can do something similar to the server:

```
diff --git a/src/gatt-database.c b/src/gatt-database.c
index db8432c6bf77..89a3dc47560b 100644
--- a/src/gatt-database.c
+++ b/src/gatt-database.c
@@ -2192,27 +2192,43 @@ static bool parse_handle(GDBusProxy *proxy, uint16_t *handle)
        return true;
 }
 
-static uint8_t dbus_error_to_att_ecode(const char *error_name, uint8_t perm_err)
+static uint8_t dbus_error_to_att_ecode(const char *name, const char *msg,
+                                      uint8_t perm_err)
 {
-       if (strcmp(error_name, ERROR_INTERFACE ".Failed") == 0)
-               return 0x80;  /* For now return this "application error" */
+       if (strcmp(name, ERROR_INTERFACE ".Failed") == 0) {
+               char *endptr = NULL;
+               uint32_t ecode;
 
-       if (strcmp(error_name, ERROR_INTERFACE ".NotSupported") == 0)
+               ecode = strtol(msg, &endptr, 0);
+
+               /* If message doesn't set an error code just use 0x80 */
+               if (!endptr || *endptr != '\0')
+                       return 0x80;
+
+               if (ecode < 0x80 || ecode > 0x9f) {
+                       error("Invalid error code: %s", msg);
+                       return BT_ATT_ERROR_UNLIKELY;
+               }
+
+               return ecode;
+       }
+
+       if (strcmp(name, ERROR_INTERFACE ".NotSupported") == 0)
                return BT_ATT_ERROR_REQUEST_NOT_SUPPORTED;
 
-       if (strcmp(error_name, ERROR_INTERFACE ".NotAuthorized") == 0)
+       if (strcmp(name, ERROR_INTERFACE ".NotAuthorized") == 0)
                return BT_ATT_ERROR_AUTHORIZATION;
 
-       if (strcmp(error_name, ERROR_INTERFACE ".InvalidValueLength") == 0)
+       if (strcmp(name, ERROR_INTERFACE ".InvalidValueLength") == 0)
                return BT_ATT_ERROR_INVALID_ATTRIBUTE_VALUE_LEN;
 
-       if (strcmp(error_name, ERROR_INTERFACE ".InvalidOffset") == 0)
+       if (strcmp(name, ERROR_INTERFACE ".InvalidOffset") == 0)
                return BT_ATT_ERROR_INVALID_OFFSET;
 
-       if (strcmp(error_name, ERROR_INTERFACE ".InProgress") == 0)
+       if (strcmp(name, ERROR_INTERFACE ".InProgress") == 0)
                return BT_ERROR_ALREADY_IN_PROGRESS;
 
-       if (strcmp(error_name, ERROR_INTERFACE ".NotPermitted") == 0)
+       if (strcmp(name, ERROR_INTERFACE ".NotPermitted") == 0)
                return perm_err;
 
        return BT_ATT_ERROR_UNLIKELY;
@@ -2236,7 +2252,7 @@ static void read_reply_cb(DBusMessage *message, void *user_data)
 
        if (dbus_set_error_from_message(&err, message) == TRUE) {
                DBG("Failed to read value: %s: %s", err.name, err.message);
-               ecode = dbus_error_to_att_ecode(err.name,
+               ecode = dbus_error_to_att_ecode(err.name, err.message,
                                        BT_ATT_ERROR_READ_NOT_PERMITTED);
                dbus_error_free(&err);
                goto done;
@@ -2415,7 +2431,7 @@ static void write_reply_cb(DBusMessage *message, void *user_data)
 
        if (dbus_set_error_from_message(&err, message) == TRUE) {
                DBG("Failed to write value: %s: %s", err.name, err.message);
-               ecode = dbus_error_to_att_ecode(err.name,
+               ecode = dbus_error_to_att_ecode(err.name, err.message,
                                        BT_ATT_ERROR_WRITE_NOT_PERMITTED);
                dbus_error_free(&err);
                goto done;
@@ -2610,7 +2626,7 @@ static void acquire_write_reply(DBusMessage *message, void *user_data)
 
                error("Failed to acquire write: %s\n", err.name);
 
-               ecode = dbus_error_to_att_ecode(err.name,
+               ecode = dbus_error_to_att_ecode(err.name, err.message,
                                        BT_ATT_ERROR_WRITE_NOT_PERMITTED);
                dbus_error_free(&err);

```

weliem, 2022-08-26:
Yes! I think that works, will give that a try in the weekend

Thank you for the fix!

Vudentz, 2022-08-29:
> Yes! I think that works, will give that a try in the weekend
> 
> Thank you for the fix!

Did you manage to test these changes?

closed 2022-08-29 by BluezTestBot

weliem, 2022-09-02:
Yes, I have tested it and it works! Thanks!