bluez/bluez #382: Bug: [5.63/5.65] Android cannot read BLE characteristic descriptors or write to it

brunovollmer, 2022-08-23:
Hey,

after upgrading to 5.65 to avoid the problem described here: https://github.com/bluez/bluez/issues/317, I'm encountering a new problem where I cannot read the descriptors of the different characteristics of my BLE server on an Android phone. On IOS everything works as expected. When I try to write to a characteristic I receive 1 as an error code. 

Any ideas where to look for the problem or what could be causing this.

Vudentz, 2022-08-23:
@brunovollmer is that perhaps because the database has changed and Android is still using the old handles to read? HCI/btmon should probably tell you exactly the error.

brunovollmer, 2022-08-24:
Forgetting the connection solved the problem. Thanks for you help again!

closed 2022-08-24 by brunovollmer