bluez/bluez #383: Bug: [5.65] Laptop bluetooth dongle power off with error in bluetoothd.service

Guztaver, 2022-08-23:
When I try to power on the Bluetooth dongle with bluetoothctl, after 1ms, they just power off, and the only error I get is in `bluetooth.service`, with systemd.
Randomly, they just work, and when I reboot, they stop to work again.

The `bluetooth.service` log:
```
Aug 23 13:42:18 death-star systemd[1]: bluetooth.service: Deactivated successfully.
Aug 23 13:42:18 death-star systemd[1]: Stopped Bluetooth service.
Aug 23 13:42:18 death-star systemd[6691]: ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 755 ConfigurationDirectoryMode: 555)
Aug 23 13:42:18 death-star systemd[1]: Starting Bluetooth service...
Aug 23 13:42:18 death-star bluetoothd[6691]: Bluetooth daemon 5.65
Aug 23 13:42:18 death-star systemd[1]: Started Bluetooth service.
Aug 23 13:42:18 death-star bluetoothd[6691]: Starting SDP server
Aug 23 13:42:18 death-star bluetoothd[6691]: Bluetooth management interface 1.22 initialized
Aug 23 13:42:18 death-star bluetoothd[6691]: Battery Provider Manager created
Aug 23 13:42:19 death-star bluetoothd[6691]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power dow
```

Some additional information:
Kernel: `5.19.3-arch1-1`
bluetoothctl: `5.65`

Vudentz, 2022-08-23:
@Guztaver are you talking about:

`Aug 23 13:42:18 death-star systemd[6691]: ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 755 ConfigurationDirectoryMode: 555)`

or 

`Aug 23 13:42:19 death-star bluetoothd[6691]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power dow`

The latter you can probably disregard, as for the former I thing it is because the bluetooth.service has changed updating the ConfigurationDirectory when previously we didn't have that, anyway please add the HCI trace (using btmon).

Guztaver, 2022-08-23:
> @Guztaver are you talking about:
>
> `Aug 23 13:42:18 death-star systemd[6691]: ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 755 [...]

I'm talking about the `btd_adv_monitor_power_down`, and how i can get the HCI trace?

Vudentz, 2022-08-23:
> I'm talking about the `btd_adv_monitor_power_down`, and how i can get the HCI trace?

btmon -w <filename>

Guztaver, 2022-08-23:
@Vudentz here is the log when i try to power on the dongle:
```log
root in ~/Downloads
➜ btmon -w hci0
Bluetooth monitor ver 5.65
= Note: Linux version 5.19.3-arch1-1 (x86_64)                                                       0.458696
= Note: Bluetooth subsystem version 2.22                                                            0.458700
= New Index: 87:37:64:88:04:03 (Primary,USB,hci0)                                            [hci0] 0.458701
@ MGMT Open: bluetoothd (privileged) version 1.22                                          {0x0001} 0.458703
@ MGMT Command: Set Powered (0x0005) plen 1                                         {0x0001} [hci0] 7.705161
        Powered: Enabled (0x01)
= Open Index: 87:37:64:88:04:03                                                              [hci0] 7.831623
< HCI Command: Reset (0x03|0x0003) plen 0                                                 #1 [hci0] 7.831642
> HCI Event: Command Complete (0x0e) plen 4                                               #2 [hci0] 7.834481
      Reset (0x03|0x0003) ncmd 1
        Status: Success (0x00)
< HCI Command: Read Local Supported Features (0x04|0x0003) plen 0                         #3 [hci0] 7.834551
> HCI Event: Command Complete (0x0e) plen 12                                              #4 [hci0] 7.836356
      Read Local Supported Features (0x04|0x0003) ncmd 1
        Status: Success (0x00)
        Features: 0xff 0xff 0x8d 0xfe 0x9b 0xf9 0x00 0x80
          3 slot packets
          5 slot packets
          Encryption
          Slot offset
          Timing accuracy
          Role switch
          Hold mode
          Sniff mode
          Park state
          Power control requests
          Channel quality driven data rate (CQDDR)
          SCO link
          HV2 packets
          HV3 packets
          u-law log synchronous data
          A-law log synchronous data
          CVSD synchronous data
          Power control
          Transparent synchronous data
          Broadcast Encryption
          Enhanced Data Rate ACL 2 Mbps mode
          Enhanced Data Rate ACL 3 Mbps mode
          Enhanced inquiry scan
          Interlaced inquiry scan
          Interlaced page scan
          RSSI with inquiry results
          Extended SCO link (EV3 packets)
          EV4 packets
          EV5 packets
          AFH capable peripheral
          AFH classification peripheral
          3-slot Enhanced Data Rate ACL packets
          5-slot Enhanced Data Rate ACL packets
          AFH capable central
          AFH classification central
          Enhanced Data Rate eSCO 2 Mbps mode
          Enhanced Data Rate eSCO 3 Mbps mode
          3-slot Enhanced Data Rate eSCO packets
          Extended features
< HCI Command: Read Local Version Information (0x04|0x0001) plen 0                        #5 [hci0] 7.836435
> HCI Event: Command Complete (0x0e) plen 12                                              #6 [hci0] 7.837359
      Read Local Version Information (0x04|0x0001) ncmd 1
        Status: Success (0x00)
        HCI version: Bluetooth 2.0 (0x03) - Revision 16384 (0x4000)
        LMP version: Bluetooth 2.0 (0x03) - Subversion 17166 (0x430e)
        Manufacturer: Broadcom Corporation (15)
          Firmware: 002.003.014
          Build: 0000
< HCI Command: Read BD ADDR (0x04|0x0009) plen 0                                          #7 [hci0] 7.837454
[... 280 lines cut ...]
          Discoverable
          Bondable
          BR/EDR
= bluetoothd: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_ma..   7.881325
@ MGMT Command: Set Powered (0x0005) plen 1                                         {0x0001} [hci0] 7.881361
        Powered: Disabled (0x00)
< HCI Command: Write Scan Enable (0x03|0x001a) plen 1                                    #47 [hci0] 7.881440
        Scan enable: No Scans (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                              #48 [hci0] 7.883354
      Write Scan Enable (0x03|0x001a) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 7                                      {0x0001} [hci0] 7.883421
      Set Powered (0x0005) plen 4
        Status: Success (0x00)
        Current settings: 0x0000009a
          Connectable
          Discoverable
          Bondable
          BR/EDR
@ MGMT Event: Class Of Device Changed (0x0007) plen 3                               {0x0001} [hci0] 7.883428
        Class: 0x000000
          Major class: Miscellaneous
          Minor class: 0x00
= Close Index: 87:37:64:88:04:03                                                             [hci0] 7.883438
```

Guztaver, 2022-08-23:
i forgot to upload the file, sorry!
[hci0.log](https://github.com/bluez/bluez/files/9410316/hci0.log)

Vudentz, 2022-08-23:
```
@ MGMT Command: Set Powered (0x0005) plen 1                                         {0x0001} [hci0] 7.881361
        Powered: Disabled (0x00)
```

This is the default setting or does main.conf has AutoEnable=false?

Guztaver, 2022-08-23:
> ```
> @ MGMT Command: Set Powered (0x0005) plen 1                                         {0x0001} [hci0] 7.881361
>         Powered: Disabled (0x00)
> ```
> 
> This is the default setting or does main.conf has AutoEnable=false?

Has `AutoEnable=false`, but it didn't work with them in true either.

Guztaver, 2022-08-23:
So, I discovered one thing: when I connect with another power plug, they just works, so the real problem isn't driver, is the power. Thanks @Vudentz for the attention!

closed 2022-08-23 by Guztaver