bluez/bluez #390: MIDI interface not exposed

frafra, 2022-09-09:
I tried to connect to a digital piano which has bluetooth MIDI, but such interface is not exposed,

```
[bluetooth]# connect E2:2A:FC:62:97:8A 
Attempting to connect to E2:2A:FC:62:97:8A
[CHG] Device E2:2A:FC:62:97:8A Connected: yes
Connection successful
[NEW] Primary Service (Handle 0x0000)
	/org/bluez/hci0/dev_E2_2A_FC_62_97_8A/service0008
	00001801-0000-1000-8000-00805f9b34fb
	Generic Attribute Profile
[NEW] Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_E2_2A_FC_62_97_8A/service0008/char0009
	00002a05-0000-1000-8000-00805f9b34fb
	Service Changed
[NEW] Descriptor (Handle 0x0000)
	/org/bluez/hci0/dev_E2_2A_FC_62_97_8A/service0008/char0009/desc000b
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[NEW] Primary Service (Handle 0x0000)
	/org/bluez/hci0/dev_E2_2A_FC_62_97_8A/service000c
	0000180a-0000-1000-8000-00805f9b34fb
	Device Information
[NEW] Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_E2_2A_FC_62_97_8A/service000c/char000d
	00002a29-0000-1000-8000-00805f9b34fb
	Manufacturer Name String
[NEW] Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_E2_2A_FC_62_97_8A/service000c/char000f
	00002a24-0000-1000-8000-00805f9b34fb
	Model Number String
[NEW] Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_E2_2A_FC_62_97_8A/service000c/char0011
	00002a25-0000-1000-8000-00805f9b34fb
	Serial Number String
[NEW] Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_E2_2A_FC_62_97_8A/service000c/char0013
	00002a27-0000-1000-8000-00805f9b34fb
	Hardware Revision String
[NEW] Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_E2_2A_FC_62_97_8A/service000c/char0015
	00002a26-0000-1000-8000-00805f9b34fb
	Firmware Revision String
[NEW] Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_E2_2A_FC_62_97_8A/service000c/char0017
	00002a28-0000-1000-8000-00805f9b34fb
	Software Revision String
[CHG] Device E2:2A:FC:62:97:8A UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device E2:2A:FC:62:97:8A UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device E2:2A:FC:62:97:8A UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Device E2:2A:FC:62:97:8A UUIDs: 03b80e5a-ede8-4b33-a751-6ce34ec4c700
[CHG] Device E2:2A:FC:62:97:8A ServicesResolved: yes
```

```
[ES920]# info
Device E2:2A:FC:62:97:8A (random)
	Name: ES920
	Alias: ES920
	Paired: yes
	Bonded: no
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (03b80e5a-ede8-4b33-a751-6ce34ec4c700)
```

```
$ aplaymidi -l
 Port    Client name                      Port name
 14:0    Midi Through                     Midi Through Port-0
$ arecordmidi -l
 Port    Client name                      Port name
 14:0    Midi Through                     Midi Through Port-0
```

```
$ lsusb | grep -i bluetooth
Bus 001 Device 004: ID 0a5c:21e6 Broadcom Corp. BCM20702 Bluetooth 4.0 [ThinkPad]
```

Device documentation declares:
> Bluetooth* Bluetooth (Ver. 4.1; GATT compatible)
> Bluetooth Low Energy MIDI Specification compliant, Bluetooth Audio

Bluez version 5.56, compiled using `--enable-midi`.

Vudentz, 2022-09-12:
@frafra could you please attach the daemon output and the btmon trace?

mwprado, 2023-11-21:
> @frafra could you please attach the daemon output and the btmon trace?
It's similar to Go:Keys issue at
https://github.com/bluez/bluez/issues/604

mwprado, 2023-12-07:
> I tried to connect to a digital piano which has bluetooth MIDI, but such interface is not exposed,
>
> ```
> [bluetooth]# connect E2:2A:FC:62:97:8A
> Attempting to connect to E2:2A:FC:62:97:8A
> [CHG] [...]

Have you tried disabling SELINUX?

closed 2025-08-17 by github-actions[bot]