bluez/bluez #391: Just a question: Bluetooth LE Audio (auracast)

StuartIanNaylor, 2022-09-11:
Anyone know the state of play with Bluetooth LE Audio specifically auracast?
WiP or roadmap or not on the horizon anyone know?

Thnx

Vudentz, 2022-09-13:
@StuartIanNaylor Upcoming Linux 6.0 will start supporting ISO sockets which implements Isochronous Channels, the userspace already contains part of the stack, (BAP, PACS, ASCS), we will probably be releasing it along side with Linux 6.0.

StuartIanNaylor, 2022-09-16:
Thnx for the info, exciting times. Ps I will close but I have been trying to work out if there is any form of time sync like snapcast or shairport-sync or broadcast in the way it works of in range devices sync is inherent.
I have been wondering about multi speaker than multi room where an active speaker is set to a channel, but thinking auracast with a sea of closed proprietary RTP or need of special radio's could be huge and a great opensource addition.
So thnx for the work as well and will be eagerly awaiting Linux 6.0 to be released.

closed 2022-09-16 by StuartIanNaylor

Septias, 2023-06-28:
What is the current state?