bluez/bluez #393: Debian bookworm: any bluetooth device cannot be connected before systemctl restart bluetooth

okias, 2022-09-14:
HW: Dell XPS 9310
SW: Debian unstable / bookworm.
/cc @VojtechMyslivec
Kernel start:
```
[    9.789627] Bluetooth: Core ver 2.22
[    9.789651] NET: Registered PF_BLUETOOTH protocol family
[    9.789652] Bluetooth: HCI device and connection manager initialized
[    9.789657] Bluetooth: HCI socket layer initialized
[    9.789660] Bluetooth: L2CAP socket layer initialized
[    9.789664] Bluetooth: SCO socket layer initialized
[    9.947263] Bluetooth: hci0: Bootloader revision 0.4 build 0 week 30 2018
[    9.948179] Bluetooth: hci0: Device revision is 2
[    9.948180] Bluetooth: hci0: Secure boot is enabled
[    9.948181] Bluetooth: hci0: OTP lock is enabled
[    9.948181] Bluetooth: hci0: API lock is enabled
[    9.948182] Bluetooth: hci0: Debug lock is disabled
[    9.948182] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[    9.952686] bluetooth hci0: firmware: direct-loading firmware intel/ibt-19-0-4.sfi
[    9.952694] Bluetooth: hci0: Found device firmware: intel/ibt-19-0-4.sfi
[    9.952726] Bluetooth: hci0: Boot Address: 0x24800
[    9.952727] Bluetooth: hci0: Firmware Version: 191-21.21
[   10.501848] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   10.501851] Bluetooth: BNEP filters: protocol multicast
[   10.501854] Bluetooth: BNEP socket layer initialized
[   11.772764] Bluetooth: hci0: Waiting for firmware download to complete
[   11.773139] Bluetooth: hci0: Firmware loaded in 1777775 usecs
[   11.773171] Bluetooth: hci0: Waiting for device to boot
[   11.788182] Bluetooth: hci0: Device booted in 14672 usecs
[   11.789891] bluetooth hci0: firmware: direct-loading firmware intel/ibt-19-0-4.ddc
[   11.789897] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-19-0-4.ddc
[   11.792167] Bluetooth: hci0: Applying Intel DDC parameters completed
[   11.793167] Bluetooth: hci0: Firmware revision 0.0 build 191 week 21 2021
[   11.881181] Bluetooth: RFCOMM TTY layer initialized
[   11.881186] Bluetooth: RFCOMM socket layer initialized
[   11.881190] Bluetooth: RFCOMM ver 1.11
```

Init:
```
Sep 14 21:52:31 newone systemd[1]: Starting Load/Save RF Kill Switch Status...
Sep 14 21:52:31 newone systemd[1]: Started Load/Save RF Kill Switch Status.
Sep 14 21:52:36 newone systemd[1]: systemd-rfkill.service: Deactivated successfully.
Sep 14 21:52:44 newone systemd[1624]: Started Application launched by gnome-shell.
Sep 14 21:52:44 newone gnome-control-c[88933]: BluetoothHardwareAirplaneMode: 0
Sep 14 21:52:44 newone dbus-daemon[1650]: [session uid=1000 pid=1650] Activating via systemd: service name='org.bluez.obex' unit='obex.service' requested by ':1.633' (uid=1000 pid=88933 comm="gnome-control-center bluetooth")
Sep 14 21:52:44 newone systemd[1624]: Starting Bluetooth OBEX service...
Sep 14 21:52:44 newone obexd[88959]: OBEX daemon 5.65
Sep 14 21:52:44 newone dbus-daemon[1650]: [session uid=1000 pid=1650] Successfully activated service 'org.bluez.obex'
Sep 14 21:52:44 newone systemd[1624]: Started Bluetooth OBEX service.
```

Connection on in GNOME panel
```
Sep 14 21:53:40 newone systemd[1624]: Reached target Bluetooth.
Sep 14 21:53:40 newone kernel: Bluetooth: hci0: link tx timeout
Sep 14 21:53:40 newone kernel: Bluetooth: hci0: Opcode 0x 401 failed: -16
```
[btmon_connect_headphones.txt](https://github.com/bluez/bluez/files/9569648/btmon_connect_headphones.txt)

Vudentz, 2022-09-14:
@okias what kernel version? Would be great if you can add the HCI traces (use btmon to collect them).

okias, 2022-09-14:
Kernel: `5.19.0-1-amd64`
I'll try to get traces tomorrow! :)

closed 2025-08-17 by github-actions[bot]