bluez/bluez #394: i have a crash（_dbus_object_tree_dispatch_and_unlock ） ,Has anyone met？

FreshLuoBoGan, 2022-09-15:
[image]

Vudentz, 2022-09-15:
@FreshLuoBoGan not really, what is bdmanager though?

FreshLuoBoGan, 2022-09-16:
the bdmanager is my app，it integrate bluez-client;

FreshLuoBoGan, 2022-09-16:
> @FreshLuoBoGan not really, what is bdmanager though?

and the object.c is from bluez-5.64/gdbus/object.c

Vudentz, 2022-09-16:
> > @FreshLuoBoGan not really, what is bdmanager though?
> 
> and the object.c is from bluez-5.64/gdbus/object.c

You are no supposed to link with it though, it is GPL and not thread safe which is probably why you are having problems.

FreshLuoBoGan, 2022-09-19:
> > > @FreshLuoBoGan not really, what is bdmanager though?
> > 
> > 
> > and the object.c is from bluez-5.64/gdbus/object.c
> 
> You are no supposed to link with it though, it is GPL and not thread safe which is probably why you are having problems.

ok,thank you ,i try to solve it

FreshLuoBoGan, 2022-09-20:
i resolved it，my app is multithread,so i add mutex in pending of object.c，thank you for your remind @Vudentz

closed 2022-09-22 by Vudentz