bluez/bluez #395: bnep: Can't add bnep0 to the bridge pan0: Invalid argument(22)

pierre-pretorius, 2022-09-15:
I've tried different versions of bluez (5.50, 5.65) with different kernels (5.9, 5.10, 5.15, 5.16) and distributions (buster, bullseye, yammy) with the Orange PI 3 LTS.  I always get the same error once I connect with a mobile phone to the Orange PI Bluetooth.  Is this a bluez error or rather a bluetooth firmware issue?

```
Sep 15 17:10:32 orangepi3-lts bluetoothd[1037]: Bluetooth management interface 1.21 initialized
Sep 15 17:15:42 orangepi3-lts bluetoothd[1037]: profiles/network/bnep.c:bnep_add_to_bridge() bnep: Can't add bnep0 to the bridge pan0: Invalid argument(22)
Sep 15 17:15:42 orangepi3-lts bluetoothd[1037]: profiles/network/server.c:bnep_setup() BNEP server cannot be added
```
https://github.com/bluez/bluez/blob/401643e5c5528a6d0c372f82b1059fc0e02aed8e/profiles/network/bnep.c#L450

### Setup from clean image

```
nano /etc/systemd/network/pan0.netdev
```
Paste content:
```
[NetDev]
Name=pan0
Kind=bridge
```

```
nano /etc/systemd/network/pan0.network
```
Paste content:
```
[Match]
Name=pan0

[Network]
Address=10.10.0.5/24
DHCPServer=true

[DHCPServer]
PoolOffset=100
PoolSize=100
EmitDNS=yes
DNS=9.9.9.9

[Link]
RequiredForOnline=false
```

```
apt-get install -y bluez bluez-tools nginx
systemctl enable systemd-networkd
systemctl enable nginx
reboot
```

### Reproduce the error

In one terminal, start bt-agent:
```
/usr/bin/yes | /usr/bin/bt-agent -c NoInputNoOutput
```

In another terminal, start bt-network:
```
/usr/bin/bt-network -s nap pan0
```

In a 3rd terminal, set properties and tail log:
```
/usr/bin/bt-adapter --set Discoverable 1
/usr/bin/bt-adapter --set Pairable 1
journalctl -u bluetooth -f
```

Now connect to Bluetooth with a mobile phone.  It prints this error on the orange PI:
```
Sep 15 17:10:31 orangepi3-lts systemd[1]: Starting Bluetooth service...
Sep 15 17:10:31 orangepi3-lts bluetoothd[1037]: Bluetooth daemon 5.55
Sep 15 17:10:31 orangepi3-lts bluetoothd[1037]: Starting SDP server
Sep 15 17:10:31 orangepi3-lts systemd[1]: Started Bluetooth service.
Sep 15 17:10:32 orangepi3-lts bluetoothd[1037]: Bluetooth management interface 1.21 initialized
Sep 15 17:15:42 orangepi3-lts bluetoothd[1037]: profiles/network/bnep.c:bnep_add_to_bridge() bnep: Can't add bnep0 to the bridge pan0: Invalid argument(22)
Sep 15 17:15:42 orangepi3-lts bluetoothd[1037]: profiles/network/server.c:bnep_setup() BNEP server cannot be added
```
It breaks here, which seems like some sort of network adapter index error?
https://github.com/bluez/bluez/blob/401643e5c5528a6d0c372f82b1059fc0e02aed8e/profiles/network/bnep.c#L450

Expected result: the mobile phone can open http://10.10.0.5 and see the nginx page.

Vudentz, 2022-09-15:
@pierre-pretorius is the bridge up? You can try adding bnep0 manually to it to see what effect it has.

pierre-pretorius, 2022-09-15:
Thank you for the quick response.  The pan0 interface is up, but there is no bnep0.
```
 pan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN group default qlen 1000
    link/ether 1e:c1:82:69:91:e0 brd ff:ff:ff:ff:ff:ff
    inet 10.10.0.5/24 brd 10.10.0.255 scope global pan0
       valid_lft forever preferred_lft forever
    inet6 fe80::1cc1:82ff:fe69:91e0/64 scope link 
       valid_lft forever preferred_lft forever
```

On a Raspberry PI where it works, there is bnep:
```
pan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether dc:a6:32:4a:a3:93 brd ff:ff:ff:ff:ff:ff
    inet 10.10.0.5/24 brd 10.10.0.255 scope global pan0
       valid_lft forever preferred_lft forever
    inet6 fe80::d4ef:bbff:fe85:d0ba/64 scope link 
       valid_lft forever preferred_lft forever
bnep0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master pan0 state UNKNOWN group default qlen 1000
    link/ether dc:a6:32:4a:a3:93 brd ff:ff:ff:ff:ff:ff
    inet 169.254.17.154/16 brd 169.254.255.255 scope global noprefixroute bnep0
       valid_lft forever preferred_lft forever
    inet6 fe80::9129:cb08:a5c9:6bc/64 scope link 
       valid_lft forever preferred_lft forever
    inet6 fe80::dea6:32ff:fe4a:a393/64 scope link 
       valid_lft forever preferred_lft forever
```
Could you give a rough indication of how to start the bridge (which I assume is bnep) manually?

Vudentz, 2022-09-15:
@pierre-pretorius hmm perhaps something in the kernel is not quite right, but we check the error on BNEPCONNADD so perhaps something is removing it after that, well we do remove it bnep_conndel if it fails to be added to the bridge, so I don't quite follow what could be preventing it to work, does the Raspberry PI set it up differently?

pierre-pretorius, 2022-09-15:
The Raspberry PI runs buster image from Raspberry PI OS:
```
lsb_release -a
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

uname -a
Linux raspberry-pi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

bluetoothctl --version
bluetoothctl: 5.50
```
The Orange PI 3 LTS runs the buster image from Orange PI:
```
lsb_release -a
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

uname -a
Linux orangepi3-lts 5.10.75-sun50iw6 #2.2.2 SMP Wed Apr 20 09:45:07 CST 2022 aarch64 GNU/Linux

bluetoothctl --version
bluetoothctl: 5.50
```
The images are built by two different companies so can differ of course, but package versions are mostly the same.

Vudentz, 2022-09-16:
@pierre-pretorius I mean if there are any different on how they create the pan0 bridge, or you have the exact same setup for pan0 in both cases? Not sure if there are things like namespaces that could interfere with bridge setup.

konkhra, 2022-11-22:
I am facing the same issue with a Nano Pi Neo Air. I think it is a Bluetooth firmware issue. 
pan0 is not automatically  created like on RPi. No issues with RPi4.

closed 2025-08-17 by github-actions[bot]