bluez/bluez #396: Double free/UAF in queue_pop_head

whiteHat001, 2022-09-16:
I did a detailed analysis of 363 
https://github.com/bluez/bluez/issues/363

```c
bool bt_gatt_client_unregister_notify(struct bt_gatt_client *client,
							unsigned int id)
{
	struct notify_data *notify_data;

	if (!client || !id)
		return false;

	notify_data = queue_remove_if(client->notify_list, match_notify_data_id,
							UINT_TO_PTR(id));
	if (!notify_data)
		return false;

	/* Remove data if it has been queued */
	queue_remove(notify_data->chrc->reg_notify_queue, notify_data);    // [0]

	/* Reset callbacks */
	notify_data->callback = NULL;
	notify_data->notify = NULL;

	complete_unregister_notify(notify_data);  //[2]
	return true;
}
```

```c
bool queue_remove(struct queue *queue, void *data)
{
	struct queue_entry *entry, *prev;

	if (!queue)
		return false;

	for (entry = queue->head, prev = NULL; entry;
					prev = entry, entry = entry->next) {
		if (entry->data != data)
			continue;

		if (prev)
			prev->next = entry->next;
		else
			queue->head = entry->next;

		if (!entry->next)
			queue->tail = prev;

		free(entry);    //[1]
		queue->entries--;

		return true;
	}

	return false;
}

```

```c
static void complete_unregister_notify(void *data)
{
	struct notify_data *notify_data = data;

	/*
	 * If a procedure to enable the CCC is still pending, then cancel it and
	 * return.
	 */
	if (notify_data->att_id) {
		bt_att_cancel(notify_data->client->att, notify_data->att_id);
		notify_data->att_id = 0;
		goto done;
	}

	if (__sync_sub_and_fetch(&notify_data->chrc->notify_count, 1) ||
						!notify_data->chrc->ccc_handle)
		goto done;

	notify_data_write_ccc(notify_data, false, disable_ccc_callback); //[3]

done:
	notify_data_unref(notify_data);
}
```

```c
static void disable_ccc_callback(uint8_t opcode, const void *pdu,
					uint16_t length, void *user_data)
{
	struct notify_data *notify_data = user_data;
	struct notify_data *next_data;

	assert(notify_data->chrc->ccc_write_id);

	notify_data->chrc->ccc_write_id = 0;

	/* This is a best effort procedure, so ignore errors and process any
	 * queued requests.
	 */
	while (1) {
		next_data = queue_pop_head(notify_data->chrc->reg_notify_queue); //[4]
		if (!next_data || notify_data_write_ccc(next_data, true,
							enable_ccc_callback))
			return;
	}
}
```

```c
void *queue_pop_head(struct queue *queue)
{
	struct queue_entry *entry;
	void *data;

	if (!queue || !queue->head)
		return NULL;

	entry = queue->head;

	if (!queue->head->next) {
		queue->head = NULL;
		queue->tail = NULL;
	} else
		queue->head = queue->head->next;

	data = entry->data;

	free(entry);  //[5]
	queue->entries--;

	return data;
}
```
The bt_gatt_client_unregister_notify function will call [0]queue_remove, queue_remove will release the entry at [1], then enter complete_unregister_notify[2], and then call the disable_ccc_callback callback function [3], the callback function will call queue_pop_head[4], and finally in the callback function It will call queue_pop_head and free entry[2] again, resulting in double free

Vudentz, 2022-09-16:
@whiteHat001 perhaps the data is added twice then since queue_pop_head shouldn't be able to find the same data as it has been remove by queue_remove, we can perhaps check if that has been added twice and then print warning then we can debug why that is the case here since I wasn't able to reproduce it perhaps that has to do with something specific behavior of the device.

Vudentz, 2022-09-22:
@whiteHat001 have it running with valgrind since it should generate a more readable backtrace.

whiteHat001, 2022-09-23:
@Vudentz  I also analyzed it according to the backtrace of 363

closed 2025-08-17 by github-actions[bot]