bluez/bluez #397: Double free in setup_cb_free

whiteHat001, 2022-09-16:
Detailed analysis of issue 327

```c
static void endpoint_setconf_cb(struct a2dp_setup *setup, gboolean ret)
{
	if (ret == FALSE) {
		setup->err = g_new(struct avdtp_error, 1);
		avdtp_error_init(setup->err, AVDTP_MEDIA_CODEC,
					AVDTP_UNSUPPORTED_CONFIGURATION);
	}

	auto_config(setup);  //[0]
	setup_unref(setup); //[2]
}
```

```c
static gboolean auto_config(gpointer data)
{
	struct a2dp_setup *setup = data;
	struct btd_device *dev = NULL;
	struct btd_service *service;

	/* Check if configuration was aborted */
	if (setup->sep->stream == NULL)
		return FALSE;

	if (setup->err != NULL)
		goto done;

	dev = avdtp_get_device(setup->session);

	avdtp_stream_add_cb(setup->session, setup->stream,
				stream_state_changed, setup->sep);

	if (setup->sep->type == AVDTP_SEP_TYPE_SOURCE) {
		service = btd_device_get_service(dev, A2DP_SINK_UUID);
		sink_new_stream(service, setup->session, setup->stream);
	} else {
		service = btd_device_get_service(dev, A2DP_SOURCE_UUID);
		source_new_stream(service, setup->session, setup->stream);
	}

done:
	if (setup->setconf_cb)
		setup->setconf_cb(setup->session, setup->stream, setup->err);

	finalize_config(setup);

	if (setup->err) {
		g_free(setup->err);
		setup->err = NULL;
	}

	setup_unref(setup);  //[1]

	return FALSE;
}
```

```c
static void setup_unref(struct a2dp_setup *setup)
{
	setup->ref--;

	DBG("%p: ref=%d", setup, setup->ref);

	if (setup->ref > 0)
		return;

	setup_free(setup);
}
```

auto_config will call auto_config[0],and then call setup_unref[1],after this function return,next line call setup_unref[2] again.if in [1] setup->ref =0,setup_free will be called,and in [2] setup->ref =-1,setup_free call again.
Finally result double free.There are many paths to call double free.

fix

if (setup->ref != 0)
		return;

Vudentz, 2022-09-16:
@whiteHat001 could you please add as a comment to the issue itself so it easier to people to find about it, it indeed sounds there is unbalanced refcount problem, but it would be better if we got the HCI traces (btmon) and perhaps run bluetoothd under valgrind so it can generate nice backtraces when this happens.

whiteHat001, 2022-09-17:
@Vudentz I will

closed 2022-09-22 by Vudentz