bluez/bluez #398: No battery indicator for Sony WH-1000XM3

Aurnytoraink, 2022-09-19:
Android and iOS naturally shows the remaining battery of my Sony headset, but this is not the case on Linux.

Here are the information that `bluetoothctl info` gave me:

```bash
Name: LE_WH-1000XM3
Alias: LE_WH-1000XM3
Class: 0x00240404
Icon: audio-headset
Paired: yes
Bonded: yes
Trusted: yes
Blocked: no
Connected: yes
LegacyPairing: no
UUID: Vendor specific           (00000000-deca-fade-deca-deafdecacaff)
UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
UUID: Amazon.com Services, Inc. (0000fe03-0000-1000-8000-00805f9b34fb)
UUID: Vendor specific           (5b833e05-6bc7-4802-8e9a-723ceca4bd8f)
UUID: Vendor specific           (5b833e06-6bc7-4802-8e9a-723ceca4bd8f)
UUID: Vendor specific           (69a7f243-e52f-4443-a7f9-cb4d053c74d6)
UUID: Vendor specific           (7b265b0e-2232-4d45-bef4-bb8ae62f813d)
UUID: Vendor specific           (81c2e72a-0591-443e-a1ff-05f988593351)
UUID: Vendor specific           (91c10d9c-aaef-42bd-b6d6-8a648c19213d)
UUID: Vendor specific           (931c7e8a-540f-4686-b798-e8df0a2ad9f7)
UUID: Vendor specific           (96cc203e-5068-46ad-b32d-e316f5e069ba)
UUID: Vendor specific           (b9b213ce-eeab-49e4-8fd9-aa478ed1b26b)
UUID: Vendor specific           (f8d1fbe4-7966-4334-8024-ff96c9330e15)
UUID: Vendor specific           (fe59bfa8-7fe3-4a05-9d94-99fadc69faff)
Modalias: usb:v054Cp0CD3d0452

```

Vudentz, 2022-09-19:
@Aurnytoraink the battery information normally fetched via AT commands which is not under bluetoothd control, BlueZ does offer some means for upper layer to add this information though:

https://github.com/bluez/bluez/blob/master/doc/battery-api.txt

Afaik the likes of pipewire do already support BatteryProvider.

Aurnytoraink, 2022-09-20:
Sorry, but I didn't understand your answer. If I get it correct, I should report the issue to PipeWire ?

Because I feel like the problem is coming from BlueZ, since when I'm doing `bluetoothctl info` on my Logitech mouse, it reports his remaining battery level

Vudentz, 2022-09-20:
> Sorry, but I didn't understand your answer. If I get it correct, I should report the issue to PipeWire ?
> 
> Because I feel like the problem is coming from BlueZ, since when I'm doing `bluetoothctl info` on my Logitech mouse, it reports his remaining battery level

There is different source of battery information, BlueZ will collect it automatically if that comes from a profile under its control, but in case of headsets that normally is implemented as part of HFP which is not under BlueZ's Control thus why BlueZ offer an interface for the upper profiles to inject the battery information via BatteryProvider.

endrift, 2023-01-28:
Most devices that report battery over Bluetooth LE use the Battery Service (BAS), which corresponds to UUID 0000180f-0000-1000-8000-00805f9b34fb. You'll notice that services is absent from this list. What @Vudentz is saying is that headphones use a different source to relay this information that bluez doesn't interface with, and something else would have to read. I don't know the details on that however.