bluez/bluez #401: I can't connect a smartphone (bt 4.2) to a PC (bt 5.3) from the PC

promeneur, 2022-09-27:
MB Asus B250M

Wi-Fi/Bluetooth card : EDUP LOVE EP-9651GS (chipset Intel AX210), Bluetooth 5.3

Samsung S7, android 8.0, Bluetooth 4

openSUSE Tumbleweed

kernel 5.19.10

Firmware date : 20220902 (date of the package "firmware all")

bluez 5.65

I don't use Wi-Fi. Just for a backup solution. I checked Wi-Fi works well.

At the PC, just disconnect the smartphone after pairing.

Wait 5 mn or more.

At the PC, connect the smartphone. The connection fails.

I did some experiments with Clearlinux and Fedora. With the two OS, I can reproduce the problem.

 
More info :

At the smartphone, I connect the PC. The connection succeeds.

At the PC, I disconnect then reconnect the smartphone. The connection succeeds.

At the PC I disconnect the smartphone.

I Wait for 5 mn or more.

At the PC I connect the smartphone. The connection fails.

There is no problem between a Bluetooth 5.0 tablet (Blackview Tab 8) and a PC equipped with Bluetooth 5.3. You can connect the smartphone from the PC.

Vudentz, 2022-09-27:
@promeneur What exactly are you trying to connect though? Are you trying A2DP (audio)? Please collect the HCI trace using btmon and bluetoothd logs.

promeneur, 2022-09-28:
At the PC, I try to connect a smartphone.

Pairing succeeds.
After pairing, the smartphone is connected.
After a while, I don't need anymore the smartphone, then at the PC I disconnect it.
One hour later or next day, I need the smartphone. Then at the PC, I connect the smartphone. The connection fails.

This problem occurs with a bt 4.2 smartphone and a bt 5.3 PC.
This problem does not occur with a bt 5.0 tablet and a bt 5.3 PC.

Forget the problem of the protocols.

promeneur, 2022-09-30:
At the PC, I connect the smartphone 

bluetoothctl connect 90:06:28:B6:B1:B6
Attempting to connect to 90:06:28:B6:B1:B6
Failed to connect: org.bluez.Error.Failed le-connection-abort-by-local

promeneur, 2022-09-30:
bluetoothctl info 90:06:28:B6:B1:B6
Device 90:06:28:B6:B1:B6 (public)
        Name: Samsung Galaxy S7 Argent
        Alias: Samsung Galaxy S7 Argent
        Class: 0x005a020c
        Icon: phone
        Paired: yes
        Bonded: yes
        Trusted: yes
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
        UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: PANU                      (00001115-0000-1000-8000-00805f9b34fb)
        UUID: NAP                       (00001116-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        UUID: SIM Access                (0000112d-0000-1000-8000-00805f9b34fb)
        UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
        UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (a23d00bc-217c-123b-9c00-fc44577136ee)
        Modalias: bluetooth:v0075p0100d0201
        Affected by Policy: no

promeneur, 2022-09-30:
bluetoothctl show
Controller F0:B6:1E:FE:FB:F4 (public)
        Name: grincheux
        Alias: grincheux
        Class: 0x007c0104
        Powered: yes
        Discoverable: no
        DiscoverableTimeout: 0x000000b4
        Pairable: yes
        UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
        UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0541
        Discovering: no
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x0c (12)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
        SupportedSecondaryChannels: 1M
        SupportedSecondaryChannels: 2M
        SupportedSecondaryChannels: Coded

promeneur, 2022-09-30:
At the PC, i connect the smartphone. the connection fails.
[dump.btsnoop.txt](https://github.com/bluez/bluez/files/9681945/dump.btsnoop.txt)

promeneur, 2022-09-30:
I forgot.

pipewire 3.58

Vudentz, 2022-09-30:
> At the PC, i connect the smartphone. the connection fails. [dump.btsnoop.txt](https://github.com/bluez/bluez/files/9681945/dump.btsnoop.txt)

Are you sure this is the correct trace? It seems it is a incoming connection which is accepted and I don't see it being disconnected either.

promeneur, 2022-09-30:
another trace
[dump.btsnoop.txt](https://github.com/bluez/bluez/files/9687419/dump.btsnoop.txt)

Vudentz, 2022-09-30:
> another trace [dump.btsnoop.txt](https://github.com/bluez/bluez/files/9687419/dump.btsnoop.txt)

There is not even a connection in there, just advertisement reports, are you running btmon -w alongside bluetoothctl when attempting to capture these traces?

promeneur, 2022-09-30:
>> are you running btmon -w alongside bluetoothctl when attempting to capture these traces?

yes

promeneur, 2022-09-30:
another try
[dump.btsnoop.txt](https://github.com/bluez/bluez/files/9687459/dump.btsnoop.txt)

promeneur, 2022-10-01:
I made an interesting experiment.
I switch the phone to "discoverable" state.
[bluetooth]# scan on
Discovery started
[CHG] Controller F0:B6:1E:FE:FB:F4 Discovering: yes
[CHG] Device 90:06:28:B6:B1:B6 RSSI: -56
[bluetooth]# scan off
Discovery stopped
[CHG] Device 90:06:28:B6:B1:B6 RSSI is nil
[CHG] Controller F0:B6:1E:FE:FB:F4 Discovering: no
[bluetooth]# connect 90:06:28:B6:B1:B6
Attempting to connect to 90:06:28:B6:B1:B6
[CHG] Device 90:06:28:B6:B1:B6 Connected: yes
[NEW] Endpoint /org/bluez/hci0/dev_90_06_28_B6_B1_B6/sep1 
[NEW] Endpoint /org/bluez/hci0/dev_90_06_28_B6_B1_B6/sep2 
[NEW] Endpoint /org/bluez/hci0/dev_90_06_28_B6_B1_B6/sep3 
[NEW] Endpoint /org/bluez/hci0/dev_90_06_28_B6_B1_B6/sep5 
[NEW] Endpoint /org/bluez/hci0/dev_90_06_28_B6_B1_B6/sep6 
[NEW] Endpoint /org/bluez/hci0/dev_90_06_28_B6_B1_B6/sep7 
[NEW] Transport /org/bluez/hci0/dev_90_06_28_B6_B1_B6/sep3/fd0 
Connection successful
[NEW] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 [default]
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 Repeat: off
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 Shuffle: off
[CHG] Device 90:06:28:B6:B1:B6 ServicesResolved: yes
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 Type: Audio
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 Subtype: Audio Book
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 Status: stopped
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 Name: Media Player
[CHG] Transport /org/bluez/hci0/dev_90_06_28_B6_B1_B6/sep3/fd0 Volume: 0x0000 (0)
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 Title: file
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 TrackNumber: 0x00000000 (0)
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 NumberOfTracks: 0x00000000 (0)
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 Duration: 0x00018737 (100151)
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 Album: Album inconnu
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 Artist: 
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 Genre: Genre inconnu
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 Position: 0x00000000 (0)
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 Position: 0x00000000 (0)
[CHG] Player /org/bluez/hci0/dev_90_06_28_B6_B1_B6/player0 Position: 0x00000000 (0)

this time the connection succeeds.

promeneur, 2022-10-01:
sudo journalctl -f -u bluetooth
oct. 01 03:05:55 grincheux bluetoothd[1054]: Endpoint registered: sender=:1.38 path=/MediaEndpoint/A2DPSource/faststream_duplex
oct. 01 03:05:55 grincheux bluetoothd[1054]: Endpoint registered: sender=:1.38 path=/MediaEndpoint/A2DPSink/opus_05
oct. 01 03:05:55 grincheux bluetoothd[1054]: Endpoint registered: sender=:1.38 path=/MediaEndpoint/A2DPSource/opus_05
oct. 01 03:05:55 grincheux bluetoothd[1054]: Endpoint registered: sender=:1.38 path=/MediaEndpoint/A2DPSource/opus_05_51
oct. 01 03:05:55 grincheux bluetoothd[1054]: Endpoint registered: sender=:1.38 path=/MediaEndpoint/A2DPSource/opus_05_71
oct. 01 03:05:55 grincheux bluetoothd[1054]: Endpoint registered: sender=:1.38 path=/MediaEndpoint/A2DPSink/opus_05_duplex
oct. 01 03:05:55 grincheux bluetoothd[1054]: Endpoint registered: sender=:1.38 path=/MediaEndpoint/A2DPSource/opus_05_duplex
oct. 01 03:05:55 grincheux bluetoothd[1054]: Endpoint registered: sender=:1.38 path=/MediaEndpoint/A2DPSource/opus_05_pro
oct. 01 03:37:39 grincheux bluetoothd[1054]: /org/bluez/hci0/dev_90_06_28_B6_B1_B6/sep3/fd0: fd(40) ready
oct. 01 03:37:57 grincheux bluetoothd[1054]: No matching connection for device

promeneur, 2022-10-01:
D-Bus messages related to the org.bluez bus during an attempt to connect the smartphone.
[buctl.log](https://github.com/bluez/bluez/files/9689399/buctl.log)

promeneur, 2022-10-01:
I try this experiment.

At the PC, I switch the internet connection from Ethernet to the PAN of the smartphone. This succeeds.

promeneur, 2022-10-05:
there is some progress with the kernel 5.19.12. At the PC, I can connect the smartphone.

closed 2022-10-05 by promeneur