bluez/bluez #402: bluez power on issue with steam session

abc-mikey, 2022-10-05:
Bluetooth works as expected when using gnome (or even steam in gnome), but when running a steam os session the bluetooth controller is continuously powered off.

[Here is an issue I opened for valve](https://github.com/ValveSoftware/steam-for-linux/issues/8794)

distribution: Fedora 35
bluez: 5.65
kernel: 5.19.12
steam-devices: 1.0.0.75
steam client version: Oct 4 2022, at 21:51:07

```
[bluetooth]# list
Controller 28:7F:CF:62:09:D9 nixon [default]
[bluetooth]# show
Controller 28:7F:CF:62:09:D9 (public)
        Name: nixon
        Alias: nixon
        Class: 0x00000000
        Powered: no
        Discoverable: yes
        DiscoverableTimeout: 0x00000000
        Pairable: yes
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0541
        Discovering: no
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x05 (5)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
[bluetooth]# power on
[CHG] Controller 28:7F:CF:62:09:D9 Class: 0x00000104
Changing power on succeeded
[CHG] Controller 28:7F:CF:62:09:D9 Powered: yes
[CHG] Controller 28:7F:CF:62:09:D9 Powered: no
[CHG] Controller 28:7F:CF:62:09:D9 Discovering: no
[CHG] Controller 28:7F:CF:62:09:D9 Class: 0x00000000
```

I also noticed an error reported by the bluetooth service that gets emitted when trying this:

```
Sep 12 13:59:19 nixon bluetoothd[987]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
```

Here is a trace with `btmon -w FILE` for 2 scenarios:

1. Start steam-session going directly into big picture mode only without a DE. Try to turn on controller. Ctrl+Alt+F3 to switch to console and start `bluetoothctl` and try to `power on`.
2. Start gnome-session starting steam then opening big picture mode. Turn on controller which connects. 

[gnome-session-steam-bpm.txt](https://github.com/bluez/bluez/files/9714400/gnome-session-steam-bpm.txt)
[steam-bpm-session.txt](https://github.com/bluez/bluez/files/9714401/steam-bpm-session.txt)

And here is the script for starting the steam-session if that helps:

[steamos-session.txt](https://github.com/bluez/bluez/files/9714404/steamos-session.txt)

Vudentz, 2022-10-05:
It seems something is requesting power off:

```
@ MGMT Command: Set Powered (0x0005) plen 1          {0x0001} [hci0] 137.172969
        Powered: Disabled (0x00)
```

So it doesn't seem to be anything BlueZ would be doing automagically but the upper layer actively disabling Bluetooth.

abc-mikey, 2022-10-06:
It must be the steam client must be misbehaving. Thanks for your support.

closed 2022-10-06 by abc-mikey