bluez/bluez #403: Sudden occurance of "Set scan parameters failed: Input/output error"

capiman, 2022-10-05:
Perhaps old thing. I saw this on a Raspberry PI CM4 with up-to-date software.

test@test:~ $ sudo hcitool --help
hcitool - HCI Tool ver 5.55

test@test:~ $ sudo hcitool dev
Devices:
        hci0    E4:5F:01:BA:08:45
test@test:~ $

test@test:~ $ sudo timeout 1s hcitool lescan
LE Scan ...
E4:5F:01:BA:05:2D (unknown)
E4:5F:01:BA:05:2D TestAdvertisement
C4:7C:8D:6D:37:02 (unknown)
C4:7C:8D:6D:37:02 Flower care
test@test:~ $ sudo timeout 1s hcitool lescan
Set scan parameters failed: Input/output error
test@test:~ $ sudo timeout 1s hcitool lescan
Set scan parameters failed: Input/output error
test@test:~ $ sudo timeout 1s hcitool lescan
Set scan parameters failed: Input/output error
test@test:~ $ sudo timeout 1s hcitool lescan
Set scan parameters failed: Input/output error
test@test:~ $ sudo timeout 1s hcitool lescan
Set scan parameters failed: Input/output error
test@test:~ $ sudo timeout 1s hcitool lescan
Set scan parameters failed: Input/output error
test@test:~ $ sudo timeout -s INT 1s hcitool lescan
Set scan parameters failed: Input/output error

After a reboot, it worked again.
I was able to reproduce it often/a lot of times.

I found a remark about it also here:
https://stackoverflow.com/questions/34189774/bluetooth-low-energy-passive-scan-but-do-not-loop-forever
But I think the workaround (ue -s INT) is not helpful.

capiman, 2022-10-13:
I have not used bluetoothctl before.
 
 I found 
 https://stackoverflow.com/questions/36607626/how-to-use-bluetoothctl-like-hcitool-lescan-to-report-repeated-proximity-beacons

Can you help me? I want to scan for (LE) advertisements and also receive --duplicates?
Optional: I get the detailed data currently with hcidump, but any other way would also be possible.

capiman, 2022-10-15:
When using hcitool lescan --duplicate together with hcidump
I sometimes received

Can't init device hci0: Connection timed out (110)
Could not open device: No such device
Can't init device hci0: Connection timed out (110)
Could not open device: No such device
Can't init device hci0: Connection timed out (110)
Could not open device: No such device
Can't init device hci0: Connection timed out (110)
Could not open device: No such device

I tried   

  hciconfig hci0 down
  hciconfig hci0 up

but seems not to be right way to recover. Is there a better one?

capiman, 2022-10-24:
Any update on this topic? It seems it is occuring again...

Vudentz, 2022-10-24:
@capiman Is that reproducible when using bluetoothctl? If the problem persists please collect the HCI traces using btmon.

capiman, 2022-10-26:
I was not able to send multiple commands to bluetoothctl, so now I used 

https://stackoverflow.com/questions/36607626/how-to-use-bluetoothctl-like-hcitool-lescan-to-report-repeated-proximity-beacons

the solution with expect.

I need to use a timeout of 2s even when we sent advertisements with a rate of 40 equal messages per second,
to receive even only 1 with a high probability. 
Used earlier the hcitool scan le and hcidump solutions where I get all and fast.
So bluetoothctl was no good experience.

Without "expect", bluetoothctl exited immediately after switching scanning on. And as written (I think) on above page,
a sleep seems to be not possible. Nevertheless I use the timeout with 2s.

I tried hcidump in parallel to bluetoothctl, but I saw only small 0x04 packets. Long ago, since I worked on HCI protocol directly, but I think the small HCI packets were only events confirming some of the sent commands.
I don't know why I don't get events with raw advertisement data even when using 

hcidump --raw

bluetoothctl seems to display BD-address, but I was not able to see raw advertised data.

To make story short: I have it running with big disadvantages (timeout, now raw data), but due to my time restriction to implement something for a test where we could verify a advertisement reception is working at all
(I have both side under control, sender sending at rate of 40 advertisements per second, needed min/max change in kernel parameter, as well as python script for sending), it has done at least till now.

I will retest perhaps mid or end of next week, when time is allowing me to do this, and if it is needed.

capiman, 2022-10-26:
Ah, just found some hcidump output in one of my old SSH windows, so I just copy&paste it here:

$ hcidump --raw
HCI sniffer - Bluetooth packet analyzer ver 5.55
device: hci0 snap_len: 1500 filter: 0xffffffff
> 04 0E 04 01 05 20 00
> 04 0E 04 01 0B 20 00
> 04 0E 04 01 0C 20 00
> 04 0E 04 01 0C 20 00
> 04 0F 04 00 01 01 04
> 04 01 01 00
^C

I saw "data from other devices" during the time the hcidump was running.

capiman, 2022-10-26:
BTW: Just for my personal interest - completely off-topic. Would I be able to open /dev/hci0 and sent raw HCI commands and receive HCI events? Some chips need firmware download during startup, this is then already done?
I think I can figure out what HCI commands need to be sent and what answers to receive, but then, I could control the on-board Bluetooth chip completely by my own. I known everything else could perhaps interfering me and must be switched off.

PS2: Tried it already some days ago with another solution:
https://github.com/bluez/bluez/issues/406
But this way was also not helpful, due to problems, dealing with --duplicate.
It could be a solution, to understand and change the sender side, if I am able to send changing advertisement data, which the filter passes every time (or only checking BD address)?

Vudentz, 2022-10-26:
> BTW: Just for my personal interest - completely off-topic. Would I be able to open /dev/hci0 and sent raw HCI commands and receive HCI events? Some chips need firmware download during startup, this [...]

Firmware upload is already handled in the kernel.

> PS2: Tried it already some days ago with another solution: #406 But this way was also not helpful, due to problems, dealing with --duplicate. It could be a solution, to understand and change the [...]

Duplicate filtering is controller specific, in practice most just check the address match, so it is not very reliable. Btw, use btmon to monitor the HCI traffic, hcidump has been deprecated and doesn't show things like the MGMT commands.

reimerp, 2025-07-06:
Hi, ran into this old issue as well, and it seems an issue related to libpcap. A similar issue was reported against bpftool ("Clear errno after libcap's checks")

`hcitool scan` is working, but lescan (needs sudo) fails.
```
$ strace -e prctl hcitool lescan
prctl(PR_CAPBSET_READ, CAP_MAC_OVERRIDE) = 1
prctl(PR_CAPBSET_READ, 0x30 /* CAP_??? */) = -1 EINVAL (Das Argument ist ungültig)
prctl(PR_CAPBSET_READ, CAP_CHECKPOINT_RESTORE) = 1
prctl(PR_CAPBSET_READ, 0x2c /* CAP_??? */) = -1 EINVAL (Das Argument ist ungültig)
prctl(PR_CAPBSET_READ, 0x2a /* CAP_??? */) = -1 EINVAL (Das Argument ist ungültig)
prctl(PR_CAPBSET_READ, 0x29 /* CAP_??? */) = -1 EINVAL (Das Argument ist ungültig)
Set scan parameters failed: Input/output error
+++ exited with 1 +++
```

I am using 
bluez          5.79-2ubuntu1.1 
libcap.so.2.73

closed 2025-09-29 by github-actions[bot]