bluez/bluez #406: Bleak -> dbus -> BlueZ: LE Set Scan Enable - filter duplicates cannot be disabled

capiman, 2022-10-05:
Details see here:

https://github.com/hbldh/bleak/issues/1065

especially here:

https://github.com/hbldh/bleak/issues/235

and here

https://github.com/torvalds/linux/commit/abfeea476c68afea54c9c050a2d3b19d5d2ee873

Would it be possible to tell me, if it is contained in a certain kernel? Or was it rejected?
Seems to be still not working in 5.17 nor 5.19. Or perhaps a different error?

dlech, 2022-10-06:
I think it is working as designed. The duplicate filter is currently only disabled if an [advertisement monitor](https://github.com/bluez/bluez/blob/master/doc/advertisement-monitor-api.txt) is in place.

So, I think this is a feature request to have a way to also not filter duplicates while actively scanning even when an advertisement monitor is not being used.

dlech, 2022-10-06:
Probably a duplicate of #119.

Vudentz, 2022-10-14:
> Probably a duplicate of #119.

Disabling duplicated is probably overkill for a normal operation, it would generate way too many D-Bus traffic, I think we need a dedicated socket type if we really want to have low access to the advertising bearer.

dlech, 2022-10-14:
So something like "AcquireNotify" but for advertisements?

Vudentz, 2022-10-14:
> So something like "AcquireNotify" but for advertisements?

Yep, it probably wouldn't even require D-Bus to begin with since we could just use setsockopt to allow the application to setup the scanning (RX) and advertising (TX) parameters/filtering, the challenge is to make this coexist with the daemon/mgmt interface to not disrupt other applications in the system.

closed 2025-08-16 by github-actions[bot]