bluez/bluez #408: Error codes in 0xE0-0xFF range

martijnvanwelie, 2022-10-10:
Hi,

I am still working on passing PTS tests for some new services that the BT SIG is developing. For one test scenario I need to return an error code 0xFF which is defined as 'Out of Range error response' and comes from a special range:

`Common Profile and Service Error Codes use 0xE0 – 0xFF and are defined in the Core Specification Supplement.`

In issue #380 you opened up the 0x80 – 0x9F range. But not yet the 0xE0-0xFF range.

Would be possible to open up de 0xE0-0xFF in the same way?

Vudentz, 2022-10-11:
You mean something like:

/*
 * Common Profile and Service Error Code descriptions (see Supplement to the
 * Bluetooth Core Specification, sections 1.2 and 2). The error codes within
 * 0xE0-0xFC are reserved for future use. The remaining 3 are defined as the
 * following:
 */
#define BT_ERROR_CCC_IMPROPERLY_CONFIGURED      0xfd
#define BT_ERROR_ALREADY_IN_PROGRESS            0xfe
#define BT_ERROR_OUT_OF_RANGE                   0xff

These are actually in use by the core already like BT_ERROR_OUT_OF_RANGE:

https://github.com/bluez/bluez/blob/master/src/shared/att.c#L1860

martijnvanwelie, 2022-10-12:
Yes, indeed those ones! So I would need to be able to return them from my application in response to a write to a descriptor and characteristic...

Vudentz, 2022-10-12:
> Yes, indeed those ones! So I would need to be able to return them from my application in response to a write to a descriptor and characteristic...

I wonder if for the likes to BT_ERROR_OUT_OF_RANGE it wouldn't be better to have some means to detect that the Value is fixed size so we can handle such errors internally instead of depending on application i.e. add ValueLength property, thoughts?

martijnvanwelie, 2022-10-13:
These error codes are really intended for the application layer and don't related to 'communication issues'. For example, in my case a central will write a descriptor with a configuration and the peripheral needs to respond whether or not it accepts the configuration. But the reasons for acceptance are totally application level specific. In my case, the parameters I receive need to be within certain limits and the limits are specific to my application.

You can see my code here:
https://github.com/weliem/ghs-server/blob/main/examples/peripheral/peripheral.c#L148

So in that code I check the MDC code I receive and the values of the other parameters. So you cannot generically do this already in the Bluez stack as only the application has this logic. The error I currently return is of course wrong, it should be 0xFF....

Since it concerns only 3 values that are in use, perhaps the easiest fix for now is to simply add them as returnable error codes. So next to BLUEZ_ERROR_REJECTED you could introduce BLUEZ_ERROR_OUT_OF_RANGE and BLUEZ_ERROR_ALREADY_IN_PROGRESS and BLUEZ_ERROR_CCC_IMPROPERLY_CONFIGURED. The rest of the range is reserved for future use....

closed 2025-08-16 by github-actions[bot]