bluez/bluez #409: Cannot send an indication to 1 specific central

martijnvanwelie, 2022-10-10:
Hi,

I am developing a 'peripheral' and as part of trying to pass PTS tests for a new BT SIG service, I need to be able to send an indication to 'specific' centrals. The scenario is as follows;

- A central does a write to change the configuration of a peripheral. Let's assume it is the clock of the peripheral.
- The peripheral then needs to send an indication to all connected centrals, except the one that did the change because that central knows it already.

This pattern is not uncommon in BT SIG services by the way. However, as far as I know, I can only trigger an indication by publishing a PropertiesChanged signal, which isn't 'directional'. So the indication goes to all centrals, but I only should send it to specific centrals.

So I was thinking: how about a new DBus method for a characteristic called 'Notify(object, value)' where object is the path of the central I want to send the indication/notification to? That way I could implement my test scenario and nothing is broken for people that want to keep using the PropertiesChanged signal.

Vudentz, 2022-10-10:
@martijnvanwelie I think it would be better to implement something at AcquireNotify level so the application can then write the notification/indication using a file descriptor and then use something like sendmsg and then we have can extra information as as part of the msghdr:

man sendmsg:
You may send control information (ancillary data) using the msg_control and msg_controllen  members.

We can then do some filtering like we have a list of bdaddr and some flags to say include/exclude logic for example.

martijnvanwelie, 2022-10-12:
Well, not sure if I understand your proposal. My app acts as a 'peripheral' and the centrals can be anything like iOS/Android/Nordic. So none of the centrals will do a AcquireNotify since that is only for Bluez centrals.

alexandrefresnais, 2022-10-29:
I would also be very interested in some sort of `Notify(object, value)` for device-specific notifications. Is there any other way we can implement some sort of Nordic UART BLE Server otherwise ? I do not see how I can answer a specific client.

closed 2025-08-16 by github-actions[bot]