bluez/bluez #410: At KDE login, the PC is not automatically connected to the Smartphone

promeneur, 2022-10-10:
Bluetooth 5.2 pci card EDUP LOVE EP-9651GS (chipset Intel AX210)
Samsung S7, Bluetooth 4.2

KDE Gears 22.08.1
Operating System: openSUSE Tumbleweed 20221003
KDE Plasma Version: 5.25.5
KDE Frameworks Version: 5.98.0
Qt Version: 5.15.6
Kernel Version: 5.19.13-1-default (64-bit)
Graphics Platform: X11

bluez 5.65
pipewire 3.58

At the PC, connect the smartphone. This succeeds.
Restart the PC. The KDE session is closed then the OS is stopped then restarted.
Open the same user KDE session.
Bluetooth is enabled.
The smartphone is not automatically connected.

KDE team says KDE is not responsible to auto-connect device.

Read at
https://bugs.kde.org/show_bug.cgi?id=460077

Vudentz, 2022-10-10:
@promeneur we do have a policy plugin which can reconnect profiles but it only really kick in if the device was not disconnected cleanly:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/main.conf#n265

Note that we can't really tell what is going on the session bus to lets say start reconnecting so if you after some action like on-login reconnect, etc, that probably would require some action from the user session.

promeneur, 2022-10-15:
@Vudentz 

I don't understand clearly. Is bluez responsible for the problem or not ?

Where I must send a bug report ?

Thanks

closed 2025-08-16 by github-actions[bot]