bluez/bluez #411: Redmi Buds 3 Lite: A2DP only after manualy reconnecting

Harvie, 2022-10-10:
Xiaomi Redmi Buds 3 Lite true wireless earbuds are comforable, play nicely and are very cheap. I have them paired and everything kinda works.

But i take earbud out of case, which turns it on. It automaticaly pairs and connects. But only in HSP/HFP mode. When i open blueman-manager, click disconnect and then connect (or i disconnect/connect it using bluetoothctl...). It gets reconnected and only then the A2DP profile appears and starts working correctly.

Running ArchLinux with bluez 5.65 and following config (tried various combinations of config, but so far nothing helped):

```ini
# grep -v ^# /etc/bluetooth/audio.conf

[General]
Enable = Source,Sink,Headset,Gateway,Control,Socket,Media

[Headset]
HFP=true
MaxConnected=1
FastConnectable=true

# grep -v ^# /etc/bluetooth/main.conf | uniq

[General]
AutoConnect=true
MultiProfile = multiple
FastConnectable = true
Experimental = true
KernelExperimental = true

[BR]

[LE]

[GATT]

[AVDTP]

[Policy]

[AdvMon]

```

Is this some kind of bug?

Vudentz, 2022-10-10:
@Harvie please attach the HCI trace using btmon.

Harvie, 2022-10-10:
@Vudentz ok i'll try. so far i only have this log.

```
říj 10 21:37:40 anemophobia bluetoothd[2257]: profiles/audio/avdtp.c:cancel_request() Discover: Connection timed out (110)
říj 10 21:37:42 anemophobia bluetoothd[2257]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 7C:C9:5E:15:A2:5A: Device or resource busy
říj 10 21:37:43 anemophobia bluetoothd[2257]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 7C:C9:5E:15:A2:5A: Device or resource busy
říj 10 21:37:48 anemophobia bluetoothd[2257]: profiles/audio/avdtp.c:cancel_request() Discover: Connection timed out (110)
```

Harvie, 2022-10-10:
Did that: [redmibuds.zip](https://github.com/bluez/bluez/files/9748562/redmibuds.zip)

Contains two files:
failedconnect.dump.txt   - unsuccessful connection when i power on the earbud and it connects to laptop
reconnect.dump.txt   - successful connection when laptop initiates reconnection

Vudentz, 2022-10-11:
```
> ACL Data RX: Handle 11 flags 0x02 dlen 6                                                                                                                                                                                                                               #54 [hci0] 10.001086
      Channel: 65 len 2 [PSM 25 mode Basic (0x00)] {chan 1}
      AVDTP: Discover (0x01) Command (0x00) type 0x00 label 14 nosp 0
< ACL Data TX: Handle 11 flags 0x00 dlen 54                                                                                                                                                                                                                              #55 [hci0] 10.001400
      Channel: 66 len 50 [PSM 25 mode Basic (0x00)] {chan 1}
      AVDTP: Discover (0x01) Response Accept (0x02) type 0x00 label 14 nosp 0
        ACP SEID: 1
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 2
          Media Type: Audio (0x00)
          SEP Type: SNK (0x01)
          In use: No
        ACP SEID: 3
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 4
          Media Type: Audio (0x00)
          SEP Type: SNK (0x01)
          In use: No
        ACP SEID: 5
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 6
          Media Type: Audio (0x00)
          SEP Type: SNK (0x01)
          In use: No
        ACP SEID: 7
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 8
          Media Type: Audio (0x00)
          SEP Type: SNK (0x01)
          In use: No
        ACP SEID: 9
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 10
          Media Type: Audio (0x00)
          SEP Type: SNK (0x01)
          In use: No
        ACP SEID: 11
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 12
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 13
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 14
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 15
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 16
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 17
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 18
          Media Type: Audio (0x00)
          SEP Type: SNK (0x01)
          In use: No
        ACP SEID: 19
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 20
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 21
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 22
          Media Type: Audio (0x00)
          SEP Type: SNK (0x01)
          In use: No
        ACP SEID: 23
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No
        ACP SEID: 24
          Media Type: Audio (0x00)
          SEP Type: SRC (0x00)
          In use: No

```

I think the device is overwelmed by this response, does Arch has this many codecs enabled? Anyway the device start collecting the capabilities for each SEID and it seems it is not able to respond to our Discover request:

```
< ACL Data TX: Handle 11 flags 0x00 dlen 6                                                                                                                                                                                                                              #180 [hci0] 10.949129
      Channel: 66 len 2 [PSM 25 mode Basic (0x00)] {chan 1}
      AVDTP: Discover (0x01) Command (0x00) type 0x00 label 14 nosp 0

```

So after 6 seconds we time out:

`= bluetoothd: profiles/audio/avdtp.c:cancel_request() Discover: Connection timed out (110)                                                                                                                                                                                          16.948664`

Id probably disable the Linux specific codecs (OPUS?), etc, to see if the device start to respond:

`Vendor ID: The Linux Foundation (0x000005f1)`

Harvie, 2022-10-11:
> Id probably disable the Linux specific codecs 

is there way to do this in bluez? or i have to do this in wireplumber (since i use pipewire)?

> So after 6 seconds we time out:

Can i just increase this timeout somehow? :innocent:

Harvie, 2022-10-11:
I think i maybe got it working!

created copy of `/usr/share/wireplumber/bluetooth.lua.d/50-bluez-config.lua`
in `/etc/wireplumber/bluetooth.lua.d/50-bluez-config.lua`

It has to contain the following

```lua
bluez_monitor.properties = {
  ["bluez5.codecs"] = "[ sbc sbc_xq aac ldac aptx aptx_hd ]",
  ["with-logind"] = true,
}
```
i am not sure if i didn't left out something actualy usefull.
by default following codecs are active:

```lua
  ["bluez5.codecs"] = "[ sbc sbc_xq aac ldac aptx aptx_hd aptx_ll aptx_ll_duplex faststream faststream_duplex ]",
```

closed 2022-10-11 by Harvie

Abradox, 2022-11-19:
Confirmed for Razer Opus. A2DP is gone if i turn off and on my headset when being paired. I left only  ["bluez5.codecs"] = "[ sbc sbc_xq ]" and problem was solved.

kuraga, 2023-01-07:
@Harvie but what codec did we have to turn off, exactly?

Harvie, 2023-01-07:
Can be seen in the highlighted code in my previos comments.

Harvie, 2025-10-06:
Yeah, it's still kinda weird. Haven't tried Redmi buds in a while, but i have some other generic brand TWS buds and sometimes they only seem to connect on second try. it does especially happen after waking from suspend-to-ram, but sometimes also randomly (restart by puting earphones back to case and removing them again seems to be enough.).

It's weird. Never had such issues on android phone...