bluez/bluez #412: Create documentation for bluetoothctl

potto216, 2022-10-11:
Create documentation for bluetoothctl in the form of a man page using [.rst](https://docutils.sourceforge.io/docs/ref/rst/restructuredtext.html). bluetoothctl.rst will reside in the client directory as shown with the forked repo.

Vudentz, 2022-10-11:
@potto216 have a look at https://github.com/bluez/bluez/commit/4314f70667217549054f61def6575085a7a18d78, do not modify the Makefile directly since that is generated from the likes of Makefile.am.

Vudentz, 2022-10-11:
@potto216 perhaps it would be a good idea to split the submenus on its own manpage like git does for its commands:

[git-add(1)](https://manpages.debian.org/stretch/git-man/git-add.1.en.html)
Add file contents to the index.

So we leave only the main menu on bluetoothctl.rst and then have bluetoothctl-advertise.rst, etc, as separate manpages that way we don't have to include everything on one manpage alone.

potto216, 2022-10-12:
@Vudentz I modified the `Makefile.tools` and when you get a chance would like your feedback on the layout for [bluetoothctl.rst](https://raw.githubusercontent.com/potto216/bluez/412-bluetoothctl-doc/client/bluetoothctl.rst) and [bluetoothctl-advertise.rst](https://raw.githubusercontent.com/potto216/bluez/412-bluetoothctl-doc/client/bluetoothctl-advertise.rst)

Vudentz, 2022-10-12:
> @Vudentz I modified the `Makefile.tools` and when you get a chance would like your feedback on the layout for [...]

As a start it seems fine, you should probably check how git list their commands as links, once we merge these we can start working on having proper examples.

Vudentz, 2023-07-17:
@potto216 are you still working on this? Id really love to see this being contributed as we will be integrating more functionality to bluetoothctl, such as the mgmt and obex submenus.

potto216, 2023-07-18:
@Vudentz that is great and I am, although being able to have full documentation on all the current features is probably not realistic in the short term because of the time required to understand the functionality from the source code and then verify it. Currently I have the docs broken into the sections as you suggested (.i.e. https://github.com/potto216/bluez/blob/412-bluetoothctl-doc/client/bluetoothctl-advertise.rst). Do you have suggestions on what general changes I need to make to my current changes to have them mergeable? Also what are the priorities on the documentation and the due date?

Vudentz, 2023-07-18:
@potto216 Id suggest you use a table to document the commands: https://patchwork.kernel.org/project/bluetooth/patch/20230717230445.168998-1-luiz.dentz@gmail.com/

Vudentz, 2023-07-18:
@potto216 btw, you can use code-block:: to document the examples:

https://raw.githubusercontent.com/bluez/bluez/master/tools/test-runner.rst

Vudentz, 2023-07-18:
@potto216 we don't have a strict timeline for adding documentation, but Id like to have them as soon as possible, even if we don't have proper example, etc, we can always extend it later.

Vudentz, 2023-07-20:
@potto216 Ive changed my mind about using tables, they are hard to keep under 80 columns, beside I think having each command as a chapter make it simpler to add examples, etc, anyway you can see how they look like in the following set where Im adding bluetoothctl-mgmt.rst and bluetoothctl-monitor.rst so I could reference them in btmgmt.rst:

https://patchwork.kernel.org/project/bluetooth/list/?series=767584

potto216, 2023-07-20:
@Vudentz I'll merge in those changes and revert my table work. Speaking of tables since they may be used again and I am new to rst I was wondering if you ever had the following two issues which seem indirectly related. First, I will see random gaps in the rendered tables when using `man` whose position change as I add rows. An example is below:
[image]

I verified there are not hidden/unusual characters and the whitespace are spaces. Additionally in experimenting with trying to resolve this I found that sometimes when I created two tables the formatting looked like:
[image]

But then if I added a table in an unrelated position much earlier in the page they suddenly rendered fine (below)
[image]

This seems like some kind of global page issue. I'm running this on Debian 11 with the latest package updates. If you have solved this problem before I would like to know about it, so far  my search for a solution or root cause has been unsuccessful.

Vudentz, 2023-07-20:
@potto216 I had the problem as well, not sure if it is rst2man or something else though, anyway you can see the final version is now pushed:

https://github.com/bluez/bluez/blob/master/tools/btmgmt.rst
https://github.com/bluez/bluez/blob/master/client/bluetoothctl-monitor.rst
https://github.com/bluez/bluez/blob/master/client/bluetoothctl-mgmt.rst

potto216, 2023-07-30:
@Vudentz is this what you are thinking of for the format https://github.com/potto216/bluez/blob/412-bluetoothctl-doc/client/bluetoothctl-scan.rst ? If so I'll commit the other files.

potto216, 2023-08-13:
@Vudentz I've finished the reformatting of the documentation including additional details. Should I do a PR from my branch https://github.com/potto216/bluez/tree/412-bluetoothctl-doc?

Vudentz, 2023-08-14:
@potto216 have the commits on a per menu basis, also when you refer to bluetoothctl-(menu) had it refer to the man section (1).

potto216, 2023-11-26:
@Vudentz I noticed with my build of bluez which includes all the [`bluetoothctl` documentation commits](https://github.com/bluez/bluez/commit/4b353ae99ab66390561730974d0c4fecc4f38d6a) that the main `bluetoothctl` documentation page isn't updated. I cannot find a `bluetoothctl.rst` file, so should I create one or is it generated some other way?

Vudentz, 2023-11-27:
> @Vudentz I noticed with my build of bluez which includes all the [`bluetoothctl` documentation commits](https://github.com/bluez/bluez/commit/4b353ae99ab66390561730974d0c4fecc4f38d6a) that the main [...]

Hmm, did I miss one of the commits? I don't think we had the main documentation done yet, so yes please create one if you have time.

potto216, 2023-11-28:
I found I had not committed bluetoothctl.rst. I'll submit a patch for it.

potto216, 2025-01-05:
I'm closing this issue because I am now focused on using [BlueR tools](https://github.com/bluez/bluer) and I don't plan on additional documentation updates on bluetoothctl.

closed 2025-01-05 by potto216