bluez/bluez #413: No Default Controller available | Invalid index (0x11)

Corrupter-rot, 2022-10-12:
Before you say go google the issue, I have already tried most of the solution but nothing seems to work. That's why I'm opening a issue here.

As for the issue 
```
$ bluetoothctl
Agent registered
[bluetooth]# scan on
No default controller available
[bluetooth]# power on
No default controller available
[bluetooth]# exit
```
This is what i get when trying to do anything with bluetoothctl.

Here is what I found in my `journalctl -u bluetooth` may help with debugging.
```
Oct 12 22:12:13 Rot systemd[1]: Started Bluetooth service.
Oct 12 22:12:13 Rot bluetoothd[2419]: Starting SDP server
Oct 12 22:12:13 Rot bluetoothd[2419]: Bluetooth management interface 1.22 initialized
Oct 12 22:19:48 Rot bluetoothd[2419]: src/adapter.c:index_removed() Ignoring index removal for a no>
Oct 12 22:19:48 Rot bluetoothd[2419]: src/adapter.c:reset_adv_monitors_complete() Failed to reset A>
Oct 12 22:19:48 Rot bluetoothd[2419]: Failed to read exp features info: Invalid Index (0x11)
Oct 12 22:19:48 Rot bluetoothd[2419]: Failed to read info for index 0: Invalid Index (0x11)
Oct 12 22:20:36 Rot bluetoothd[2419]: src/adapter.c:index_removed() Ignoring index removal for a no>
Oct 12 22:20:36 Rot bluetoothd[2419]: src/adapter.c:reset_adv_monitors_complete() Failed to reset A>
Oct 12 22:20:36 Rot bluetoothd[2419]: Failed to read exp features info: Invalid Index (0x11)
Oct 12 22:20:36 Rot bluetoothd[2419]: Failed to read info for index 0: Invalid Index (0x11)
Oct 12 22:20:36 Rot bluetoothd[2419]: Terminating
Oct 12 22:20:36 Rot systemd[1]: Stopping Bluetooth service...
Oct 12 22:20:36 Rot bluetoothd[2419]: Stopping SDP server
Oct 12 22:20:36 Rot bluetoothd[2419]: Exit
Oct 12 22:20:36 Rot systemd[1]: bluetooth.service: Deactivated successfully.
Oct 12 22:20:36 Rot systemd[1]: Stopped Bluetooth service.
```

Earlier the problem was not there but after upgrading recently, this issue occured. Sadly I didn't took any backup so I can't rollback.

If you guys need anymore info, let me know.

`lsusb` showes my bluetooth controller
```
Bus 001 Device 003: ID 13d3:3548 IMC Networks Bluetooth Radio
```

`sudo systemctl status bluetooth.service`
```
sudo systemctl status bluetooth.service
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: disabled)
     Active: active (running) since Thu 2022-10-13 01:11:48 IST; 1h 11min ago
       Docs: man:bluetoothd(8)
   Main PID: 7142 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 8835)
     Memory: 840.0K
        CPU: 62ms
     CGroup: /system.slice/bluetooth.service
             └─7142 /usr/lib/bluetooth/bluetoothd

Oct 13 01:11:48 Rot systemd[1]: Starting Bluetooth service...
Oct 13 01:11:48 Rot bluetoothd[7142]: Bluetooth daemon 5.65
Oct 13 01:11:48 Rot systemd[1]: Started Bluetooth service.
Oct 13 01:11:48 Rot bluetoothd[7142]: Starting SDP server
Oct 13 01:11:48 Rot bluetoothd[7142]: Bluetooth management interface 1.22 initialized
```

Vudentz, 2022-10-12:
@Corrupter-rot please add the HCI/btmon traces, what kernel version and distro are you using?

Corrupter-rot, 2022-10-12:
OS : Arch Linux
Kernel : 5.19.13-arch1-1

As for btmon traces I don't quite know how to do it. Can you please guide me.

Vudentz, 2022-10-12:
@Corrupter-rot oh, looks like meshd is grabing the controller, @bgix I that explain why bluetoothd cannot use the index?

Corrupter-rot, 2022-10-12:
@Vudentz is there a quick fix to this, because I'm unable to use my Bluetooth?

Vudentz, 2022-10-12:
> @Vudentz is there a quick fix to this, because I'm unable to use my Bluetooth?

Try disabling meshd service, not sure why Arch had it enabled to begin with, it is probably not useful for you anyway.

Corrupter-rot, 2022-10-12:
It finally worked!! Man I spent my whole day searching for solution but nothing worked🥲.

bgix, 2022-10-12:
If you are on a single controller system, you will (as noted by @Vudentz ) need to use one daemon or the other.  This will change shortly after v6 of the kernel is released, when the bluetooth-meshd will no longer reserve one of the controllers for exclusive use.

To disable the bluetooth-meshd daemon:
$ systemctl disable bluetooth-mesh
$ systemctl stop bluetooth-mesh

closed 2022-10-12 by Corrupter-rot