bluez/bluez #414: systemd logs `ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 755 ConfigurationDirectoryMode: 555)`

paulmenzel, 2022-10-13:
On Debian sid/unstable with *bluez* 5.65-1, starting the Bluetooth service with `systemctl start bluetooth.service`, systemd logs to the journal:

    systemd[8067]: ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 755 ConfigurationDirectoryMode: 555)

Vudentz, 2022-10-13:
@paulmenzel I think this is due to the update in bluetooth.service (https://github.com/bluez/bluez/commit/00cfb36e20e3c35db2150e7d0351ad7b8442e2d8), before we used to create it which is probably why it is cause this change, not sure if the is a way to inform systemd to update/force the mode to avoid this warning.

maxnikulin, 2024-01-02:
If your intention is really to have `/etc/bluetooth` read-only (#329) then remove `ConfigurationDirectory=bluetooth` and `ConfigurationDirectoryMode=0555` from `bluetooth.service.in`. In presence of `ProtectSystem=strict` the former directive causes `/etc`  mounted in `rw` mode and the latter one leads to the warning due to inconsistency with actual permissions of `/etc/bluetooth`

```
# nsenter --all --target "$(systemctl show --value --property=MainPID bluetooth.service)"
# touch /etc/bluetooth/ro-test
# rm /etc/bluetooth/ro-test
# exit
```

A workaround: create the following systemd configuration file overriding a part of the unit directives.
`/etc/systemd/system/bluetooth.service.d/disable-configuration-directory.conf`
```
[Service]
ConfigurationDirectory=
ConfigurationDirectoryMode=
```
```
# systemctl daemon-reload && systemctl restart bluetooth.service
# nsenter --all --target "$(systemctl show --value --property=MainPID bluetooth.service)"
# touch /etc/bluetooth/ro-test
touch: cannot touch '/etc/bluetooth/ro-test': Read-only file system
# exit
```

I am trying to debug an issue with Intel AX200 and I have been confused whether `bluetoothd` needs `/etc/bluetooth` being writable. There is a number of forum threads and distribution-specific bug reports related to this warning:
- https://bugs.debian.org/1017988
- https://launchpad.net/bugs/2037212

prahal, 2025-08-03:
This issue still stands with Debian bluez 5.82-1.1. Attempting to remove stale with a comment.

@hadess  in https://github.com/bluez/bluez/issues/329#issuecomment-1102459104 tells `ConfigurationDirectoryMode=0555` has to be added to make /etc/bluetooth read only (which was done in https://github.com/bluez/bluez/commit/00cfb36e20e3c35db2150e7d0351ad7b8442e2d8 ).
From the context, the aim is to make /etc/bluetooth read only to the systemd service processes, not per se to the other host processes.

Note that not using ConfigurationDirectory alters behavior 
 ```
Except in case of ConfigurationDirectory=, the innermost specified directories will be owned by the user and group specified in User= and Group=.
```

Wouldn't `ConfigurationDirectory=bluetooth::ro` be more appropriate than a mode that enforce host access modes? Though only available since systemd v257.
https://www.freedesktop.org/software/systemd/man/latest/systemd.exec.html#RuntimeDirectory 
```
RuntimeDirectory=, StateDirectory=, CacheDirectory= and LogsDirectory= optionally support two more parameters, separated by ":". The second parameter will be interpreted as a destination path that will be created as a symlink to the directory. The symlinks will be created after any BindPaths= or TemporaryFileSystem= options have been set up, to make ephemeral symlinking possible. The same source can have multiple symlinks, by using the same first parameter, but a different second parameter. The third parameter is a flags field, and since v257 can take a value of ro to make the directory read only for the service. This is also supported for ConfigurationDirectory=. If multiple symlinks are set up, the directory will be read only if at least one is configured to be read only. To pass a flag without a destination symlink, the second parameter can be empty, for example:

ConfigurationDirectory=foo::ro
```

Or maybe replaces ConfigurationDirectoryMode=0555 by ReadOnlyPaths=/etc/bluetooth to only apply read only to the bluetoothd systemd service processes not to the whole host.

And last made it explicit that bluetoothd requires /etc/bluetooth read only for all host processes, so distribution maintainers migrates the /etc/bluetooth host access modes.

lonnic, 2026-01-07:
same error on Debian 13.2 with bluez 5.85-1

Jan 07 16:10:03 hostname (bluetoothd)[1426]: bluetooth.service: ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 755 ConfigurationDirectoryMode: 555)

Package: bluez
Version: 5.85-1
Priority: optional
Section: admin
Maintainer: Debian Bluetooth Maintainers <team+pkg-bluetooth@tracker.debian.org>
Installed-Size: 5,123 kB
Depends: libasound2t64 (>= 1.0.17), libc6 (>= 2.42), libdbus-1-3 (>= 1.10), libdw1t64 (>= 0.127), libglib2.0-0t64 (>= 2>
Suggests: pulseaudio-module-bluetooth
Homepage: http://www.bluez.org
Tag: admin::hardware, admin::kernel, devel::lang:c, devel::library,
 hardware::bluetooth, implemented-in::c, protocol::TODO,
 role::devel-lib, role::program, use::driver, use::synchronizing
Download-Size: 1,507 kB
APT-Manual-Installed: yes
APT-Sources: http://deb.debian.org/debian sid/main amd64 Packages
Description: Bluetooth tools and daemons
 This package contains tools and system daemons for using Bluetooth devices.

mvz, 2026-02-07:
What activity can we perform except commenting once a month?

mvz, 2026-08-20:
Every month i get a notification from the bot and I run something like this:

```shell
journalctl --boot | grep "ConfigurationDirectory 'bluetooth' already exists"
```
Every month I get journal entries that tell me this issue is still relevant.

mvz, 2026-08-20:
Oh @hadess I missed that you already removed the `stale` label.

closed 2026-09-10 by BluezTestBot

mvz, 2026-09-11:
Thanks @hadess for taking care of this!