bluez/bluez #415: can't connect to paired audio device

Sikeen, 2022-10-13:
### SUMMARY
when trying to connect to a paired device, it displays "connection to device failed"

STEPS TO REPRODUCE
1. pair device
2. successfully connect to device
3. restart
4. try to connect to device

OBSERVED RESULT
it fails, and says "connection to device failed" 
you have to unpair and pair the device again for it to connect.
after a restart it fails again

EXPECTED RESULT
the audio device(earbuds) successfully connects

SOFTWARE/OS VERSIONS
Linux/KDE Plasma: 

KDE Plasma Version: 5.25.5
KDE Frameworks Version: 5.97.0
Qt Version: 5.15.5

ADDITIONAL INFORMATION

INXI:
```
System:
  Host: simon-m-zephyrusg14-ga402rkga402rk Kernel: 6.0.0-1-MANJARO
    arch: x86_64 bits: 64 Desktop: KDE Plasma v: 5.25.5 Distro: Manjaro Linux
Machine:
  Type: Laptop System: ASUSTeK product: ROG Zephyrus G14 GA402RK_GA402RK
    v: 1.0 serial: <superuser required>
  Mobo: ASUSTeK model: GA402RK v: 1.0 serial: <superuser required>
    UEFI: American Megatrends LLC. v: GA402RK.315 date: 09/21/2022
Battery:
  ID-1: BAT0 charge: 39.2 Wh (59.9%) condition: 65.4/76.0 Wh (86.0%)
    volts: 15.9 min: 15.9
CPU:
  Info: 8-core model: AMD Ryzen 9 6900HS with Radeon Graphics bits: 64
    type: MT MCP cache: L2: 4 MiB
  Speed (MHz): avg: 1813 min/max: 1600/4934 cores: 1: 3300 2: 1600 3: 1600
    4: 1600 5: 1600 6: 1395 7: 1822 8: 1600 9: 3300 10: 1600 11: 1600 12: 1600
    13: 1600 14: 1600 15: 1600 16: 1600
Graphics:
  Device-1: AMD Navi 23 [Radeon RX 6650 XT] driver: amdgpu v: kernel
  Device-2: AMD Rembrandt [Radeon 680M] driver: amdgpu v: kernel
  Device-3: Quanta USB2.0 HD UVC WebCam type: USB driver: uvcvideo
  Display: wayland server: X.org v: 1.21.1.4 with: Xwayland v: 22.1.3
    compositor: kwin_wayland driver: X: loaded: amdgpu
    unloaded: modesetting,radeon gpu: amdgpu,amdgpu resolution: 1707x1067
  OpenGL: renderer: AMD Radeon RX 6800S (dimgrey_cavefish LLVM 14.0.6 DRM
    3.48 6.0.0-1-MANJARO) v: 4.6 Mesa 22.1.7
Audio:
  Device-1: AMD Navi 21/23 HDMI/DP Audio driver: snd_hda_intel
  Device-2: AMD Rembrandt Radeon High Definition Audio
    driver: snd_hda_intel
  Device-3: AMD ACP/ACP3X/ACP6x Audio Coprocessor driver: snd_pci_acp6x
  Device-4: AMD Family 17h/19h HD Audio driver: snd_hda_intel
  Sound Server-1: ALSA v: k6.0.0-1-MANJARO running: yes
  Sound Server-2: PulseAudio v: 16.1 running: yes
  Sound Server-3: PipeWire v: 0.3.57 running: yes
Network:
  Device-1: MEDIATEK driver: mt7921e
  IF: wlp5s0 state: up mac: b4:8c:9d:5c:77:db
Bluetooth:
  Device-1: IMC Networks Wireless_Device type: USB driver: btusb
  Report: rfkill ID: hci0 state: up address: see --recommends
Drives:
  Local Storage: total: 1.89 TiB used: 118.16 GiB (6.1%)
  ID-1: /dev/mmcblk0 model: SN32G size: 29.72 GiB
  ID-2: /dev/nvme0n1 vendor: Toshiba model: N/A size: 1.86 TiB
Partition:
  ID-1: / size: 891.53 GiB used: 48.19 GiB (5.4%) fs: ext4
    dev: /dev/nvme0n1p5
  ID-2: /boot/efi size: 256 MiB used: 67.8 MiB (26.5%) fs: vfat
    dev: /dev/nvme0n1p1
  ID-3: /home size: 93.13 GiB used: 69.9 GiB (75.1%) fs: ntfs
    dev: /dev/nvme0n1p6
Swap:
  Alert: No swap data was found.
Sensors:
  System Temperatures: cpu: 70.5 C mobo: N/A
  Fan Speeds (RPM): cpu: 0
  GPU: device: amdgpu temp: 66.0 C fan: 0 device: amdgpu temp: 64.0 C
Info:
  Processes: 364 Uptime: 1h 4m Memory: 30.59 GiB used: 4.76 GiB (15.6%)
  Shell: Bash inxi: 3.3.21
```

Vudentz, 2022-10-13:
@Sikeen please add the HCI/btmon traces, also what kernel are you using?

YourFriendCaspian, 2022-10-19:
I'm having the same problem.

OS: Nobara Linux 36 (KDE Plasma) x86_64
Host: 82B1 Lenovo Legion 5 15ARH05H
Kernel: 5.19.15-202.fsync.fc36.x86_64
Intel Corp. AX200 Bluetooth

Vudentz, 2022-10-19:
> I'm having the same problem.
> 
> OS: Nobara Linux 36 (KDE Plasma) x86_64 Host: 82B1 Lenovo Legion 5 15ARH05H Kernel: 5.19.15-202.fsync.fc36.x86_64 Intel Corp. AX200 Bluetooth
> 
> Bluetooth headphones tested: Samsung Buds Plus headphones (2 different sets) Razer Hammerhead TWS
> 
> Edit: My bluetooth keyboard reconnects every time without any issues but headphones have to be removed and then repaired every time.

Could you please collect the logs with btmon?

simgunz, 2022-11-17:
I have a QCY T18 and I'm facing a similar problem. I add a different set of steps to reproduce the issue that may shed some light on the issue.

I suspect that my device can be paired only to two devices, and when I pair the third the issue arises.

STEPS TO REPRODUCE
1. pair linux computer 1
2. successfully connect to linux computer 1
3. (disconnect and reconnect works)
4. pair linux computer 2
5. successfully connect to linux computer 2
6. (disconnect and reconnect works both on computer 1 and computer 2)
7. (I can reboot both computers and still can connect)
8. pair android device
9. successfully connect to android device

OBSERVED RESULT
At this point, either linux computer 1 or 2 fails to connect, and says "connection to device failed".
You have to unpair and pair the device again for it to connect.

Note that I can do the operations in a different order and the result is the same. 
If the android device is paired first, after pairing the two computers and trying to reconnect to the android device what happens is that Android somehow detects the problem and prompts for repairing.

EXPECTED RESULT
the audio device(earbuds) successfully connects

SOFTWARE/OS VERSIONS
Operating System: Manjaro Linux
KDE Plasma Version: 5.26.3
KDE Frameworks Version: 5.99.0
Qt Version: 5.15.7
Kernel Version: 5.15.78-1-MANJARO (64-bit)
Graphics Platform: X11

INXI:
Bluetooth:
  Device-1: Intel Bluetooth wireless interface type: USB driver: btusb v: 0.8
    bus-ID: 1-7:3 chip-ID: 8087:0a2b class-ID: e001
  Report: rfkill ID: hci0 rfk-id: 0 state: up address: see --recommends

LOG:
[bluemon.log](https://github.com/bluez/bluez/files/10028911/bluemon.log)

closed 2023-02-07 by Sikeen