bluez/bluez #416: Headphone automatically powers off when connecting to computer through bluez

Kazuma-chan, 2022-10-14:
I'm using ArcoLinux with kernel version 6.0.1-arch1-1, completely up-to-date.

Whenever I try to connect to the headphone through either `bluetoothctl` or `blueberry-tray` (xfce), it just turns off without warning, so I'm unable to use it at all.

The headphone model is a Multilaser PH308, it has audio feedback for connecting and disconnecting and turning on and off. But it when I try to connect to it, it just turns off (not just disconnect) without any feedback at all.

Everything was working yesterday, apart from some other issues, like having to auto connect not working properly and it turning off once, but working after turning on again.

I'm not sure if it was an update, since I update my system everyday and I had the same problem with previous kernel version.

Other devices work, particularly my airpod knockoff (I think its model is i12 TWS, but I'm really not sure, it's a knockoff after all).

It does work on my smartphone, so I'm fairly certain it's not just a problem with the headphone, but rather a combination or a strange interaction between it and bluez.

(Sorry for the segmented issue, I was a bit confused how to phrase things since I didn't see a template I could use here on github)

Vudentz, 2022-10-14:
@Kazuma-chan check that you don't have too many codecs enabled, it seems some headset get overwhelmed by that (https://github.com/bluez/bluez/issues/411), it seem Arch started to support some Linux specific codecs as well so perhaps that breaks your headset.

Kazuma-chan, 2022-10-14:
How do I do that? I don't really understand what was done in that issue you mentioned to solve it since none of those directories in the last comment exist for me.

Kazuma-chan, 2022-10-14:
I think I managed to make a btmon dump of me trying to connect with `sudo btmon -w /tmp/btmon.dump.txt`:
[btmon.dump.txt](https://github.com/bluez/bluez/files/9781499/btmon.dump.txt)
What should I do now?

Vudentz, 2022-10-14:
@Kazuma-chan looks like there something disrupting the link, there are several instances it is disconnected due to a timeout:

```
> HCI Event: Disconnect Complete (0x05) plen 4                                                                           #277 [hci0] 68.192940
        Status: Success (0x00)
        Handle: 17
        Reason: Connection Timeout (0x08)

```

Perhaps there is something with the antenna, maybe try switching to a different controller? (USB BT dongle)

Kazuma-chan, 2022-10-14:
I don't own one of those and don't think it's worth buying. Like I said, my smartphone can connect just fine to it, so it's clearly not a problem with the headphone.

Also, I just tried again with a different desktop environment (Cinnamon) and it worked fine. Same computer, same headphone, I haven't updated anything yet.

Before logging off and trying with Cinnamon, I tried once again with xfce and the same problem happened.

Kazuma-chan, 2022-10-14:
I tried using the same command to log it working, but it randomly stopped working again even with Cinnamon. I have no idea why it worked once.

Kazuma-chan, 2022-10-14:
I somehow managed to make it work by updating, and and turning it on repeatedly, making it autoconnect multiple times. Here's my successful attempt:
[btmon.dump.txt](https://github.com/bluez/bluez/files/9790510/btmon.dump.txt)

Kazuma-chan, 2022-10-18:
I forgot to update this issue on why this is still open. Turns out that didn't work at all. It seemed like it worked because the headphone showed it was connected and the pc showed it as connected too, but neither the headphone would play anything from the pc, nor the pc would show it in the devices list in the sound driver selection. Sadly it's extremely finicky reproducing that and logging, so I couldn't do it.

closed 2025-08-16 by github-actions[bot]