bluez/bluez #417: MIDI interface is not working

tsutsarin, 2022-10-16:
Version: bluez 5.64-0ubuntu1 compiled with --enable-midi 
OS: Kubuntu 22.04

I connected my Sensel Morph MIDI controller via bluetooth and it shows here:
```
master@asgard:~$ aconnect -i
client 0: 'System' [type=kernel]
    0 'Timer           '
    1 'Announce        '
client 14: 'Midi Through' [type=kernel]
    0 'Midi Through Port-0'
client 128: 'Morph-4065' [type=user,pid=37797]
    0 'Morph-4065 Bluetooth'
```
When I try pressing some keys on controller nothing shows as dump here:

```
master@asgard:~$ aseqdump -p 128:0
Waiting for data. Press Ctrl+C to end.
Source  Event                  Ch  Data
```
Expected: MIDI BLE working with DAW and showing output in aseqdump, like in this example:
```
Waiting for data. Press Ctrl+C to end.                                                                                                                                                                                                                                         
Source  Event                  Ch  Data                                                                                                                                                                                                                                        
 20:0   Note on                 0, note 60, velocity 62                                                                                                                                                                                                                        
 20:0   Note off                0, note 60    
```
Controller was tested in Windows 11 OS and it does work through MIDIberry and loopMIDI tools.

Vudentz, 2022-10-17:
@tsutsarin not what these commands do, does aconnect list devices connected or just paired/bonded? Id probably suggest making sure bluetoothctl is used to connect the device first, also make sure to include the HCI traces using btmon.

dlundqvist, 2023-03-14:
I have the same problem, but with a Roland FP-30X and BlueZ 5.66 with MIDI enabled (Debian bookworm). As for the other poster it works fine in Windows with MIDIberry and loopMIDI. 

Here is a [snoop file](https://www.malarhojden.nu/files/bluez-417/btmon.dump) where I do the following:

- Enable Bluetooth on the piano
- Play a few keys

After Bluetooth is enabled, ALSA client exists:

```
% aconnect -l      
client 0: 'System' [type=kernel]
    0 'Timer           '
        Connecting To: 145:0
    1 'Announce        '
        Connecting To: 129:0, 145:0
client 14: 'Midi Through' [type=kernel]
    0 'Midi Through Port-0'
client 16: 'Virtual Raw MIDI 0-0' [type=kernel,card=0]
    0 'VirMIDI 0-0     '
client 20: 'UltraLite-mk5' [type=kernel,card=1]
    0 'UltraLite-mk5 MIDI 1'
client 128: 'FP-30X MIDI' [type=user,pid=22527]
    0 'FP-30X MIDI Bluetooth'
client 130: 'Virtual Keyboard' [type=user,pid=9564]
    0 'Virtual Keyboard'
client 145: 'PipeWire-System' [type=user,pid=9841]
    0 'input           '
        Connected From: 0:1, 0:0
client 146: 'PipeWire-RT-Event' [type=user,pid=9841]
    0 'input           '
```

As I play the keys nothing appears in btmon nor aseqdump:
```
% aseqdump -p 128:0
Waiting for data. Press Ctrl+C to end.
Source  Event                  Ch  Data
```

What other information can I provide to help debugging this issue?

dlundqvist, 2023-03-15:
Whenever I pair with a MIDI enabled BlueZ this appear in its log:
```
bluetoothd[126792]: profiles/midi/midi.c:handle_characteristic() Failed to obtain characteristic data
bluetoothd[126792]: profiles/midi/midi.c:handle_characteristic() Failed to obtain characteristic data
```
And I also see this in kernel log:
```
[Mar15 17:36] Bluetooth: hci0: unexpected SMP command 0x06 from cb:81:f4:7f:7d:69
[  +0.099516] Bluetooth: hci0: unexpected SMP command 0x07 from cb:81:f4:7f:7d:69
[  +0.099930] Bluetooth: hci0: unexpected SMP command 0x08 from cb:81:f4:7f:7d:69
[  +0.000563] Bluetooth: hci0: unexpected SMP command 0x09 from cb:81:f4:7f:7d:69
[  +0.149364] Bluetooth: hci0: unexpected SMP command 0x0a from cb:81:f4:7f:7d:69
[Mar15 17:55] Bluetooth: hci0: unexpected SMP command 0x06 from cb:81:f4:7f:7d:69
[  +0.000687] Bluetooth: hci0: unexpected SMP command 0x07 from cb:81:f4:7f:7d:69
[  +0.000606] Bluetooth: hci0: unexpected SMP command 0x08 from cb:81:f4:7f:7d:69
[  +0.000614] Bluetooth: hci0: unexpected SMP command 0x09 from cb:81:f4:7f:7d:69
[  +0.000795] Bluetooth: hci0: unexpected SMP command 0x0a from cb:81:f4:7f:7d:69
[ +23.448232] Bluetooth: hci0: unexpected SMP command 0x06 from cb:81:f4:7f:7d:69
[  +0.000714] Bluetooth: hci0: unexpected SMP command 0x07 from cb:81:f4:7f:7d:69
[  +0.000702] Bluetooth: hci0: unexpected SMP command 0x08 from cb:81:f4:7f:7d:69
[  +0.000532] Bluetooth: hci0: unexpected SMP command 0x09 from cb:81:f4:7f:7d:69
[  +0.000750] Bluetooth: hci0: unexpected SMP command 0x0a from cb:81:f4:7f:7d:69
```

closed 2025-08-16 by github-actions[bot]