bluez/bluez #419: Wireless Ear Buds get randomly disconnected.

Corrupter-rot, 2022-10-30:
Basic problem is that my wireless buds get randomly disconnect and then sometime reconnects and sometimes not.
This problem doesn't occur with my Bluetooth Speaker. It only happens when I connect my buds to my laptop.

`sudo journalctl -b -u bluetooth.service` gives the following.

```
Oct 30 19:16:09 Rot bluetoothd[578]: /org/bluez/hci0/dev_FC_58_FA_32_0C_D4/sep1/fd6: fd(30) ready
Oct 30 19:16:36 Rot bluetoothd[578]: /org/bluez/hci0/dev_D8_6C_02_50_B0_1B/fd7: fd(34) ready
Oct 30 19:20:15 Rot bluetoothd[578]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Oct 30 19:20:16 Rot bluetoothd[578]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for D8:6C:02:50:B0:1B: Device or resource busy
Oct 30 19:20:16 Rot bluetoothd[578]: plugins/policy.c:reconnect_timeout() Reconnecting services failed: Device or resource busy (16)
Oct 30 19:20:18 Rot bluetoothd[578]: /org/bluez/hci0/dev_D8_6C_02_50_B0_1B/fd8: fd(29) ready
Oct 30 19:25:26 Rot bluetoothd[578]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Oct 30 19:25:28 Rot bluetoothd[578]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for D8:6C:02:50:B0:1B: Device or resource busy
Oct 30 19:25:28 Rot bluetoothd[578]: plugins/policy.c:reconnect_timeout() Reconnecting services failed: Device or resource busy (16)
Oct 30 19:25:29 Rot bluetoothd[578]: /org/bluez/hci0/dev_D8_6C_02_50_B0_1B/fd9: fd(29) ready
```

Search online but didn't find any useful information or help for my case.
- As for my EarBuds they are **Realme AirBuds 3**
- **Bluetoothctl: 5.65**

If you need any other information, please let me know.
Thanks.

Vudentz, 2022-11-01:
@Corrupter-rot please add the HCI traces, you can collect it using btmon.

Corrupter-rot, 2022-11-02:
Sometime it happens rarely and something every other minute. So it hard for me to catch the random disconnect in the Bluetooth. 

Here is log file of when earbuds were connected and no audio was playing.
[hcihump.log](https://github.com/bluez/bluez/files/9917994/hcihump.log)

But the random disconnect didn't occur during this time.

Corrupter-rot, 2022-11-05:
I was able to catch the random disconnect. Here is the log file for that.
[hcihump2.log](https://github.com/bluez/bluez/files/9943893/hcihump2.log)

Vudentz, 2022-11-07:
```
> HCI Event: Mode Change (0x14) plen 6                                                                                                                                                                                                                               #19091 [hci0] 231.755604
        Status: Success (0x00)
        Handle: 2
        Mode: Sniff (0x02)
        Interval: 500.000 msec (0x0320)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                                                                                                       #19092 [hci0] 253.639344
        Status: Success (0x00)
        Handle: 2
        Reason: Connection Timeout (0x08)
```

So this is down in the link layer, apparently the device is entering sniff mode and then the connection times out, so probably the controller could not get acks for some reason which ends up causing the connection to time out.

Corrupter-rot, 2022-11-12:
Any reason why it does that?

class101, 2022-11-21:
@Vudentz 

Do you think the following error is connected to this ticket ?

```
Nov 21 12:52:45 steamdeck bluetoothd[921]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Nov 21 12:52:45 steamdeck bluetoothd[921]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Nov 21 12:52:45 steamdeck bluetoothd[921]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
```

Seen on Steam Deck, SteamOS 3.3.2, I think they are using a vulnerable kernel `5.13.0-valve21.3-1-neptune` because when I connect the WF100XM4 headphone + gamepad , the gamepad can't disconnects after 2 min and the journal is showing the errors

Unfiortunately, the Valve developers don't responde to my ticket so I've to dig myself to find the responsible parties :(

henil, 2023-02-11:
facing the same issue, any way i can help debug this further?

henil, 2023-02-11:
facing the same issue, any way i can help debug this further?

could be similar issue to https://github.com/bluez/bluez/issues/130.

I am using Realtek MT7922. @Corrupter-rot what wireless card are you using?

henil, 2023-02-11:
facing the same issue, any way i can help debug this further?

could it be similar issue to https://github.com/bluez/bluez/issues/130?

I am using Realtek MT7922. @Corrupter-rot what wireless card are you using?

I get couple different but similar kinds of error from journalctl:

```log
Feb 11 17:15:17 henil bluetoothd[596]: profiles/audio/avdtp.c:cancel_request() Suspend: Connection timed out (110)
Feb 11 17:15:18 henil bluetoothd[596]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Feb 11 17:15:18 henil bluetoothd[596]: profiles/audio/avdtp.c:handle_unanswered_req() No reply to Suspend request
Feb 11 17:15:19 henil bluetoothd[596]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 74:D7:13:00:9F:95: Device or resource busy
Feb 11 17:15:19 henil bluetoothd[596]: plugins/policy.c:reconnect_timeout() Reconnecting services failed: Device or resource busy (16)
Feb 11 17:15:25 bubble bluetoothd[596]: /org/bluez/hci0/dev_74_D7_13_00_9F_95/fd5: fd(40) ready 
```

```log
Feb 11 17:21:49 henil bluetoothd[596]: profiles/audio/avdtp.c:cancel_request() Suspend: Connection timed out (110)
Feb 11 17:21:51 henil bluetoothd[596]: profiles/audio/avdtp.c:handle_unanswered_req() No reply to Suspend request
Feb 11 17:21:51 henil bluetoothd[596]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint
is not connected (107)
Feb 11 17:22:20 henil bluetoothd[596]: /org/bluez/hci0/dev_74_D7_13_00_9F_95/sep2/fd6: fd(42) ready
```

```log
Feb 11 17:39:37 henil bluetoothd[596]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint>
```

Whimfoome, 2023-02-11:
Facing the same problem.

- Fedora 37 with Gnome
- Linux 6.1.9-200.fc37.x86_64
- CPU: 11th Gen Intel® Core™ i5-1135G7 × 8
- Device: realme buds air 2
- I think I am running with pipewire
- bluez 5.65

Logs from `journalctl | grep -i “blue*” `:
```
окт 30 14:10:41 fedora bluetoothd[782]: profiles/audio/avdtp.c:cancel_request() Suspend: Connection timed out (110)
окт 30 14:10:42 fedora bluetoothd[782]: profiles/audio/avdtp.c:handle_unanswered_req() No reply to Suspend request
окт 30 14:10:42 fedora bluetoothd[782]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
окт 30 14:10:44 fedora bluetoothd[782]: plugins/policy.c:reconnect_timeout() Reconnecting services failed: Operation already in progress (114)
```

henil, 2023-02-21:
might be related to this bug:

recently when i using buds some state got messed up and bluetooth started disconnecting very frequently and wasn't able to play any audio.

So to reduce surface area to debug, I:
- Disabled Dual-device connection
- Disconnected other connected device (still the same)
- Buds buds case and closed the lid so it gets disconnected completely and tried to reconnect (still the same)
- Tried restarting `bluetooth.service` and `wireplumber` (still the same)
- Keep buds in case with lid closed and restarted `bluetooth.service` and `wireplumber` and then tried to reconnect (still the same)
- Restarted computer (worked now)

So looks like some temporary state got corrupted, which isn't able to fix itself by restarting bluetooth related services.

henil, 2023-02-21:
After restart (with Dual-device connection disabled and buds connected to laptop only), I got a disconnection **exactly 5 minute** after the connection.

See 16:13:47 -> 16:18:47
<details>
<summary>journalctl -xe -u bluetooth</summary>

```
Feb 21 16:12:30 henil systemd[1]: Starting Bluetooth service...
░░ Subject: A start job for unit bluetooth.service has begun execution
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░
░░ A start job for unit bluetooth.service has begun execution.
░░
░░ The job identifier is 159.
Feb 21 16:12:30 henil bluetoothd[592]: Bluetooth daemon 5.66
Feb 21 16:12:30 henil bluetoothd[592]: Starting SDP server
Feb 21 16:12:30 henil bluetoothd[592]: profiles/audio/vcp.c:vcp_init() D-Bus experimental not enabled
Feb 21 16:12:30 henil bluetoothd[592]: src/plugin.c:plugin_init() Failed to init vcp plugin
Feb 21 16:12:30 henil bluetoothd[592]: profiles/audio/mcp.c:mcp_init() D-Bus experimental not enabled
Feb 21 16:12:30 henil bluetoothd[592]: src/plugin.c:plugin_init() Failed to init mcp plugin
Feb 21 16:12:30 henil bluetoothd[592]: profiles/audio/bap.c:bap_init() D-Bus experimental not enabled
Feb 21 16:12:30 henil bluetoothd[592]: src/plugin.c:plugin_init() Failed to init bap plugin
Feb 21 16:12:30 henil systemd[1]: Started Bluetooth service.
░░ Subject: A start job for unit bluetooth.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░
░░ A start job for unit bluetooth.service has finished successfully.
░░
░░ The job identifier is 159.
Feb 21 16:12:30 henil bluetoothd[592]: Bluetooth management interface 1.22 initialized
Feb 21 16:12:37 henil bluetoothd[592]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSink/aac
Feb 21 16:12:37 henil bluetoothd[592]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/aac
Feb 21 16:13:47 henil bluetoothd[592]: /org/bluez/hci0/dev_74_D7_13_00_72_5C/sep2/fd0: fd(41) ready
Feb 21 16:18:47 henil bluetoothd[592]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint>
Feb 21 16:18:49 henil bluetoothd[592]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 74:D7:13:00:72:5C: Device or resource busy
Feb 21 16:18:49 henil bluetoothd[592]: plugins/policy.c:reconnect_timeout() Reconnecting services failed: Device or resource busy (16)
Feb 21 16:18:50 henil bluetoothd[592]: /org/bluez/hci0/dev_74_D7_13_00_72_5C/fd1: fd(40) ready
```

</details>

Here is  [HCI traces](https://github.com/bluez/bluez/files/10792846/hci.log)  of that time.

I don't understand HCI protocol but from my initial look at logs unlike @Corrupter-rot's case doesn't look like disconnection was due to device going in `Sniff` mode. Any ideas @Vudentz ?

Corrupter-rot, 2023-02-23:
@henil I think my bluetooth and wireless card are integrated in one, therefore the card is `Realtek RTL8822CE`.
Also I think the problem doesn't occur when using Pulseaudio. It only happens when using Pipewire.

henil, 2023-02-23:
> @henil I think my bluetooth and wireless card are integrated in one, therefore the card is `Realtek RTL8822CE`. Also I think the problem doesn't occur when using Pulseaudio. It only happens when using Pipewire.

My bluetooth and wireless card are integrated as well, I am on `Realtek MT7922`. Don't think that effects much.

I will try using pulseaudio instead of pipewire and report back.

pv, 2023-02-23:
> I will try using pulseaudio instead of pipewire and report back.

Make sure to use the same audio codec in the comparisons, in case it depends on that.

henil, 2023-03-03:
I tried to switch from pipewire to pulseaudio on my machine but it was breaking a lot of other things so i couldn't test it properly. 

And I switched back to pipewire (with wireplumber), I tried using `SBC` bluetooth codec but still experienced disconnects, although with a different error message this time (see around timestamp 17:42).

<details>
<summary>journalctl -xe -u bluetooth</summary>

```
░░ The job identifier is 159.
Mar 03 15:10:03 henil bluetoothd[519]: Bluetooth daemon 5.66
Mar 03 15:10:03 henil bluetoothd[519]: Starting SDP server
Mar 03 15:10:03 henil bluetoothd[519]: profiles/audio/vcp.c:vcp_init() D-Bus experimental not enabled
Mar 03 15:10:03 henil bluetoothd[519]: src/plugin.c:plugin_init() Failed to init vcp plugin
Mar 03 15:10:03 henil bluetoothd[519]: profiles/audio/mcp.c:mcp_init() D-Bus experimental not enabled
Mar 03 15:10:03 henil bluetoothd[519]: src/plugin.c:plugin_init() Failed to init mcp plugin
Mar 03 15:10:03 henil bluetoothd[519]: profiles/audio/bap.c:bap_init() D-Bus experimental not enabled
Mar 03 15:10:03 henil bluetoothd[519]: src/plugin.c:plugin_init() Failed to init bap plugin
Mar 03 15:10:03 henil systemd[1]: Started Bluetooth service.
░░ Subject: A start job for unit bluetooth.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░
░░ A start job for unit bluetooth.service has finished successfully.
░░
░░ The job identifier is 159.
Mar 03 15:10:03 henil bluetoothd[519]: Bluetooth management interface 1.22 initialized
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/ldac
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSink/aptx_hd
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/aptx_hd
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSink/aptx
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/aptx
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSink/aac
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/aac
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSink/sbc
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/sbc
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSink/sbc_xq
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/sbc_xq
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/faststream
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/faststream_duplex
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSink/opus_05
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/opus_05
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSink/opus_05_duplex
Mar 03 15:10:13 henil bluetoothd[519]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/opus_05_duplex
Mar 03 15:10:17 henil bluetoothd[519]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
Mar 03 17:23:11 henil bluetoothd[519]: /org/bluez/hci0/dev_74_D7_13_00_72_5C/sep1/fd0: fd(41) ready
Mar 03 17:42:16 henil bluetoothd[519]: profiles/audio/avdtp.c:handle_unanswered_req() No reply to Close request
Mar 03 17:42:24 henil bluetoothd[519]: /org/bluez/hci0/dev_74_D7_13_00_72_5C/sep1/fd2: fd(40) ready
```
</details>

Anyone else able to use it properly on pulseaudio and not on pipewire?

henil, 2023-03-12:
> I tried to switch from pipewire to pulseaudio on my machine but it was breaking a lot of other things so i couldn't test it properly.

Okay so somehow i was able to setup pulseaudio so i could test properly (although it has other unrelated issue so i cannot use it as daily driver), and I can confirm I am not seeing those reconnections while using pulseaudio @pv.

One thing to note is that i tried exclusively using `SBC` codec on pipewire as well but issue still persisted, so this is not related to codec as well.

pv, 2023-03-12:
In your btmon trace above, the disconnect is apparently due to remote device stopping responding. Since there's no access to software running on the device, it's just blind guessing now to see what is it that triggers the device to do that.

The next step would be to make sure the SBC settings are identical, they are generally not starting from frequency etc. You should recompile pipewire from this branch https://gitlab.freedesktop.org/pvir/pipewire/-/tree/sbc-settings see pipewire wiki Troubleshooting for how to do it without overriding system installation.

pv, 2023-03-15:
> pulseaudio-working.btsnoop.log, pipewire-0.3.67-working.btsnoop.log, pipewire-sbc-settings-working.btsnoop.log

Don't see much difference in these. Updated the sbc-settings branch again, this time disabled also delay reporting which was one remaining difference, maybe you can try to see if it changes something.

henil, 2023-03-17:
Unfortunately still experiencing the problem. Though I was able to catch the disconnection with the lastest `sbc-settings` branch.

journalctl error was:
> Mar 17 22:48:32 henil bluetoothd[556]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint>

Because I knew dump could have gotten very large (first disconnection happened after 5 hour) i tried to split of the initial handshake from other data. What I did is started the dump connected the device play something for 5 seconds and pause and let all the communication complete.

Then basically restart the btmon in few intervals where everything is going smooth, but dump still got around 56 MB (so attaching a google drive link instead)... The relevant part should be at very last. Also I would like to say thank you for taking your time to debug this!

initial dump: [pipewire-sbc-settings-latest.btsnoop.log](https://github.com/bluez/bluez/files/11004060/pipewire-sbc-new-working.btsnoop.log)
dump with disconnection: [pipewire-sbc-settings-latest-data.btsnoop.log](https://drive.google.com/file/d/1sxQi5OyrVdw4jWy5smsRy69GjwDPrktT/view?usp=share_link)

henil, 2023-03-17:
Also frequency of disconnection increases by alot after the first disconnection (now happening every 15-20 mins)

henil, 2023-03-18:
Actually one interesting thing to note is that, after the initial disconnection i am seeing subsequent disconnections after roughly **multiple of 5 mins** (mostly after 5 minutes but can happen occasionally after 10 minutes) journalctl log. 

I was using them by only connecting them to laptop. Once I connected my mobile (realme buds support dual device pairing) at around `3:25` and next disconnection happened after 15 or so mintues. 

Also when disconnections happens with mobile connected both mobile and laptops bluetooth get disconnected not laptop alone.

Which leads me to believe this might be a bug in liveliness check between the devices (from quick search it seems it is called `Supervision Timeout`). And when something unexpected happens device just decide to restart which leads to even mobile's bluetooth connection getting resetted.

Does pipewire differ in anyway to pulseaudio in matter of liveliness checks of devices @pv ?

<details>
<summary> journalctl -xe -u bluetooth </summary>

```log
Mar 18 12:04:16 henil systemd[1]: Starting Bluetooth service...
░░ Subject: A start job for unit bluetooth.service has begun execution
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit bluetooth.service has begun execution.
░░ 
░░ The job identifier is 159.
Mar 18 12:04:16 henil bluetoothd[507]: Bluetooth daemon 5.66
Mar 18 12:04:16 henil bluetoothd[507]: Starting SDP server
Mar 18 12:04:16 henil bluetoothd[507]: profiles/audio/vcp.c:vcp_init() D-Bus experimental not enabled
Mar 18 12:04:16 henil bluetoothd[507]: src/plugin.c:plugin_init() Failed to init vcp plugin
Mar 18 12:04:16 henil bluetoothd[507]: profiles/audio/mcp.c:mcp_init() D-Bus experimental not enabled
Mar 18 12:04:16 henil bluetoothd[507]: src/plugin.c:plugin_init() Failed to init mcp plugin
Mar 18 12:04:16 henil bluetoothd[507]: profiles/audio/bap.c:bap_init() D-Bus experimental not enabled
Mar 18 12:04:16 henil bluetoothd[507]: src/plugin.c:plugin_init() Failed to init bap plugin
Mar 18 12:04:16 henil systemd[1]: Started Bluetooth service.
░░ Subject: A start job for unit bluetooth.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit bluetooth.service has finished successfully.
░░ 
░░ The job identifier is 159.
Mar 18 12:04:16 henil bluetoothd[507]: Bluetooth management interface 1.22 initialized
Mar 18 12:04:27 henil bluetoothd[507]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSink/sbc
Mar 18 12:04:27 henil bluetoothd[507]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/sbc
Mar 18 12:32:05 henil bluetoothd[507]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Mar 18 12:53:16 henil bluetoothd[507]: /org/bluez/hci0/dev_74_D7_13_00_72_5C/fd1: fd(40) ready
Mar 18 16:00:14 henil bluetoothd[507]: profiles/audio/avdtp.c:handle_unanswered_req() No reply to Close request
Mar 18 16:00:19 henil bluetoothd[507]: /org/bluez/hci0/dev_74_D7_13_00_72_5C/sep1/fd2: fd(41) ready
Mar 18 17:03:07 henil bluetoothd[507]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Mar 18 17:03:16 henil bluetoothd[507]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Mar 18 17:03:26 henil bluetoothd[507]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Mar 18 17:03:38 henil bluetoothd[507]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Mar 18 17:03:54 henil bluetoothd[507]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Mar 18 17:04:18 henil bluetoothd[507]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Mar 18 17:04:58 henil bluetoothd[507]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Mar 18 17:06:10 henil bluetoothd[507]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Mar 18 17:33:04 henil bluetoothd[507]: /org/bluez/hci0/dev_74_D7_13_00_72_5C/fd3: fd(40) ready
Mar 18 17:33:28 henil bluetoothd[507]: /org/bluez/hci0/dev_74_D7_13_00_72_5C/sep1/fd4: fd(41) ready
Mar 18 17:33:49 henil bluetoothd[507]: /org/bluez/hci0/dev_74_D7_13_00_72_5C/sep1/fd5: fd(41) ready
Mar 18 17:34:10 henil bluetoothd[507]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Mar 18 17:34:21 henil bluetoothd[507]: /org/bluez/hci0/dev_74_D7_13_00_72_5C/fd6: fd(40) ready
Mar 18 18:27:50 henil bluetoothd[507]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Mar 18 18:51:30 henil bluetoothd[507]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Mar 18 20:42:53 henil bluetoothd[507]: /org/bluez/hci0/dev_74_D7_13_00_72_5C/sep1/fd8: fd(41) ready
Mar 18 22:53:13 henil bluetoothd[507]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
[... cut ...]
Mar 19 03:38:55 henil bluetoothd[507]: profiles/audio/avdtp.c:handle_unanswered_req() No reply to Suspend request
Mar 19 03:38:55 henil bluetoothd[507]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Mar 19 03:38:57 henil bluetoothd[507]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Operation already in progress
Mar 19 03:38:59 henil bluetoothd[507]: plugins/policy.c:reconnect_timeout() Reconnecting services failed: Operation already in progress (114)
```
</details>

pv, 2023-03-18:
> Does pipewire differ in anyway to pulseaudio in matter of liveliness checks of devices

I don't think either of them does something related to this, it is lower level thing. PA/PW only talk to the device via bluez media API and kernel socket for audio data. I think the issue here is some indirect consequence. E.g. it may well be the issue is not reproducible with a different bluetooth adapter.

To reduce number of variables, you should also disable HFP/HSP in pipewire, create file e.g. named `~/.config/wireplumber/bluetooth.lua.d/51-hfp.lua` containing the line `bluez_monitor.properties["bluez5.hfphsp-backend"] = "none"`.

henil, 2023-04-06:
Okay, I can confirm after removing that fix I am again seeing those disconnections. I have also confirmed this with one of my friend who had similar issues and disabling `hfphsp-backend` mitigates the issue. 

Pulseaudio also does support hfp backend but I haven't experience disconnections on it (i mostly use a2dp on both pipewire and pulseaudio), so the bug could be in how those backends are handled in pipewire?

CC: @pv

henil, 2023-04-06:
Okay, I can confirm after removing that fix I am again seeing those disconnections. I have also confirmed this with one of my friend who had similar issues and disabling `hfphsp-backend` mitigates the issue. 

CC: @pv

pv, 2023-04-06:
Possibly the device does not like some of the HFP commands we send via RFCOMM. So one should go through the PA and PW btmon dumps, look through what goes differently in the RFCOMM part, and change things in PW one by one to be the same, to determine what is the difference the device does not like.

henil, 2023-04-07:
Does pipewire send HFP commands even when just using A2DP on SBC?

MASHtm, 2023-04-07:
I can confirm that with disabled bluez5.hfphsp-backend I have no disconnects on both my Laptop and Desktop PC as well.
I used my "earfun free pro" headphones as before with SBC-XQ Audio profile, because this triggered the disconnects in the shortest time (about 5-10 minutes).  I also reconfirmed that the disconnects happen again as soon as I reactivate hfphsp-backend.

My Laptop is a
Lenovo Carbon X1 G9 (latest bios/firmware) with Intel AX201
On my Desktop I use a 
7392:c611 Edimax Technology Co., Ltd Edimax Bluetooth Adapter
with rtl8761bu

Both run on Fedora 37 with
bluez-5.66-4.fc37.x86_64
pipewire-0.3.67-1.fc37.x86_64
wireplumber-0.4.14-1.fc37.x86_64

pv, 2023-04-07:
Devices use HFP for e.g. battery status reporting.

We seem to not be replying OK to AT+IPHONEACCEV. Try this: https://gitlab.freedesktop.org/pvir/pipewire/-/tree/hfp-ok

MASHtm, 2023-04-07:
I rebuilt the f37 SRPM with your patch applied and got no disconnects for about an hour now with hfphsp-backend reenabled.

Whimfoome, 2023-04-17:
Since updating pipewire to version 0.3.69 I no longer experience this issue as well

henil, 2023-04-18:
for some reason i am seeing it again, after i cleaned up development build install and installed 0.3.69 from official repo

K4R7IK, 2023-05-21:
After updating to 0.3.69 I don't experience random disconnects anymore. Thanks for fixing it.

closed 2023-05-21 by K4R7IK

musakarimli, 2024-04-24:
guys, maybe someone gets this issue again? I have this problem and I can't it figure it out what could be issue, I tried many thing updating bluez (5.75), updating pipewire (1.0.3), playing with bluez config and etc. But it is still getting disconnecting in handset profile every 3 minutes. I am on ubuntu 22.04.4 and kernel is 6.5.0-28-generic. My bluetooth adapter is Intel Corp. AX201 Bluetooth. I think it is not appropriate place to post on but maybe some of you could point to the right direction. Thanks in advance

Those are important logs (I think) collected by me:

```
wireplumber[750973]: Failed to release transport /org/bluez/hci0/
dbus-daemon[885]: [system] Rejected send message, 0 matched rules; type="method_return", 
gsd-media-keys[4778]: gvc_mixer_card_get_index: assertion 'GVC_IS_MIXER_CARD (card)' failed

src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected 
src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
wireplumber[2634]: Failed to release transport /org/bluez/hci0//sep2/fd6: Method "Release" with signature "" on interface "org.bluez.MediaTransport1" doesn't exist

Failed to release transport: Method "Release" with signature "" on interface "org.bluez.MediaTransport1" doesn't exist

bluetoothd[7441]: src/profile.c:ext_io_disconnected() Unable to get io data for GalaxyBudsClient: getpeername: Transport endpoint is not connected (107)
bluetoothd[7441]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)

  69 69 69 69 69 69 78                             iiiiiix         
> HCI Event: Disconnect Complete (0x05) plen 4                #2201 [hci0] 24.260540
        Status: Success (0x00)
        Handle: 256
        Reason: Remote User Terminated Connection (0x13)
@ MGMT Event: Device Disconnected (0x000c) plen 8          {0x0001} [hci0] 24.260646
        BR/EDR Address: E8:7F:6B:0B:09:C2 (Samsung Electronics Co.,Ltd)
        Reason: Connection terminated by remote host (0x03)
= bluetoothd: src/profile.c:ext_io_disconnected() Unable to get io dat..   24.299345

 
```