bluez/bluez #420: Failed to stop discovery: org.bluez.Error.Failed

JonasDaWi, 2022-11-01:
in bluetoothctl, when I type
` [bluetooth]# scan off `
I get the error:
`Failed to stop discovery: org.bluez.Error.Failed`

How do I stop scanning?

Vudentz, 2022-11-01:
@JonasDaWi Is it scanning in the first place? Note that if there is another application active scanning you can't stop its session, only the sessions created by its own client instance can be stopped like that, also if the current scanning is passive then you should leave it be because that means some devices are in auto-connect mode.

gadekula1, 2025-01-21:
> The same problem.
> 
> Can't `scan off` when graphical client is off or is on:
> 
> bluetoothctl 
> Agent registered
> [CHG] Controller %EDITED% Pairable: yes
> [bluetooth]# scan off
> Failed to stop discovery: org.bluez.Error.Failed
> But when graphical client is on I can see new devices, trust device. Can't pair via terminal.

@Justuser3310 by any changes Mesh enabled and already started passive scanning ?

justuser-31, 2025-02-15:
@gadekula1 , I don't know, don't see any passive scanning in bluetoothctl.
When I turned on graphical client on:
```
[CHG] Controller %EDITED% Class: 0x006e010c
[CHG] Controller %EDITED% Powered: yes
```
And it's all, no "new device found..."

With user's rights all should be okay, at the least I see this artifact not only in one distro.

Filip3Kx, 2025-04-10:
Don't know if relevant but for me the problem was caused by Home Assistant using the same bluetooth controller and performing the scans. Turning off the entity in HA fixed it.

In summary some external process maybe performing the scans and you can't manage that through `bluetoothctl`. You have to do it from withing the service that's using the controller itself

closed 2025-08-15 by github-actions[bot]

hakanb05, 2025-10-20:
You can simply first
`quit`

Then, restart your bluetooth by running the following command:
`sudo systemctl restart bluetooth`

Now it should have been turned off, run
`bluetoothctl`
and check it for yourself