bluez/bluez #421: Track position not in sync with phone

OliW2, 2022-11-02:
[Old issue which explains my current issue, apart from in also affects androids now](https://github.com/bluez/bluez/issues/232)

When I restart a song more than once on a Samsung S8+/Samsung S22/iPhone X (iPhones have a whole host of time sync issues) all fail to correctly reset the position of the track to the beginning. Is there a way to manually poll "PropertiesChanged" every 50ms or so? I know this isn't technically the correct solution as @Vudentz pointed out, but I kind of need this functionality. Being able to use iPhones at all without time sync issues would great.

Vudentz, 2022-11-02:
Like Ive commented on https://github.com/bluez/bluez/issues/232#issuecomment-965742045, we could in theory resync if client attempt to read Position property, that still need to be implemented though, are you sure this is not application/player specific problem though?

Btw, Im sort of surprise by requiring to sync at ms granularity, Ive never seem any player that would report sub-second progress, in fact the process of reading the status via AVRCP_Get_Status may take more than 50ms.

OliW2, 2022-11-03:
> Like Ive commented on https://github.com/bluez/bluez/issues/232#issuecomment-965742045, we could in theory resync if client attempt to read Position property, that still need to be implemented though, are you sure this is not application/player specific problem though?

Do you have a rough idea on how to resync the position? It's definitely not an application/player specific problem because it happens when I read the Bluez messages directly from the dbus with a very basic 20 line Python application for setting up the connection (Also tried it setting up the connection manually/ Another bluetooth application designed for making your PC into a speaker using BlueZ). When I get all the properties for "org.bluez.MediaPlayer1" the values seem correct, however when I restart the song on Android (or do anything on iPhone like skipping to a different time on the phone) then the position carries on incrementing and not reset/jumping.

> Btw, Im sort of surprise by requiring to sync at ms granularity, Ive never seem any player that would report sub-second progress, in fact the process of reading the status via AVRCP_Get_Status may take more than 50ms.

So that might be overkill, but having it update fast enough so that when someone resets or jumps to a new position it looks instantaneous is the best outcome, but anything other than

OliW2, 2022-11-03:
After playing around a bit with the source code, I can see how the time is update now just using timedelta in the funcion "media_player_get_position" in profiles/audio/player.c. I was also able to add another dbus method to allow for it to be called instead of every time "get_position" is called and that propagates through from player.c to avrcp.c correctly, however with either method I was unable to actually work out how the resync should work?

OliW2, 2022-11-03:
Okay, just got back to this now and added a hotfix that works. Every time I call my dbus method it calls "avrcp_get_play_status" and this fixes the position, I think this works for what I want atm

closed 2025-08-15 by github-actions[bot]