bluez/bluez #422: bluetoothctl --timeout 5 connect <MAC> always returns success

macmpi, 2022-11-03:
I need to connect with a timeout, as default timeout seems quite long (about 30sec).

Doing some tests (i.e. trying to connect to a registered paired devices, now turned off) shows `bluetoothctl` returns success even if **timeout** is reached with built-in option:
```
~$ if bluetoothctl --timeout 5 connect <MAC>; then echo hello; fi
Attempting to connect to <MAC>
hello
```
Using timeout on shell-side does same:
```
~$ if timeout 5 bluetoothctl connect <MAC>; then echo hello; fi
Attempting to connect to <MAC>
hello
```
Only doing it the "brutal" way works as expected:
```
~$ if timeout -s KILL 5 bluetoothctl connect <MAC>; then echo hello; fi
Attempting to connect to <MAC>
Killed
```
Shouldn't built-in --timeout option affect return status if action is not successful?
(this is with bluez 5.65)

Vudentz, 2022-11-03:
> I need to connect with a timeout, as default timeout seems quite long (about 30sec).

In case of LE we need to scan the device before connecting.

> Doing some tests (i.e. trying to connect to a registered paired devices, now turned off) shows `bluetoothctl` returns success even if **timeout** is reached with built-in option:
>
> ```
> ~$ if [...]

This is sort of undefined currently since it depends on the command, for instance --timeout was originally intended for scan command in which case it would be fine to interrupt after timeout which don't set any error, that said it should be possible to enable a timeout with:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/client/main.c#n2058
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/gdbus/client.c#n1009
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/gdbus/client.c#n23

We probably need to introduce a function that takes a timeout instead of using METHOD_CALL_TIMEOUT.

macmpi, 2022-11-03:
Thanks for the prompt feedback
> In case of LE we need to scan the device before connecting.

In that specific case, it is std audio connection to headset/speaker (SBC or AAC).

Vudentz, 2022-11-04:
> Thanks for the prompt feedback
> 
> > In case of LE we need to scan the device before connecting.
> 
> In that specific case, it is std audio connection to headset/speaker (SBC or AAC).

For classic/BREDR it is most likely the page timeout that controls the maximum timeout, that said the 30 seconds could be due to some other reason but in that case we need more logs to be able to tell where exactly is coming from, anyway I don't think we want to hardcode it to 5 or anything like that thus it probably needs to be handled in the client if it doesn't want to wait until the procedure is completed.

macmpi, 2022-11-04:
> thus it probably needs to be handled in the client if it doesn't want to wait until the procedure is completed.

sure.
a working  **--timeout** option would really help in such usecase, particularly as using the external timeout command:
```
~$ if timeout -s KILL 5 bluetoothctl connect <MAC>; then echo hello; fi
Attempting to connect to <MAC>
Killed
```
is not fully satisfying since subsequent similar call will fail (`Failed to connect: org.bluez.Error.Failed Operation already in progress`) until the first call internal timeout (30sec?) is exhausted...

Easy to replicate (just trying to connect to a powered-off paired-device), but let me know which log you may want if you need some data-points.

macmpi, 2022-11-04:
It seems like your previous comment has been repeated twice 🤔

macmpi, 2022-12-09:
> We probably need to introduce a function that takes a timeout instead of using METHOD_CALL_TIMEOUT.

any plan to improve such timeout handling for next release?
Thanks for consideration.

closed 2025-08-15 by github-actions[bot]