bluez/bluez #424: headless controller cannot re-pair to Bluetooth

Phlip, 2022-11-14:
Try this experiment with a cheap Bluetooth speaker with no user interface:

- pair your tablet with the Bluetooth speaker
- turn the speaker off
- unpair the tablet from the speaker
- turn the speaker on
- pair the tablet successfully with the speaker

Now try the same experiment with a headless controller running Bluez. If inside the controller (such as a Raspberry Pi) you use hciconfig hci0 sspmode 1 then the controller will automatically pair with the tablet's Bluetooth.

But if you then turn off the controller, un-pair the tablet, and turn the controller back on, it will now refuse to pair. The tablet shows the controller in its 'advertising' list, eligible for pairing. But inside the controller, the tablet appears in its 'paired' list. The working theory is the controller thinks the tablet is paired and hence refuses to pair again.

How do cheap headless Bluetooth speakers get around this problem? Opening up the controller with SSH or a keyboard are not optional.

Phlip, 2022-11-14:
Raspberry Pis come with Bluez 5.50. I built the current master version and got 5.65, and it still has the bug.

bojanpotocnik, 2022-11-29:
Might be related to https://github.com/bluez/bluez/issues/433

Phlip, 2022-11-29:
I built Bluez and added the line device_remove_stored(device); (and `sudo make install` didn't copy `bluetoothd` into `/usr/lib/bluetooth/bluetoothd` so I did that manually)

and I still can't re-pair. Android just says 'Couldn't pair. Check settings for this device and try again.'

Vudentz, 2022-12-16:
@Phlip you are talking about this patch:

```
diff --git a/src/device.c b/src/device.c
index 995d39f2c..a41370ade 100644
--- a/src/device.c
+++ b/src/device.c
@@ -6048,6 +6048,9 @@ void device_bonding_complete(struct btd_device *device, uint8_t bdaddr_type,
                        btd_device_set_temporary(device, true);
 
                device_bonding_failed(device, status);
+
+               device_remove_stored(device); /* or device_remove(device, TRUE); */
+
                return;
        }
```

We never applied this upstream, our recommendation was to remove the device from upper layer, the above call is not exactly what we would do in case of remove_device.

Phlip, 2022-12-16:
but ya that's the patch that apparently doesn't work in my case

bojanpotocnik, 2024-01-08:
@Phlip have you managed to solve your issue somehow?

Phlip, 2024-04-04:
it turns out I needed to upgrade to the latest Bluez and then enable the line in the Bluez configuration "JustWorksRepairing." The confusion was because Raspberry Pi thwarted my every attempt to upgrade Bluez, such as by putting it in the wrong folder etc

closed 2025-08-15 by github-actions[bot]