bluez/bluez #427: Stopped working with Intel AX200 on Arch Linux

ad-on-is, 2022-11-19:
As of a recent updates, Bluetooth has stopped working, with `Version         : 5.66-1`

`bt-adapter -l` => empty

`bt-adapter -i`
```
**
ERROR:lib/bluez/adapter.c:165:adapter_get_dbus_object_path: assertion failed: (ADAPTER_IS(self))
Bail out! ERROR:lib/bluez/adapter.c:165:adapter_get_dbus_object_path: assertion failed: (ADAPTER_IS(self))
zsh: IOT instruction (core dumped)  bt-adapter -i
```
Starting `blueberry` or `bluetooth-manager` results in a similar error like

```
Could not retrieve the BT adapter name with 'bt-adapter -i': Command '['timeout', '2s', 'bt-adapter', '-i']' died with <Signals.SIGABRT: 6>.

```

Am I missing something?

endrift, 2023-01-28:
Anything in dmesg?

tebicap, 2023-05-11:
same issue here, (Kubuntu 23.10)

 ≫bt-adapter -l
No adapters found

 ≫bt-adapter -i
**
ERROR:lib/bluez/adapter.c:165:adapter_get_dbus_object_path: assertion failed: (ADAPTER_IS(self))
Bail out! ERROR:lib/bluez/adapter.c:165:adapter_get_dbus_object_path: assertion failed: (ADAPTER_IS(self))
Abortado (`core' generado)

----

Operating System: Kubuntu 23.04
KDE Plasma Version: 5.27.4
KDE Frameworks Version: 5.104.0
Qt Version: 5.15.8
Kernel Version: 6.2.0-20-generic (64-bit)
Graphics Platform: X11
Processors: 4 × Intel® Core™ i3-3240 CPU @ 3.40GHz
Memory: 5.7 GiB of RAM
Graphics Processor: NVD9