bluez/bluez #428: Failed to connect after forgetting while using NoInputNoOutput with bluetooth agent

AravindGopala, 2022-11-19:
System is running below:
Bluez_version: 5.64
Kernel version: 5.15

In this case let's call the system that's having the issue is called DUT

Steps to reproduce:

1. Set the bluetooth agent capability NoInputNoOutput for the default-agent on the DUT using bluetoothctl
2. Take a phone turn on bluetooth and search for the DUT 
3. No once it shows up in the phone BT search menu, click on the DUT so that pairing is initiated
4. Now the phone is connected to the DUT without issues and no passcode is displayed as NoInputNoOutput is selected on the DUT in step 1
5. Now forget the DUT from the phone
6. Now turn of BT on the phone, turn it back on 
7. Now wait for the DUT to show up in the phone BT menu
8. Now try to connect

The connection fails --> Issue

The hcidump when the connect failed is as below:
```
> ACL data: handle 11 flags 0x02 dlen 12
    L2CAP(s): Disconn req: dcid 0x0040 scid 0x0408
< ACL data: handle 11 flags 0x00 dlen 12
    L2CAP(s): Disconn rsp: dcid 0x0040 scid 0x0408
> HCI Event: Link Supervision Timeout Change (0x38) plen 4
    handle 11 timeout 32000
> HCI Event: IO Capability Response (0x32) plen 9
    bdaddr 14:F2:87:BA:90:94 capability 0x01 oob 0x00 auth 0x04
    Capability: DisplayYesNo (OOB data not present)
    Authentication: General Bonding (No MITM Protection)
> HCI Event: IO Capability Request (0x31) plen 6
    bdaddr 14:F2:87:BA:90:94
< HCI Command: IO Capability Request Reply (0x01|0x002b) plen 9
    bdaddr 14:F2:87:BA:90:94 capability 0x03 oob 0x00 auth 0x04
    Capability: NoInputNoOutput (OOB data not present)
    Authentication: General Bonding (No MITM Protection)
> HCI Event: Command Complete (0x0e) plen 10
    IO Capability Request Reply (0x01|0x002b) ncmd 1
    status 0x00 bdaddr 14:F2:87:BA:90:94
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 11 packets 1
> HCI Event: User Confirmation Request (0x33) plen 10
    bdaddr 14:F2:87:BA:90:94 passkey 713305
< HCI Command: User Confirmation Request Negative Reply (0x01|0x002d) plen 6
    bdaddr 14:F2:87:BA:90:94
> HCI Event: Command Complete (0x0e) plen 10
    User Confirmation Request Negative Reply (0x01|0x002d) ncmd 1
    status 0x00 bdaddr 14:F2:87:BA:90:94
> HCI Event: Simple Pairing Complete (0x36) plen 7
    status 0x05 bdaddr 14:F2:87:BA:90:94
    Error: Authentication Failure
```

Piocky, 2023-02-06:
Hi @AravindGopala, I'm "happy" to see that I'm not the only one facing this exact same issue.. I'm working on a project with custom Yocto image on a Raspberry Pi, but I just discovered that it's behaving the same way on my Raspbian dev Pi. Did you find any solution to this problem?

AravindGopala, 2023-02-06:
@Piocky yes, the solution is to use `JustWorksRepairing = always` in the `etc/bluetooth/main.conf`, this looks like a regression in the latest kernel after 4.19 and we need the JustWorksRepairing to make this to work.

FYI, More discussion can be found [here](https://github.com/bluez/bluez/issues/93)

Related Patch discussions:

['[PATCH 1/1] Bluetooth: Fix Just-Works re-pairing' - MARC](https://marc.info/?l=linux-bluetooth&m=161262095811187&w=2) 

['[PATCH 0/1] Bluetooth: Fix Just-Works re-pairing' - MARC](https://marc.info/?l=linux-bluetooth&m=161262093311177&w=2)

closed 2023-02-06 by AravindGopala

Piocky, 2023-02-06:
@AravindGopala you just made my day!! Thank you so much!

AravindGopala, 2023-02-06:
@Piocky you are welcome, glad that resolved the issue.