bluez/bluez #430: BLE application in C on Raspberry Pi

kahlenberg, 2022-11-25:
Hi,
I couldn't find any proper place to ask my question. If it is not suitable here, apologizes.
I want to develop a Bluetooth application in **C** language on Raspberry Pi. I already developed a firmware for nRF52840 Bluetooth module from Nordic Semiconductor. Furthermore, I can connect to it with my phone using nRFConnect App and I can subscribe to a notification service. It works fine. Now I want to develop a C application on Raspberry Pi sending data, receiving data, reading notification and so on.
But I could not find any proper example code. What I found is only scanning applications or pyhton scripts.
How can I implement a **C** application for that?
Thanks.

Vudentz, 2022-11-28:
@kahlenberg there is the likes of bluetoothctl that is written in C:

https://github.com/bluez/bluez/blob/master/client/main.c

The GATT parts you can find here:

https://github.com/bluez/bluez/blob/master/client/gatt.c

It fairly complete and you can do must things that nRFConnect can.

weliem, 2022-11-30:
Alternatively you could have a look here: https://github.com/weliem/bluez_inc