bluez/bluez #431: A crash (rare) on headphones disconnection

kuraga, 2022-11-27:
bluetoothctl 5.65 on Calculate Linux (Gentoo).

A crash (rare) on headphones disconnection.

```
Nov 26 20:10:05 bluetoothd[4221]: profiles/audio/avdtp.c:avdtp_close() avdtp_close: rejecting since close is already initiated
Nov 26 20:10:05 bluetoothd[4221]: profiles/audio/avdtp.c:cancel_request() Close: Operation canceled (125)
Nov 26 20:10:05 bluetoothd[4221]: profiles/audio/avdtp.c:session_cb() Transaction label doesn't match
Nov 26 20:10:05 kernel: bluetoothd[4221]: segfault at fffffffffffffff9 ip 00007f33c64e7c5a sp 00007ffc90a5f6a0 error 5 in libc.so.6[7f33c6478000+152000]
Nov 26 20:10:05 kernel: Code: 00 e8 ea 5e fe ff 66 2e 0f 1f 84 00 00 00 00 00 48 85 ff 0f 84 bf 00 00 00 55 48 8d 77 f0 53 48 83 ec 18 48 8b 1d a6 a1 13 00 <48> 8b 47 f8 64 8b 2b a8 02 75 5b 48 8b 15 2c a1 13 00 64 48 83 3a
```

kuraga, 2023-01-20:
BlueZ 5.66 on Calculate Linux (Gentoo).

A crash on headphones disconnection.

Q: How do I reproduce this crash?
A:
1. Switch profiles.
2. Talk with Telegram.
3. Switch off Bluetooth connection.

Seems like disconnect callback is called twice...

```
Nov 26 20:10:05 bluetoothd[4221]: profiles/audio/avdtp.c:avdtp_close() avdtp_close: rejecting since close is already initiated
Nov 26 20:10:05 bluetoothd[4221]: profiles/audio/avdtp.c:cancel_request() Close: Operation canceled (125)
Nov 26 20:10:05 bluetoothd[4221]: profiles/audio/avdtp.c:session_cb() Transaction label doesn't match
Nov 26 20:10:05 kernel: bluetoothd[4221]: segfault at fffffffffffffff9 ip 00007f33c64e7c5a sp 00007ffc90a5f6a0 error 5 in libc.so.6[7f33c6478000+152000]
Nov 26 20:10:05 kernel: Code: 00 e8 ea 5e fe ff 66 2e 0f 1f 84 00 00 00 00 00 48 85 ff 0f 84 bf 00 00 00 55 48 8d 77 f0 53 48 83 ec 18 48 8b 1d a6 a1 13 00 <48> 8b 47 f8 64 8b 2b a8 02 75 5b 48 8b 15 2c a1 13 00 64 48 83 3a
```

```gdb
Program received signal SIGSEGV, Segmentation fault.
0x00007fda2118773a in __GI___libc_free (mem=0x1) at malloc.c:3362
3362	  if (chunk_is_mmapped (p))                       /* release mmapped memory. */
```

```gdb
(gdb) bt
#0  0x00007fda2118773a in __GI___libc_free (mem=0x1) at malloc.c:3362
#1  0x00007fda21377283 in g_free (mem=<optimized out>) at ../glib-2.74.4/glib/gmem.c:229
#2  0x00007fda2136da55 in g_source_unref_internal (source=source@entry=0x56269f2be6a0, context=context@entry=0x56269f299930, have_lock=have_lock@entry=1) at ../glib-2.74.4/glib/gmain.c:2397
#3  0x00007fda2136dc23 in g_source_destroy_internal (source=0x56269f2be6a0, context=0x56269f299930, have_lock=have_lock@entry=0) at ../glib-2.74.4/glib/gmain.c:1427
#4  0x00007fda2136eb26 in g_source_destroy (source=<optimized out>) at ../glib-2.74.4/glib/gmain.c:1464
#5  0x000056269e36d6fa in timeout_remove (id=<optimized out>) at src/shared/timeout-glib.c:72
#6  0x000056269e2c9767 in session_cb (chan=0x56269f2be5f0, cond=G_IO_IN, data=data@entry=0x56269f2cfc30) at profiles/audio/avdtp.c:2281
#7  0x00007fda213c54f1 in g_io_unix_dispatch (source=source@entry=0x56269f2fcf50, callback=0x56269e2c94d1 <session_cb>, user_data=0x56269f2cfc30) at ../glib-2.74.4/glib/giounix.c:168
#8  0x00007fda2136e078 in g_main_dispatch (context=context@entry=0x56269f299930) at ../glib-2.74.4/glib/gmain.c:3454
#9  0x00007fda213717af in g_main_context_dispatch (context=context@entry=0x56269f299930) at ../glib-2.74.4/glib/gmain.c:4172
#10 0x00007fda21371924 in g_main_context_iterate (context=0x56269f299930, block=block@entry=1, dispatch=dispatch@entry=1, self=self@entry=0x56269f2a0230) at ../glib-2.74.4/glib/gmain.c:4248
#11 0x00007fda21371cbf in g_main_loop_run (loop=0x56269f29aae0) at ../glib-2.74.4/glib/gmain.c:4448
#12 0x000056269e36d821 in mainloop_run () at src/shared/mainloop-glib.c:66
#13 0x000056269e36dc92 in mainloop_run_with_signal (func=func@entry=0x56269e2f65a8 <signal_callback>, user_data=user_data@entry=0x0) at src/shared/mainloop-notify.c:188
#14 0x000056269e2f875c in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:1289
```

```gdb
(gdb) bt -full
#0  0x00007fda2118773a in __GI___libc_free (mem=0x1) at malloc.c:3362
        ar_ptr = <optimized out>
        p = <optimized out>
        err = <optimized out>
#1  0x00007fda21377283 in g_free (mem=<optimized out>) at ../glib-2.74.4/glib/gmem.c:229
No locals.
#2  0x00007fda2136da55 in g_source_unref_internal (source=source@entry=0x56269f2be6a0, context=context@entry=0x56269f299930, have_lock=have_lock@entry=1) at ../glib-2.74.4/glib/gmain.c:2397
        old_cb_data = 0x0
        old_cb_funcs = 0x0
        __func__ = "g_source_unref_internal"
#3  0x00007fda2136dc23 in g_source_destroy_internal (source=0x56269f2be6a0, context=0x56269f299930, have_lock=have_lock@entry=0) at ../glib-2.74.4/glib/gmain.c:1427
        tmp_list = <optimized out>
        old_cb_data = 0x56269f2bcc50
        old_cb_funcs = <optimized out>
#4  0x00007fda2136eb26 in g_source_destroy (source=<optimized out>) at ../glib-2.74.4/glib/gmain.c:1464
        context = <optimized out>
        __func__ = "g_source_destroy"
#5  0x000056269e36d6fa in timeout_remove (id=<optimized out>) at src/shared/timeout-glib.c:72
        source = <optimized out>
#6  0x000056269e2c9767 in session_cb (chan=0x56269f2be5f0, cond=G_IO_IN, data=data@entry=0x56269f2cfc30) at profiles/audio/avdtp.c:2281
        session = 0x56269f2cfc30
        header = 0x56269f2cf220
        size = <optimized out>
[... 29 lines cut ...]
        begin_time_nsec = 0
#11 0x00007fda21371cbf in g_main_loop_run (loop=0x56269f29aae0) at ../glib-2.74.4/glib/gmain.c:4448
        self = 0x56269f2a0230
        __func__ = "g_main_loop_run"
#12 0x000056269e36d821 in mainloop_run () at src/shared/mainloop-glib.c:66
No locals.
#13 0x000056269e36dc92 in mainloop_run_with_signal (func=func@entry=0x56269e2f65a8 <signal_callback>, user_data=user_data@entry=0x0) at src/shared/mainloop-notify.c:188
        data = 0x56269f2a6560
        io = 0x56269f2a6580
        ret = <optimized out>
#14 0x000056269e2f875c in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:1289
        context = 0x56269f297d30
        err = 0x0
        sdp_mtu = 0
        sdp_flags = <optimized out>
        gdbus_flags = <optimized out>
        __func__ = "main"
```

```gdb
(gdb) generate-core-file /home/sasha/bluez-core
warning: Memory read failed for corefile section, 4096 bytes at 0xffffffffff600000.
Saved corefile /home/sasha/bluez-core
```
https://gist.github.com/kuraga/85cbb1dc461db005e0fb00ab2b660146#file-bluez-core

ValeZAA, 2025-04-03:
Does not happen here, with this very scenario.

But i do not understand why it warns when you disconnect or when somebody else connects and thus makes your headphones disconnect.

ValeZAA, 2025-06-10:
> [@ValeZAA](https://github.com/ValeZAA) , were you turning off from the headset profile?

Yes.

closed 2025-09-24 by github-actions[bot]