bluez/bluez #432: MX Keys Mini - Doesn't establish connection when trying to pair

SonidoCristalino, 2022-11-29:
Hello guys, how are you?
I got the following [keyboard](https://www.logitech.com/en-us/products/keyboards/mx-keys-mini.920-010473.html)
I had no problem pairing this keyboard with my work laptop, which is running Windows 10.
But I did have problems with my personal laptop on which I only have ArchLinux installed and using `bluetoothctl` to try to pair with it. 
I have followed the steps described on the following [Archlinux page](https://wiki.archlinux.org/title/Bluetooth_keyboard)

**Problem:** 
The problem is that although it detects the device, after entering the command `pair [device]` it does not return the access key for pairing. After a few seconds it disconnects and reconnects to the keyboard in an infinite loop.

**Information**
* [uname.log](https://github.com/bluez/bluez/files/10108897/uname.log)
* [pacmanBluez.log](https://github.com/bluez/bluez/files/10108900/pacmanBluez.log)
* [rfkill.log](https://github.com/bluez/bluez/files/10108929/rfkill.log)

**Btmon traces**
The device I want to connect to is _DE:6A:C5:1E:B9:96, MX Keys Mini_
* [btmon.log](https://github.com/bluez/bluez/files/10108935/btmon.log)

**System information after attempting to connect**
After unsuccessfully trying to pair the keyboard, you can see in the following files some errors that also appear in btmon ():
* [dmesgBluetooth.log](https://github.com/bluez/bluez/files/10108955/dmesgBluetooth.log)
* [journalctl.log](https://github.com/bluez/bluez/files/10108959/journalctl.log)

I browsed for almost two weeks through the issues in the repository trying other users' solutions, as well as trying some solutions provided by the Archlinux forums, but I couldn't find the key to the problem.
Sorry in advance if I'm missing something to provide more information. And thank you very much for any information that will help me to clarify the problem.

Vudentz, 2022-12-02:
```
< ACL Data TX: Handle 1 flags 0x00 dlen 11                                                                                                                                                                                                                              #38 [hci0] 106.861280
      SMP: Pairing Request (0x01) len 6
        IO capability: KeyboardDisplay (0x04)
        OOB data: Authentication data not present (0x00)
        Authentication requirement: Bonding, MITM, SC, No Keypresses, CT2 (0x2d)
        Max encryption key size: 16
        Initiator key distribution: EncKey Sign LinkKey (0x0d)
        Responder key distribution: EncKey IdKey Sign LinkKey (0x0f)
> ACL Data RX: Handle 1 flags 0x02 dlen 11                                                                                                                                                                                                                              #42 [hci0] 107.039804
      SMP: Pairing Response (0x02) len 6
        IO capability: KeyboardOnly (0x02)
        OOB data: Authentication data not present (0x00)
        Authentication requirement: Bonding, MITM, SC, Keypresses (0x1d)
        Max encryption key size: 16
        Initiator key distribution: <none> (0x00)
        Responder key distribution: EncKey IdKey (0x03)
< ACL Data TX: Handle 1 flags 0x01 dlen 5                                                                                                                                                                                                                               #45 [hci0] 107.060040
      SMP: Pairing Public Key (0x0c) len 64
        X: 9457a7e227b0e33ddb57a3b9aaf4233fa19fa2d9d5c971a758ffa891475f52ba
        Y: f33ae697809ee64f8e6e1758a35baed85922b1e776aef82b5ba1308602bf6744

```

After that it doesn't seem the device responds anymore.

SonidoCristalino, 2022-12-22:
Hello guys how are you?
I am closing this issue because the only solution I found to this problem was to buy a [Logitech Bolt receiver](https://www.logitech.com/es-ar/products/mice/logi-bolt-usb-receiver.956-000007.html) so that I could establish a connection with my MX Keys Mini with my Arch Linux

Thank you very much for your assistance!

closed 2022-12-22 by SonidoCristalino