bluez/bluez #435: Mouse does not work after computer wakes up from sleep

nekromoff, 2022-12-01:
STEPS TO REPRODUCE
1. Suspend computer
2. Wake up computer

OBSERVED RESULT
Mouse not working (cursor not moving)

EXPECTED RESULT
Mouse working, cursor moving

SOFTWARE/OS VERSIONS
Operating System: KDE neon 5.26
KDE Plasma Version: 5.26.3
KDE Frameworks Version: 5.100.0
Qt Version: 5.15.7
Kernel Version: 5.15.0-53-generic (64-bit)
Graphics Platform: Wayland

ADDITIONAL INFORMATION
HW restart of mouse (turn off/turn on) helps. This has started to happen after the latest Neon upgrade to Jammy.

```
dmesg:
[156632.525056] Bluetooth: hci0: Waiting for firmware download to complete
[156632.525715] Bluetooth: hci0: Firmware loaded in 1457966 usecs
[156632.525761] Bluetooth: hci0: Waiting for device to boot
[156632.540748] Bluetooth: hci0: Device booted in 14671 usecs
[156632.540758] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-20-1-3.ddc
[156632.544759] Bluetooth: hci0: Applying Intel DDC parameters completed
[156632.550737] Bluetooth: hci0: Firmware revision 0.3 build 126 week 5 2022

$ bluetoothctl 
Agent registered
[400-MA102]

Device B4:BC:7C:2F:CF:A5 (public)
Name: 400-MA102
Alias: 400-MA102
Appearance: 0x03c2
Icon: input-mouse
Paired: yes
Trusted: yes
Blocked: no
Connected: yes
WakeAllowed: yes
LegacyPairing: no
UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
Modalias: bluetooth:v000Dp0000d0110
Battery Percentage: 0x32 (50)
```

also dmesg after wakeup:
```
[308383.450558] Bluetooth: hci0: Bootloader revision 0.3 build 0 week 24 2017
[308383.451640] done.
[308383.453537] Bluetooth: hci0: Device revision is 1
[308383.453543] Bluetooth: hci0: Secure boot is enabled
[308383.453544] Bluetooth: hci0: OTP lock is enabled
[308383.453545] Bluetooth: hci0: API lock is enabled
[308383.453546] Bluetooth: hci0: Debug lock is disabled
[308383.453547] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[308383.454544] Bluetooth: hci0: Found device firmware: intel/ibt-20-1-3.sfi
```

Vudentz, 2022-12-02:
@nekromoff I suspect it could be kernel related, can you try with a more recent kernel? Also does it work if you attempt to initiate the connection via bluetoothctl?

nekromoff, 2022-12-04:
Kernel:
5.15.0-53-generic

Mouse will start to work when I run `bluetoothctl` and enter `info` command.

nekromoff, 2022-12-29:
using `bluetoothctl` I get this after about 10 seconds:

```
[CHG] Device B4:BC:7C:2F:CF:A5 ServicesResolved: yes
[CHG] Device B4:BC:7C:2F:CF:A5 Paired: yes
[CHG] Device B4:BC:7C:2F:CF:A5 ServicesResolved: no
[CHG] Device B4:BC:7C:2F:CF:A5 Connected: no
[CHG] Device B4:BC:7C:2F:CF:A5 Connected: yes
[NEW] Primary Service (Handle 0x0000)
[snip]
[CHG] Device B4:BC:7C:2F:CF:A5 ServicesResolved: yes
```

Mouse will work after that.