bluez/bluez #437: power off doesn't claim an error if called with rfkilled device

pizdjuk, 2022-12-08:
Device: RTL8723BS
Kernel: linux-sunxi, 6.1 version

Steps to reproduce:

1. Enable Bluetooth
2. bluetoothctl discoverable on
3. rfkill the bluetooth device (rfkill all, for example)
4. bluetoothctl power off
5. the device continues to send discover messages

in place the steps:

....
3. bluetoothctl power off

and the device sends nothing more.

In the first sequence the bluetoothctl power off should return an error message, but does nothing.

Vudentz, 2022-12-08:
@pizdjuk that is weird, is RTL8723BS using btusb driver? The rfkill should have stopped the discovery and powered it down if I remember it correctly.

Vudentz, 2022-12-08:
sudo rfkill block <id>

```
bluetoothd[650607]: src/shared/mgmt.c:can_read_data() [0x0001] event 0x0007
bluetoothd[650607]: src/adapter.c:dev_class_changed_callback() Class: 0x000000
bluetoothd[650607]: src/shared/mgmt.c:can_read_data() [0x0001] event 0x0006
bluetoothd[650607]: src/adapter.c:new_settings_callback() Settings: 0x00000ac0
bluetoothd[650607]: src/adapter.c:settings_changed() Changed settings: 0x00000001
bluetoothd[650607]: src/adapter.c:settings_changed() Pending settings: 0x00000000
bluetoothd[650607]: src/adapter.c:cancel_passive_scanning() 
bluetoothd[650607]: src/adapter.c:adapter_stop() adapter /org/bluez/hci1 has been disabled
bluetoothd[650607]: src/rfkill.c:rfkill_event() RFKILL event idx 15 type 2 op 2 soft 1 hard 0
bluetoothd[650607]: src/rfkill.c:rfkill_event() RFKILL unblock for hci1
bluetoothd[650607]: src/adapter.c:adapter_set_power_state() off-blocked
```

```
< HCI Command: Reset (0x03|0x0003) plen 0                                                                         #2519 [hci1] 13:51:53.690728
> HCI Event: Command Complete (0x0e) plen 4                                                                       #2520 [hci1] 13:51:53.811288
      Reset (0x03|0x0003) ncmd 2
        Status: Success (0x00)
@ MGMT Event: Class Of Device Changed (0x0007) plen 3                                                          {0x0001} [hci1] 13:51:53.811405
        Class: 0x000000
          Major class: Miscellaneous
          Minor class: 0x00
@ MGMT Event: New Settings (0x0006) plen 4                                                                     {0x0001} [hci1] 13:51:53.811410
        Current settings: 0x00000ac0
          Secure Simple Pairing
          BR/EDR
          Low Energy
          Secure Connections
```

Here it seems to work just fine with an Intel Controller, it even reset if I rfkill it.

pizdjuk, 2022-12-09:
> @pizdjuk that is weird, is RTL8723BS using btusb driver? The rfkill should have stopped the discovery and powered it down if I remember it correctly.

No, btrtl + hci_uart,.

closed 2025-08-14 by github-actions[bot]