bluez/bluez #438: Segfault in bluetoothd when getting services/characteristics for a specific device

bdr99, 2022-12-12:
Operating System: Ubuntu 22.04 (Linux 5.15.0-56-generic)
BlueZ version: 5.64

I have a certain device (Eufy Smart Scale P2) which exposes three BLE services. The issue I'm having is that BlueZ can only see the first two, and the third one (UUID 0xFFF0) is not visible.

When I ran `sudo bluetoothd --debug --nodetach` to get the debug output, I noticed that `bluetoothd` at first sees all three services, but then segfaults. My guess is that the first two services are getting added correctly, but it segfaults before it finishes adding the third one, hence why it is not visible in `bluetoothctl`.

I'm pretty new to BlueZ so please let me know what info I can provide to help debug this issue.

Here is the log output of `bluetoothd` showing the segfault:

```
bluetoothd[5978]: src/device.c:device_connect_le() Connection attempt to: CF:E6:08:09:01:02
bluetoothd[5978]: src/adapter.c:connected_callback() hci0 device CF:E6:08:09:01:02 connected eir_len 30
bluetoothd[5978]: attrib/gattrib.c:g_attrib_ref() 0x560c15e89c70: g_attrib_ref=1
bluetoothd[5978]: src/device.c:load_gatt_db() Restoring CF:E6:08:09:01:02 gatt database from file
bluetoothd[5978]: src/device.c:load_gatt_db() No cache for CF:E6:08:09:01:02
bluetoothd[5978]: src/gatt-client.c:btd_gatt_client_connected() Device connected.
bluetoothd[5978]: src/device.c:gatt_debug() MTU exchange complete, with MTU: 128
bluetoothd[5978]: src/device.c:gatt_debug() Primary services found: 3
bluetoothd[5978]: src/device.c:gatt_debug() start: 0x0001, end: 0x000b, uuid: 0000180a-0000-1000-8000-00805f9b34fb
bluetoothd[5978]: src/device.c:gatt_debug() start: 0x000c, end: 0x000f, uuid: 0000180f-0000-1000-8000-00805f9b34fb
bluetoothd[5978]: src/device.c:gatt_debug() start: 0x0010, end: 0x0018, uuid: 0000fff0-0000-1000-8000-00805f9b34fb
bluetoothd[5978]: src/device.c:gatt_debug() Secondary services found: 3
bluetoothd[5978]: src/device.c:gatt_debug() start: 0x0001, end: 0x000b, uuid: 0000180a-0000-1000-8000-00805f9b34fb
bluetoothd[5978]: src/device.c:gatt_debug() start: 0x000c, end: 0x000f, uuid: 0000180f-0000-1000-8000-00805f9b34fb
bluetoothd[5978]: src/device.c:gatt_debug() start: 0x0010, end: 0x0018, uuid: 0000fff0-0000-1000-8000-00805f9b34fb
bluetoothd[5978]: src/device.c:gatt_debug() Characteristics found: 9
bluetoothd[5978]: src/device.c:gatt_debug() start: 0x0002, end: 0x0003, value: 0x0003, props: 0x02, uuid: 00002a27-0000-1
bluetoothd[5978]: src/device.c:gatt_debug() start: 0x0004, end: 0x0005, value: 0x0005, props: 0x02, uuid: 00002a26-0000-1
bluetoothd[5978]: src/device.c:gatt_debug() start: 0x0006, end: 0x0007, value: 0x0007, props: 0x02, uuid: 00002a29-0000-1
bluetoothd[5978]: src/device.c:gatt_debug() start: 0x0008, end: 0x0009, value: 0x0009, props: 0x02, uuid: 00002a24-0000-1
bluetoothd[5978]: src/device.c:gatt_debug() start: 0x000a, end: 0x000c, value: 0x000b, props: 0x02, uuid: 00002a25-0000-1
bluetoothd[5978]: src/device.c:gatt_debug() start: 0x000d, end: 0x0010, value: 0x000e, props: 0x12, uuid: 00002a19-0000-1
bluetoothd[5978]: src/device.c:gatt_debug() start: 0x0011, end: 0x0012, value: 0x0012, props: 0x08, uuid: 0000fff1-0000-1
bluetoothd[5978]: src/device.c:gatt_debug() start: 0x0013, end: 0x0015, value: 0x0014, props: 0x10, uuid: 0000fff2-0000-1
bluetoothd[5978]: src/device.c:gatt_debug() start: 0x0016, end: 0x0018, value: 0x0017, props: 0x10, uuid: 0000fff4-0000-1
[1]    5976 segmentation fault  sudo bluetoothd --debug --nodetach
```

Here is a screenshot of nRF connect on my phone, showing all three services:

[image]

Here is the output when I run `bluetoothctl info` on the device (notice the FFF0 service is not shown):
```
Device CF:E6:08:09:01:02 (public)
        Name: eufy T9148
        Alias: eufy T9148
        Paired: no
        Trusted: no
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
        ManufacturerData Key: 0xff5f
        ManufacturerData Value:
  cf e6 08 09 01 02 cf 00 00 3a 02 00 00 00 01 01  .........:......
  00 00 f7                                         ...
        Battery Percentage: 0x5f (95)
```

Vudentz, 2022-12-12:
@bdr99 try running bluetoothd with valgrind to generate a backtrace.

bdr99, 2022-12-12:
@Vudentz Sure, here are the logs when running with valgrind:

[valgrind.log](https://github.com/bluez/bluez/files/10212072/valgrind.log)

Also, running bluetoothd with valgrind actually fixes the issue, I assume because it allows bluetoothd to continue despite the segfault:

```
➜  ~ bluetoothctl info CF:E6:08:09:01:02
Device CF:E6:08:09:01:02 (public)
        Name: eufy T9148
        Alias: eufy T9148
        Paired: no
        Trusted: no
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000fff0-0000-1000-8000-00805f9b34fb)
```

Vudentz, 2022-12-12:
@bdr99 please install the debug symbols otherwise valgrind can't decode them:

```

==6843== Invalid read of size 8
==6843==    at 0x1C0F67: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1C1647: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1CA687: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1C015B: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1C4AE8: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x48B9C43: g_main_context_dispatch (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==6843==    by 0x490E6C7: ??? (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==6843==    by 0x48B92B2: g_main_loop_run (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==6843==    by 0x129C6E: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x4A15D8F: (below main) (libc_start_call_main.h:58)
==6843==  Address 0x53a7220 is 0 bytes inside a block of size 128 free'd
==6843==    at 0x484B27F: free (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==6843==    by 0x1C771F: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1D0E3C: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1BC9B2: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1C1ABB: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1CA687: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1C015B: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1C4AE8: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x48B9C43: g_main_context_dispatch (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==6843==    by 0x490E6C7: ??? (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==6843==    by 0x48B92B2: g_main_loop_run (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==6843==    by 0x129C6E: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==  Block was alloc'd at
==6843==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==6843==    by 0x1C5646: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1C79FD: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1BC8E1: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1C1DB2: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1CA687: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1C015B: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x1C4AE8: ??? (in /usr/lib/bluetooth/bluetoothd)
==6843==    by 0x48B9C43: g_main_context_dispatch (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==6843==    by 0x490E6C7: ??? (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==6843==    by 0x48B92B2: g_main_loop_run (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==6843==    by 0x129C6E: ??? (in /usr/lib/bluetooth/bluetoothd
```

Btw, any chance to test with the latest to see if still reproducible with it?

bdr99, 2022-12-12:
@Vudentz With help from @dlech I was able to get the debug symbols installed. Here is the valgrind log with debug symbols:

[valgrind_with_symbols.log](https://github.com/bluez/bluez/files/10212611/valgrind_with_symbols.log)

I would still need some guidance in order to test with the latest commit.

dlech, 2022-12-12:
Instructions to build BlueZ from source are in the main README.

bdr99, 2022-12-13:
Thanks again, @dlech.

@Vudentz I tested with the latest master commit, and I am no longer getting a segfault from bluetoothd. However, the original problem still remains, which is that BlueZ does not see the service with UUID 0xFFF0. 5.65 behaves the same as latest master. So, to summarize:

5.64: Segfaults, and service does not show.
5.65: Does not segfault, but service does not show.
latest master: Does not segfault, but service does not show.

Here is the output of `bluetoothctl connect` and `bluetoothctl info` with the latest master, showing that only two of the three services are visible:
```
 ➜  ~ bluetoothctl connect CF:E6:08:09:01:02
Attempting to connect to CF:E6:08:09:01:02
[CHG] Device CF:E6:08:09:01:02 Connected: yes
Connection successful
 ➜  ~ bluetoothctl info CF:E6:08:09:01:02
Device CF:E6:08:09:01:02 (public)
        Name: eufy T9148
        Alias: eufy T9148
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
```

Here is the debug log from bluetoothd with the latest master commit, which does not segfault:

[bluetoothd_latest_master.log](https://github.com/bluez/bluez/files/10213130/bluetoothd_latest_master.log)

bdr99, 2022-12-13:
@Vudentz I bisected the segfault to this commit: https://github.com/bluez/bluez/commit/e0870ce5e1fe842e11f5d2a8425b4c4594ca0799. I noticed that commit only changed the debug logging, so I tried running `bluetoothd` without `--debug`, and it did not segfault. So it looks like the segfault is actually unrelated to the issue with the missing service.

I've updated the issue title and description to reflect these findings, and I'd appreciate any input you may have for this issue. Please let me know if I can provide any other info.

Vudentz, 2022-12-15:
@bdr99 Seems like there is some bugs with the last service:

```
bluetoothd[46555]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x0011, end: 0x0012, value: 0x0012, props: 0x08, uuid: 0000fff1-0000-1000-8000-00805f9b34fb
bluetoothd[46555]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x0013, end: 0x0015, value: 0x0014, props: 0x10, uuid: 0000fff2-0000-1000-8000-00805f9b34fb
bluetoothd[46555]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x0016, end: 0x0018, value: 0x0017, props: 0x10, uuid: 0000fff4-0000-1000-8000-00805f9b34fb
bluetoothd[46555]: src/shared/gatt-client.c:discovery_op_complete() service disappeared: start 0x0010 end 0x0018
bluetoothd[46555]: src/device.c:gatt_service_removed() start: 0x0010, end: 0x0018
```

Were you running with: 

https://github.com/bluez/bluez/commit/43d71b8410ff8cf1abcb4ec5353041268f49b037

Vudentz, 2022-12-16:
@bdr99 try with the following change:

https://patchwork.kernel.org/project/bluetooth/patch/20221215234404.2735181-1-luiz.dentz@gmail.com/

bdr99, 2022-12-16:
@Vudentz Unfortunately, that patch didn't resolve the issue.

The debug output is the same as before, I still see the "service disappeared" line after the characteristics are discovered.

Vudentz, 2022-12-16:
> @Vudentz Unfortunately, that patch didn't resolve the issue.
> 
> The debug output is the same as before, I still see the "service disappeared" line after the characteristics are discovered.

Ive sent a v2:

https://patchwork.kernel.org/project/bluetooth/patch/20221216210611.2990552-1-luiz.dentz@gmail.com/

bdr99, 2022-12-16:
@Vudentz That version didn't work either unfortunately. Again, the log output is the same as before, and I don't see the any occurrences of the new `Unable to get service data at 0x%04x` log that you added.

Vudentz, 2022-12-16:
> @Vudentz That version didn't work either unfortunately. Again, the log output is the same as before, and I don't see the any occurrences of the new `Unable to get service data at 0x%04x` log that you added.

Hmm, you don't seem to have any included service, not really sure why, can you upload the btmon traces?

bdr99, 2022-12-17:
I already sent this trace to @Vudentz via Slack, but here it is in case anyone else wants to take a look: 
[issue438.log](https://github.com/bluez/bluez/files/10252769/issue438.log)

bdr99, 2022-12-17:
@Vudentz Your v4 patch seems to have fixed the issue for me!

https://patchwork.kernel.org/project/bluetooth/patch/20221217011059.3087848-1-luiz.dentz@gmail.com/
https://patchwork.kernel.org/project/bluetooth/patch/20221217011059.3087848-2-luiz.dentz@gmail.com/

closed 2022-12-19 by BluezTestBot