bluez/bluez #441: Cleaning up SAP initialization error

lorenz, 2022-12-17:
Currently, BlueZ always logs two errors when starting:
```
sap-server: Operation not permitted (1)
profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
```
As far as I can tell both of those are caused by BlueZ's default dbus policy (https://github.com/bluez/bluez/blob/3d6e4bf14abfe592bb95471eb989a7899febe779/src/bluetooth.conf) as it does not not permit `org.bluez.SimAccessTest1. This does not cause an immediate problem with the rest of bluez and can be fully worked around by just disabling the SAP plugin on the command line, but IMO either should the plugin be disabled by default in BlueZ or the default dbus policy should work with it.

maxnikulin, 2024-01-03:
It sounds like a distribution-specific issue in the sense that the SAP plugin is not built by default, see
https://github.com/bluez/bluez/blob/7ad5669402c9acff8e4cc808edc12a41df36654e/configure.ac#L170-L172
https://github.com/bluez/bluez/blob/7ad5669402c9acff8e4cc808edc12a41df36654e/Makefile.plugins#L24-L30
E.g. Debian has `--enable-sap` in `debian/rules`, so it is affected <https://bugs.debian.org/803265> while Ubuntu is not <https://launchpad.net/bugs/1629632>

If the plugin still can do anything useful then [`src/bluetooth.conf`](../blob/master/src/bluetooth.conf) should be generated basing on the `configure` options.

When noticed first time such warnings are confusing, they make troubleshooting more time consuming.

lorenz, 2024-01-03:
Right, I don't know how I missed that, I thought I checked if nixpkgs was enabling this explicitly. But apparently they did and only recently stopped (https://github.com/NixOS/nixpkgs/commit/77c75090b69fbc7e628b5d5c5d56e95171c1976d). In that case this is not really a BlueZ bug as the defaults are working, so closing this.

closed 2024-01-03 by lorenz

maxnikulin, 2024-01-10:
I believe that it is a bug that D-Bus policy is inconsistent with `configure` option. On the other hand there is no reason to fix it if SAP server is unsupported.

I have no idea if `ofono` or some other tool may interact with SAP in `bluez`.