bluez/bluez #443: HID service characteristics not supported with dbus interfaces.

Timfuture, 2022-12-19:
Hi, all

I got a problem that when I try to connect a hid device with bluez, I got service like next:
[HID]# list-attributes 
Primary Service (Handle 0x0000)
	/org/bluez/hci0/dev_24_D7_EB_19_5D_96/service0028
	0000180f-0000-1000-8000-00805f9b34fb
	Battery Service
Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_24_D7_EB_19_5D_96/service0028/char0029
	00002a19-0000-1000-8000-00805f9b34fb
	Battery Level
Descriptor (Handle 0x0000)
	/org/bluez/hci0/dev_24_D7_EB_19_5D_96/service0028/char0029/desc002c
	00002904-0000-1000-8000-00805f9b34fb
	Characteristic Format
Descriptor (Handle 0x0000)
	/org/bluez/hci0/dev_24_D7_EB_19_5D_96/service0028/char0029/desc002b
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
Primary Service (Handle 0x0000)
	/org/bluez/hci0/dev_24_D7_EB_19_5D_96/service0001
	00001801-0000-1000-8000-00805f9b34fb
	Generic Attribute Profile
Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_24_D7_EB_19_5D_96/service0001/char0002
	00002a05-0000-1000-8000-00805f9b34fb
	Service Changed
Descriptor (Handle 0x0000)
	/org/bluez/hci0/dev_24_D7_EB_19_5D_96/service0001/char0002/desc0004
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration

You can see there is no HID relevant services and characteristics.
But according the wireshark info, I think this device do provided HID relevant services and characteristics. And my ubuntu did discovered such services correctly:

Bluetooth Attribute Protocol
    Opcode: Read By Group Type Response (0x11)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 0001 = Method: Read By Group Type Response (0x11)
    Length: 6
    Attribute Data, Handle: 0x0001, Group End Handle: 0x0005, UUID: Generic Attribute Profile
        Handle: 0x0001 (Generic Attribute Profile)
            [UUID: Generic Attribute Profile (0x1801)]
        Group End Handle: 0x0005
        UUID: Generic Attribute Profile (0x1801)
    Attribute Data, Handle: 0x0014, Group End Handle: 0x001c, UUID: Generic Access Profile
        Handle: 0x0014 (Generic Access Profile)
            [UUID: Generic Access Profile (0x1800)]
        Group End Handle: 0x001c
        UUID: Generic Access Profile (0x1800)
    Attribute Data, Handle: 0x0028, Group End Handle: 0x002c, UUID: Battery Service
        Handle: 0x0028 (Battery Service)
            [UUID: Battery Service (0x180f)]
        Group End Handle: 0x002c
        UUID: Battery Service (0x180f)
    **Attribute Data, Handle: 0x002d, Group End Handle: 0xffff, UUID: Human Interface Device
        Handle: 0x002d (Human Interface Device)
            [UUID: Human Interface Device (0x1812)]
        Group End Handle: 0xffff
        UUID: Human Interface Device (0x1812)**
    [UUID: GATT Primary Service Declaration (0x2800)]
    [Request in Frame: 66]

Bluetooth Attribute Protocol
    ...
    Attribute Data, Handle: 0x0036, Characteristic Handle: 0x0037, UUID: Protocol Mode
        Handle: 0x0036 (Human Interface Device: Report Map: GATT Characteristic Declaration)
            [Service UUID: Human Interface Device (0x1812)]
            [Characteristic UUID: Report Map (0x2a4b)]
            [UUID: GATT Characteristic Declaration (0x2803)]
        Characteristic Properties: 0x0a, Write, Read
            0... .... = Extended Properties: False
            .0.. .... = Authenticated Signed Writes: False
            ..0. .... = Indicate: False
            ...0 .... = Notify: False
            .... 1... = Write: True
            .... .0.. = Write without Response: False
            .... ..1. = Read: True
            .... ...0 = Broadcast: False
        Characteristic Value Handle: 0x0037 (Human Interface Device: Protocol Mode)
            [Service UUID: Human Interface Device (0x1812)]
            [UUID: Protocol Mode (0x2a4e)]
        UUID: Protocol Mode (0x2a4e)
    ...
    [Request in Frame: 89]

Is anyone know why the dbus interfaces didn't include such HID services??
Any of your advices will be appreciate.

endrift, 2023-01-28:
I've run into this too, and best I can tell it's intentional, though I'm not sure why.

Timfuture, 2023-01-31:
@endrift , thank you for your reply.

endrift, 2025-06-11:
Hello. Stale bots are a pox. This issue hasn't been resolved.

endrift, 2026-03-02:
I just tried to reproduce this and something...else weird happened. The controller sends HIDP messages but Bluez doesn't show it at all. Nothing in evtest either.

```
$ bluetoothctl
[NEW] Media /org/bluez/hci0
        SupportedUUIDs: 0000110a-0000-1000-8000-00805f9b34fb
        SupportedUUIDs: 0000110b-0000-1000-8000-00805f9b34fb
Agent registered
[Xbox Wireless Controller]> menu gatt
[Xbox Wireless Controller]> list-attributes
[Xbox Wireless Controller]>
```

[hid-bluez.pcapng.gz](https://github.com/user-attachments/files/25676752/hid-bluez.pcapng.gz)

I don't remember exactly what was going on before, but Xbox Wireless Controllers definitely have historically worked.