bluez/bluez #444: bluez-5.66 fails to build with -Werror=strict-prototypes

pacho2, 2022-12-19:
As reported downstream:
https://bugs.gentoo.org/874369

bluez fails to build when -Werror=strict-prototypes is used and, then, it will break with future clang versions, that will force this:
```
x86_64-pc-linux-gnu-gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/sysprof-4 -pthread  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/sysprof-4 -pthread  -I./lib  -fPIC      -O2 -Werror=strict-prototypes -c -o tools/hid2hci.o tools/hid2hci.c
/bin/sh ./libtool  --tag=CC   --mode=compile x86_64-pc-linux-gnu-gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/sysprof-4 -pthread  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/sysprof-4 -pthread  -I./lib  -fPIC      -fvisibility=hidden -O2 -Werror=strict-prototypes -c -o plugins/sixaxis_la-sixaxis.lo `test -f 'plugins/sixaxis.c' || echo './'`plugins/sixaxis.c
/usr/sbin/mkdir -p src/ && rst2man.py --strict --no-raw --no-generator --no-datestamp src/bluetoothd.rst src/bluetoothd.8
mesh/manager.c:66:13: error: function declaration isn’t a prototype [-Werror=strict-prototypes]
   66 | static void free_pending_add_call()
      |             ^~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [Makefile:7526: mesh/manager.o] Error 1
make[1]: *** Waiting for unfinished jobs....
libtool: compile:  x86_64-pc-linux-gnu-gcc -DHAVE_CONFIG_H -I. -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/sysprof-4 -pthread -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/sysprof-4 -pthread -I./lib -fPIC -fvisibility=hidden -O2 -Werror=strict-prototypes -c plugins/sixaxis.c  -fPIC -DPIC -o plugins/.libs/sixaxis_la-sixaxis.o
make: *** [Makefile:4468: all] Error 2
```

Thanks

Vudentz, 2022-12-19:
@bgix Can you give it a looks?

inga-s, 2023-04-06:
I believe this was fixed as part of commit:
ommit f3243ecba0a2a062021e461afc4fdc91e480f510
Author: Brian Gix <brian.gix@intel.com>
Date:   Tue Oct 25 14:33:48 2022 -0700

    mesh: Add Remote Provisioning

closed 2025-08-14 by github-actions[bot]