bluez/bluez #445: TemporaryTimeout doesn't apply to all BLE devices that use ADV_IND

jeremyherbert, 2022-12-20:
Hi,

(all of the log outputs are taken from Ubuntu 22.04 with bluez 5.64)

When discovery is terminated, as far as I can tell bluez immediately deletes/frees all BLE devices that do not have general or limited discoverable mode bit set in the advertisement flags, even if the ADV_IND PDU was used. However, use of ADV_IND implies that the device is connectable. As such, I would like to propose that it should delete these devices only after the `TemporaryTimeout` period expires if ADV_IND was used (as opposed to `ADV_NONCONN_IND` which can be deleted immediately). 

The reason is that there are some non-standard devices out there which do not include the advertisement flags that indicate connectability, but they are still connectable as implied by the use of ADV_IND. This is presumably done to reduce the packet size for ultra long battery life and/or to fit more data in in the advertisement. Indeed, `btmon` assumes that `ADV_IND` implies connectable without the advertising flags present as it calls ADV_IND "Connectable undirected":

```
> HCI Event: LE Meta Event (0x3e) plen 31              #1538 [hci0] 1871.008016
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
        Address: A4:C1:38:3E:F2:4E (Telink Semiconductor (Taipei) Co. Ltd.)
        Data length: 19
        Service Data (UUID 0x181a): 4ef23e38c1a4200a2514110b465d05
        RSSI: -45 dBm (0xd3)
```

Indeed the bluetoothd logs show:

```
Dec 20 16:24:54 gnuradio-vm bluetoothd[1850]: src/adapter.c:device_found_callback() hci0 addr A4:C1:38:3E:F2:4E, rssi -45 flags 0x0000 eir_len 31
```

where the flags 0x0000 indicate that the `MGMT_DEV_FOUND_NOT_CONNECTABLE` (0x04) bit is not set, so the device is indeed being reported as connectable.

I note that Android has no problem connecting to these type of devices (ie ADV_IND but no advertising flags), and `bluetoothctl` can connect to them as long as the scan is already running - once `scan off` is run, they are deleted and no longer connectable.

Vudentz, 2022-12-21:
@jeremyherbert you might be missing the following change:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=3fb4e1cca2eaddc413d1ebed9d862084351efdbc

It was release with 5.65.

jeremyherbert, 2022-12-21:
Yes, you are correct! I was testing against 5.64 and then ported my changes across. I should have tested the latest version first.

Thank you for the pointer and sorry for the noise.

closed 2022-12-21 by jeremyherbert