bluez/bluez #448: Custom HOG device working fine on android, linux fails to read the report's CCC

aguaviva, 2022-12-21:
The battery notifications seem to work fine but the report's CCC can't be read, see pic below:

[image]

This is the full trace:

[nordic_custom_HOG.zip](https://github.com/bluez/bluez/files/10278179/nordic_custom_HOG.zip)

 I tried debugging "bluetoothd -n - d" but couldn't find anything relevant. I will be happy to try any debugging experiments you suggest.

Thanks!

Vudentz, 2022-12-21:
@aguaviva that message is just generated before there is a disconnection:
```

> HCI Event: Disconnect Complete (0x05) plen 4                                                                           #218 [hci0] 44.256486
        Status: Success (0x00)
        Handle: 3585
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                    {0x0001} [hci0] 44.256552
        LE Address: 01:01:01:01:01:01 (OUI 01-01-01)
        Reason: Connection timeout (0x01)
```

It seems the link is not really stable if you keep getting connection timeout like this, it is down in the controller which control the link-layer so if you want to further investigate you will have to contact the manufacturer or perhaps switch controllers.

aguaviva, 2022-12-21:
I was wondering why this works consistently on Android but it fails on Linux.

Having said that I am using a nrf51288 and it is a real pain to work with.

aguaviva, 2022-12-26:
I found the reason.

the Nordic stack will stall on CCCD reads unless
- they have been initialized
- a peer_manager is being used 
- the missing attribute is handed: see https://devzone.nordicsemi.com/f/nordic-q-a/57996/nrf-connect-desktop-error-failed-to-get-descriptor

It turns out that android sets the CCCD to 1 without reading it first, that is why it works, and on linux this CCCD is read before setting it and that is why it hangs.

Linux should try either:
- not read CCCDs to be a bit defensive against shitty Nordic products, as Android does.
- show a warning on CCCD reads that hang

Please remind me not to use Nordic products :/ What a pain.