bluez/bluez #449: Linklike Elite 10 not working (Wireplumber)

philfry, 2022-12-22:
Hi all,

I have a Linklike Elite 10 headset which seems to be incompatible with bluez and/or Wireplumber on Fedora 37. IIRC this headset worked fine in earlier Fedora versions which still hat Pulseaudio as standard sound system. Anyway.

The headset pairs and autoconnects fine, but after a couple of errors the headset gets disconnected:
```
bluetoothd[44606]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 20:05:20:07:5C:57: Device or resource busy
bluetoothd[44606]: profiles/audio/avdtp.c:cancel_request() Discover: Connection timed out (110)
```
I attached a btmon hci trace that should show the problem quite well:
[btmon-linklike-elite10.zip](https://github.com/bluez/bluez/files/10288197/btmon-linklike-elite10.zip)

fwiw: my sound system stack works fine with an older headset of that brand, a Linklike Fly 9.

Any ideas on what I can try to make the Elite 10 work?

Thanks!

Vudentz, 2022-12-22:
The connection is timing out:

```
< ACL Data TX: Handle 256 flags 0x00 dlen 6                                                                              #114 [hci0] 11.857889
      Channel: 67 len 2 [PSM 25 mode Basic (0x00)] {chan 1}
      AVDTP: Discover (0x01) Command (0x00) type 0x00 label 0 nosp 0
= bluetoothd: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 20:05:20:07:5C:57: Device or resource busy    12.857676
= bluetoothd: profiles/audio/avdtp.c:cancel_request() Discover: Connection timed out (110)                                           17.861452
> HCI Event: Disconnect Complete (0x05) plen 4                                                                           #117 [hci0] 39.397996
        Status: Success (0x00)
        Handle: 256
        Reason: Connection Timeout (0x08)
```

We waited for 6 seconds without any response.

philfry, 2022-12-23:
Thanks.
Since the headset is working fine with other devices – like an iphone and android phone, is there anything I can do? I suspect increasing the timeout (assuming that possibility is given) might not help much because the *device or resource busy*-thingy might lead to a situation where the headset won't answer at all.

pv, 2022-12-27:
This is likely Pipewire bug (malformed +CIND) fixed in master branch, so probably will be fixed in next update.

philfry, 2023-01-06:
Hi,
I switched from pipewire to pulseaudio using
```
dnf swap --allowerasing pipewire-pulseaudio pulseaudio
```
but no luck either. Guess I'll wait for the next pipewire release and then try again.