bluez/bluez #452: Zsh completion for options is broken

ratijas, 2023-01-07:
Zsh completion is broken for options, i.e. if you type in `bluetoothctl -` and hit Tab.

[image]

ratijas, 2023-01-07:
Arch linux, latest, bluez-utils 5.66-1

ratijas, 2025-06-08:
What do you mean "automatically closed"?! Zsh completions are still broken!

Vudentz, 2025-07-17:
@ratijas does this fix your problem: https://patchwork.kernel.org/project/bluetooth/patch/20250624003551.107489-1-ronan@rjp.ie/

ratijas, 2025-07-23:
Idk, is there a simple way to get the final version of that file without cloning the whole Linux kernel tree and figuring out how email?

ratijas, 2025-07-23:
With shell.c patch applied and installed, the base command's completions don't bug out like on that left pane anymore. But everything else like `bluetoothctl -<tab>` is still broken.